Koenkk/zigbee2mqtt #14332: RTCGQ11LM (Aquara motion sensor) pairs but never returns any state data

cayco, 2022-10-04:
### What happened?

I was able to pair the sensor (tried 2 of them) but they only report data such as: IEEE Address, Network address and LQI (always 255) . They never report occupancy, batter, light, temp, etc. I tried 3 new batteries, disabled Conbee dongle, re-paired and rebooted zigbee2mqqt + HA several times but no change.

Here's the view after pairing: https://imgur.com/UjBQVPR

Other zigbee devices so far work fine. This Aquara sensor works without a problem on Conbee. 

Coordinator type: EZSP v8

I'm attaching a log from pairing.

### What did you expect to happen?

Sensor should report state data

### How to reproduce it (minimal and precise)

Pair RTCGQ11LM device

### Zigbee2MQTT version

1.28.0-1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus-E (EFR32MG21 + CH9102)

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9709334/log.txt)

kirovilya, 2022-10-04:
thanks for the log. I see interesting things in it

cayco, 2022-10-04:
You're welcome! Let me know if/how I can help!

kirovilya, 2022-10-04:
can you try using a stick with usb cable? looking by the log - some problems with data transfer between z2m and stick

cayco, 2022-10-04:
I was using cable (the same that Conbee works on) but I admin I also used externally powered USB hub. Without the hub - after re-pairing - everything works! Interestingly, Conbee works on that USB hub, no problem at all. 

That's for the lead!

closed 2022-10-04 by cayco