Koenkk/zigbee2mqtt #14333: Moes ZTS-US_1gang wrongly paired as Moes ZS_EUB_1gang

relombardo, 2022-10-04:
### What happened?

My Moes ZTS-US_1gang wall switch is wrongly paired as Moes [ZS_EUB_1gang](https://www.zigbee2mqtt.io/devices/ZS-EUB_1gang.html#moes-zs-eub_1gang). I do have 3gang version (Moes ZTS-US_3gang) that is also wrongly paired as Tuya [TS0013](https://www.zigbee2mqtt.io/devices/TS0013.html#tuya-ts0013). 

I had entered this [problem report](https://github.com/Koenkk/zigbee2mqtt/issues/13448), for my 3-gang switch, that was actually resolved as per my [latest comment](https://github.com/Koenkk/zigbee2mqtt/issues/13448#issuecomment-1267351810).

My goal is to have the entities "_backlight_mode" and "_power_on_behavior" available for my 1-gang version switch, no matter if it is still being paired as a different model. I am wondering if it would be possible to mimic the same existing implementation of the 3-gang version, even if it is pointing to Tuya TS0013 or even TS0011 which is the 1-gang version.

### What did you expect to happen?

 Entities "_backlight_mode" and "_power_on_behavior" available even if the model is paired as a different model

### Zigbee2MQTT version

1.28.0-1

### Adapter firmware version

0x26780700

### Adapter

ConBee2

Koenkk, 2022-10-05:
ZS_EUB_1gang supports backlight_mode and power_on_behaviour: https://www.zigbee2mqtt.io/devices/ZS-EUB_1gang.html#power-on-behavior-enum

relombardo, 2022-10-05:
This is true. Thanks for sharing. I just found the entities to switch these options and they have worked.

closed 2022-10-05 by Koenkk