Koenkk/zigbee2mqtt #14335: Yale YDF40 door locker with "Null" payload for fingerprint unlocking

relombardo, 2022-10-04:
### What happened?

Yale YDF40 door locker is sending "Null" payload for fingerprint unlocking as follows:

Zigbee2MQTT:info  2022-10-04 15:37:11: MQTT publish: topic 'zigbee2mqtt/Fechadura Entrada', payload '{"action":null,"action_source_name":null,"action_user":null,"battery":90,"linkquality":255,"lock_state":"unlocked","state":"UNLOCK"}'

I can map all other actions. For examplesusers unlocking with passcode, which sends a message payload.action_user with the respective user through the source payload.action_source =  0 as follows: 

Zigbee2MQTT:info  2022-10-04 15:43:11: MQTT publish: topic 'zigbee2mqtt/Fechadura Entrada', payload '{"action":"unlock","action_source":0,"action_source_name":"keypad","action_user":65535,"battery":90,"linkquality":255,"lock_state":"locked","state":"LOCK"}'

Regarding the fingerprint unlocking, I have seen YMF40 users being able to capture that through the payload.action_source = 4. I believe I should be able to do the same with the YDF40 as their documentation seems to be the same in that aspect.

### What did you expect to happen?

The Fingerprint unlocking action should present some specific payload message, different than Null, especially because it can have multiple users associated, such as Keypad unlocking action. YMF40 does that, so I expected YDF40 to do the same.

### Zigbee2MQTT version

1.28.0-1

### Adapter firmware version

0x26780700

### Adapter

ConBee2

everson7, 2022-10-04:
I'm having the same problem!

danieledwardgeorgehitchcock, 2022-10-05:
Could you enable the Z2M debug logging and the Herdsman debug logging, and perform a fingerprint unlock on the device - there may be a lot of log chat in your console, but it may help track down the issue..

relombardo, 2022-10-07:
Hi @Koenkk ! Would it be possible to use external converter of YMF40 in my YDF40? Or is there a way to fix the YDF40 itself?

everson7, 2022-10-08:
Debug 2022-10-07 21:43:33Received Zigbee message from 'Yale', type 'commandCheckIn', cluster 'genPollCtrl', data '{}' from endpoint 1 with groupID null
Debug 2022-10-07 21:45:03Received Zigbee message from 'Yale', type 'attributeReport', cluster 'closuresDoorLock', data '{"lockState":2}' from endpoint 1 with groupID null
Info 2022-10-07 21:45:03MQTT publish: topic 'zigbee2mqtt/Yale', payload '{"action_source_name":null,"action_user":null,"battery":82,"linkquality":61,"lock_state":"unlocked","state":"UNLOCK"}'
Debug 2022-10-07 21:45:03Received Zigbee message from 'Yale', type 'commandCheckIn', cluster 'genPollCtrl', data '{}' from endpoint 1 with groupID null
Debug 2022-10-07 21:45:11Received Zigbee message from 'Yale', type 'attributeReport', cluster 'closuresDoorLock', data '{"lockState":2}' from endpoint 1 with groupID null
Info 2022-10-07 21:45:11MQTT publish: topic 'zigbee2mqtt/Yale', payload '{"action_source_name":null,"action_user":null,"battery":82,"linkquality":28,"lock_state":"unlocked","state":"UNLOCK"}'
Debug 2022-10-07 21:45:15Received Zigbee message from 'Yale', type 'attributeReport', cluster 'closuresDoorLock', data '{"lockState":1}' from endpoint 1 with groupID null
Info 2022-10-07 21:45:15MQTT publish: topic 'zigbee2mqtt/Yale', payload '{"action_source_name":null,"action_user":null,"battery":82,"linkquality":91,"lock_state":"locked","state":"LOCK"}'

everson7, 2022-10-08:
As you can see mine is always null.
I can only see the quality of the link, battery and lock state

relombardo, 2022-11-08:
Any help? The issue was not resolved and it is affecting other users of Yale YDF40.

relombardo, 2022-12-01:
Hi @Koenkk, could you help us with this issue?

gregolin, 2022-12-20:
The same happening here.
The full log (I started with a new pairing etc) is attached.
As you can see in the log, it shows me correctly when I open the lock internally (line 90 for instance), but when I open it externally with fingerprint, I can't see the action, action_source_name or action_user (next-to-last, 94, for instance). The most strange is that everything works fine when I try to open it with password (lin
e 95).
Thanks.

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10264090/log.txt)

gregolin, 2022-12-20:
@relombardo , any news about this issue from other source?

relombardo, 2022-12-20:
> @relombardo , any news about this issue from other source?

@gregolin , unfortunately I haven't found a solution. Hopefully we can get some guidance shortly. Another user on this thread, @everson7, also had same issue, which makes me think it is persistent outcome of this model so far. And you are right, using password is fine, the problem occurs with fingerprint.

relombardo, 2022-12-20:
Hi @danieledwardgeorgehitchcock, I noticed @everson7 shared logs you asked and now @gregolin also shared his logs. Would you have any comment on it?

gregolin, 2022-12-20:
@danieledwardgeorgehitchcock , I have one more log file (from other period of the day) with Herdsman log as well. Included here.
[log_herdsman.txt](https://github.com/Koenkk/zigbee2mqtt/files/10271806/log_herdsman.txt)

Koenkk, 2022-12-21:
Let's continue in https://github.com/Koenkk/zigbee2mqtt/issues/15665

closed 2022-12-21 by Koenkk