Koenkk/zigbee2mqtt #14336: TS011F_din_smart_relay can´t update. But other from same manufracture can

henkiejan1, 2022-10-04:
### What happened?

A few days ago is saw there was a firmware update for some Tuys TS011F power plugs and also for my recently bought din rail switch. All installed without problems. But strang enough do i got for my other din rail switch an error when i check for a firmware update. But bot look very similar. Could someone help?

Status info of the one that can not be updated:
`{
    "current": 0,
    "energy": 407.8,
    "indicator_mode": "on_off",
    "last_seen": "2022-10-04T19:25:17.301Z",
    "linkquality": 149,
    "power": 0,
    "power_outage_memory": "restore",
    "state": "OFF",
    "voltage": 226,
    "device": {
        "applicationVersion": 67,
        "dateCode": "",
        "friendlyName": "Boiler switch",
        "hardwareVersion": 1,
        "ieeeAddr": "0x84ba20fffe6d50c9",
        "manufacturerID": 4098,
        "manufacturerName": "_TZ3000_8bxrzyxz",
        "model": "TS011F_din_smart_relay",
        "networkAddress": 27479,
        "powerSource": "Mains (single phase)",
        "stackVersion": 0,
        "type": "Router",
        "zclVersion": 3
    },
    "update": {
        "state": null
    },
    "update_available": null
}`

And status info for the one that could update:

`{
    "current": 0.19,
    "energy": 14.37,
    "indicator_mode": "on_off",
    "last_seen": "2022-10-04T19:27:15.384Z",
    "linkquality": 98,
    "power": 28,
    "power_outage_memory": "restore",
    "state": "OFF",
    "update": {
        "state": "idle"
    },
    "update_available": false,
    "voltage": 233,
    "device": {
        "applicationVersion": 192,
        "dateCode": "",
        "friendlyName": "WTW Boost switch",
        "hardwareVersion": 1,
        "ieeeAddr": "0xa4c138c2b90edb44",
        "manufacturerID": 4417,
        "manufacturerName": "_TZ3000_ky0fq4ho",
        "model": "TS011F_din_smart_relay",
        "networkAddress": 48073,
        "powerSource": "Mains (single phase)",
        "stackVersion": 0,
        "type": "Router",
        "zclVersion": 3
    }
}`

### What did you expect to happen?

That the update could be installed.

### How to reproduce it (minimal and precise)

Click on the ´Check for update button under OTA and get the message:
Failed to check if update available for 'Boiler switch' (No image available for imageType '5634')","status":"error","transaction":"i1nhw-6"}'

### Zigbee2MQTT version

1.28.1

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-P

### Debug log

info  2022-10-04 21:37:13: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Checking if update available for 'Boiler switch'","meta":{"device":"Boiler switch","status":"checking_if_available"}>
debug 2022-10-04 21:37:13: Check if update available for '0x84ba20fffe6d50c9' (TS011F)
debug 2022-10-04 21:37:13: Using endpoint '1'
debug 2022-10-04 21:37:13: Received Zigbee message from 'Boiler switch', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":67,"imageType":5634,"manufacturerCode":4098}>
info  2022-10-04 21:37:13: MQTT publish: topic 'zigbee2mqtt/Boiler switch', payload '{"current":0,"device":{"applicationVersion":67,"dateCode":"","friendlyName":"Boiler switch","hardwareVersion":1,"ieeeAddr":"0>
debug 2022-10-04 21:37:13: Got OTA request '{"fieldControl":0,"manufacturerCode":4098,"imageType":5634,"fileVersion":67}'
debug 2022-10-04 21:37:14: ZigbeeOTA: downloaded main index
info  2022-10-04 21:37:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Failed to check if update available for 'Boiler switch' (No image available for imageType '5634')","meta":{"device":>
info  2022-10-04 21:37:14: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Boiler switch"},"error":"Failed to check if update available for 'Boiler switch' (No>
error 2022-10-04 21:37:14: Failed to check if update available for 'Boiler switch' (No image available for imageType '5634')
debug 2022-10-04 21:37:14: Error: No image available for imageType '5634'
    at getImageMeta (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/zigbeeOTA.js:113:15)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at isNewImageAvailable (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/common.js:265:18)
    at Object.isUpdateAvailable (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/common.js:256:23)
    at OTAUpdate.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/otaUpdate.ts:194:39)
debug 2022-10-04 21:37:16: Received Zigbee message from 'Boiler switch', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":67,"imageType":5634,"manufacturerCode":4098}>
info  2022-10-04 21:37:16: MQTT publish: topic 'zigbee2mqtt/Boiler switch', payload '{"current":0,"device":{"applicationVersion":67,"dateCode":"","friendlyName":"Boiler switch","hardwareVersion":1,"ieeeAddr":"0>
debug 2022-10-04 21:37:16: Device 'Boiler switch' requested OTA

henkiejan1, 2022-10-05:
> We don't have the OTA updates for all manufaturerCodes yet. To grab yours see https://www.zigbee2mqtt.io/advanced/more/tuya_xiaomi_ota_url.html

Do you think it´s a different manufacturer?  The switch broke if i should flash the other one i think? I don´t have a Tuya Bridge... :)

Koenkk, 2022-10-05:
I dont know if it bricks but we had such cases in the past therefore the manufacturername check has been added.

closed 2022-11-12 by github-actions[bot]