Koenkk/zigbee2mqtt #14337: Autodiscovery HomeAssistant realization in Z2M maybe is a bit wrong?

Nikolay-Ch, 2022-10-04:
### What happened?

According to the docs of the HA): _The discovery topic needs to follow a specific format:_
_<discovery_prefix>/<cоmponent>/[<node_id>/]<object_id>/config_

Unique **object ID** need to be _forth_ part of the topic, but Z2M placed it on _third_ place. Third part are optional and ignored by HA. 
EX: topic, that Z2M create is like this now:
homeassistant/sensor/0x8cf681aaaaaaaa42/voltage/config

But more correct topic need to be like this (to be unique):
homeassistant/sensor/0x8cf681aaaaaaaa42-voltage/config

### Zigbee2MQTT version

 1.28.0 commit: 03ba647

### Adapter firmware version

20211222

### Adapter

zStack30x

Koenkk, 2022-10-05:
Does this cause any issue currently?

Nikolay-Ch, 2022-10-07:
I do not know, maybe. Or maybe will cause a bug in the future...
I just saw an inconsistency between the implementation in Z2M and the HA documentation

m0wlheld, 2022-10-14:
As you said - node_id is optional and missing, making object_id the 3rd path element. 

Not a bug.

Nikolay-Ch, 2022-10-15:
> As you said - node_id is optional and missing, making object_id the 3rd path element.
> 
> Not a bug.

Not quite... **node_id** is optional you are right, but **object_id** is not optional and **object_id** need to be at forth place if **node_id** is present.

Current realization Z2M use both identifiers, but **node_id** at the forth place and **object_id** at third place. I think, that is wrong.