Koenkk/zigbee2mqtt #14338: [Feature request]: Support fractions in transitions.

SelfDestructor, 2022-10-04:
### Is your feature request related to a problem? Please describe

I'm currently attempting to move my Zigbee lamps off my Philips Hue bridges and I have hit a bit of a snag with transition times.

When changing a light state via MQTT I am able to use fractions in the transition field.
Hue defaults to 0.4 seconds which is not too slow or too quick to my liking and I can do so through MQTT but I am not able to set transition times for scene changes which is my primary way to turn on lamps as different brands has slightly different colors and white tones.

If I set the transition time to 0.-something it becomes zero seconds.

Example in configuration.yaml: 

  '3':
    friendly_name: office_lights
    transition: 0.4
    devices:
      - 0x0/1

Example through MQTT:

{
    "scene_add": {
        "ID": 0,
        "name": "'Light scene",
        "transition": 0.4
    }
}

### Describe the solution you'd like

Since I can send`{"transition": 0.4, "kelvin": 2700, "brightness_pct": 100}` through MQTT I would love to be able to set scenes and group(light fade times to something shorter than a second to mirror how my setup was with the Hue hub.

### Describe alternatives you've considered

The quickest option would be to go back to a Hue hub and continue calling scenes from there.

Or I can send MQTT messages to the group. I would have to find a way to deal with the non-Hue bulbs I currently have to match colors.

### Additional context

Nothing else to add.

closed 2022-11-11 by github-actions[bot]

Flavien, 2023-07-09:
Is there a workaround for this? I'm trying to have a transition that is just a few milliseconds, but it seems right now, it's either 0 second or 1 second, nothing in between.

Zoooook, 2024-07-02:
I'm having the same issue, and I have to use scenes because that's the only way to transition brightness and color at the same time. If I simply set the state of the lights, then it fades the brightness and then fades the color. But with scenes the minimum non-instant transition is 1 second.

Flavien, 2024-07-08:
This was fixed in [1.35.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.35.0) (PR: https://github.com/Koenkk/zigbee-herdsman-converters/pull/6656).

Stooovie, 2025-10-29:
But the Z2M UI (on each device) doesn't let me set a fractional. It rounds up or down to a full second. For my[ IKEA bulb](https://www.zigbee2mqtt.io/devices/LED2003G10.html#ikea-led2003g10) anyway.