Koenkk/zigbee2mqtt #14344: Frontend dashboard devices list is repeatedly empty

blacknell, 2022-10-05:
### What happened?

Recently, the list of devices  in the zigbee2mqtt dashboard comes up blank. I have to refresh it several times and/or start the dashboard map and then revisit the devices tab (this doesn't always work). All the devices themselves are working as expected.

### What did you expect to happen?

I expect to see the list of devices almost immediately.

### How to reproduce it (minimal and precise)

My system is currently seeing this all the time.

### Zigbee2MQTT version

1.28.0

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus × 1

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9713582/log.txt)

JackTalisker, 2022-10-05:
I had a similar problem.
Devices list became empty but in my case all devices had gone and unoperational. My database.db suddenly dropped from 32k to 1k.
I had to reinstall 1.27.2 with a database backup then upgrade again to 1.28.0.
I'm still experimenting strage behaviours from some end devices which are under investigation.

blacknell, 2022-10-10:
Any ideas @Koenkk ?

Anything I can enable to try and uncover what is going on?

blacknell, 2022-10-22:
This is very frustrating. @Koenkk - Anything we can do to help diagnose?

Koenkk, 2022-10-22:
As stated [here](https://github.com/Koenkk/zigbee2mqtt/issues/new/choose) issues related to the frontend should be created here: https://github.com/nurikk/zigbee2mqtt-frontend/issues

blacknell, 2022-10-24:
Moved to [frontend repo](https://github.com/nurikk/zigbee2mqtt-frontend/issues/1587)

closed 2022-10-24 by blacknell