Koenkk/zigbee2mqtt #14347: NYCE NCZ-3041-HA not reporting temperature and humidity

andresp, 2022-10-05:
### What happened?

Joined NCZ-3041-HA won't report humidity or temperature. Currently it just reports the following state:

```json
{
    "battery_low": false,
    "last_seen": "2022-10-05T13:57:33.381Z",
    "linkquality": 66,
    "occupancy": false,
    "tamper": false
}
```

### What did you expect to happen?

Humidity and Temperature being reported at least once an hour.

### Zigbee2MQTT version

1.28.0

### Adapter firmware version

20210708

### Adapter

Sonoff_Zigbee_3.0

andresp, 2022-11-05:
Can the stale tag be removed as this issue persists?

Koenkk, 2022-11-05:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['3041'],
    model: 'NCZ-3041-HA',
    vendor: 'Nyce',
    description: 'Wall motion sensor',
    fromZigbee: [fz.occupancy, fz.humidity, fz.temperature, fz.ignore_basic_report, fz.ignore_genIdentify, fz.ignore_poll_ctrl,
        fz.battery, fz.ignore_iaszone_report, fz.ias_occupancy_alarm_2],
    toZigbee: [],
    meta: {battery: {dontDividePercentage: true}},
    exposes: [e.occupancy(), e.humidity(), e.temperature(), e.battery(), e.battery_low(), e.tamper()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['msTemperatureMeasurement', 'msRelativeHumidity']);
        await reporting.temperature(endpoint);
        await reporting.humidity(endpoint);
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

closed 2022-12-14 by github-actions[bot]

andresp, 2022-12-14:
Apologies for responding late, but I have in fact used the following converter (almost identical to the one suggested above):

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['3041'],
    model: 'NCZ-3041-HA',
    vendor: 'Nyce',
    description: 'Ceiling motion sensor CUSTOM',
    fromZigbee: [fz.occupancy, fz.humidity, fz.temperature, fz.ignore_basic_report, fz.ignore_genIdentify, fz.ignore_poll_ctrl,
        fz.battery, fz.ignore_iaszone_report, fz.ias_occupancy_alarm_2],
    toZigbee: [],
    exposes: [e.occupancy(), e.humidity(), e.temperature(), e.battery(), e.battery_low(), e.tamper()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'msTemperatureMeasurement', 'msRelativeHumidity']);
        await reporting.batteryPercentageRemaining(endpoint);
        await reporting.temperature(endpoint);
        await reporting.humidity(endpoint);
    },
};

module.exports = definition;
```

With this converter the sensors do function in reporting the extra data, however new sensors can no longer be joined with the converter in place. The converter has to be disabled before a new sensor can be joined, so something is blocking the discovery of a new sensor with the converter in place.