Koenkk/zigbee2mqtt #14350: Centralite 4200-C plug fails to interview

akrauze, 2022-10-05:
### What happened?

Centralite 4200-C plug fails to interview with message: Failed to interview '0x5c0272fffecaedd4', device has not successfully been paired
Device remains in pairing mode and shows as:
Device type:  Router
Zigbee Model:
Zigbee Manufacturer: 
Support status: Not supported

Note1: Tested on Zigbee2MQTT on both latest and latest-dev docker images.
Note2: Same Centralite 4200-C plug with same controller works with Home Assistant's ZHA integration just fine

### What did you expect to happen?

Device should pair.

### How to reproduce it (minimal and precise)

Reset 4200-C plug. Pairing starts and fails, but device remains in pairing mode.

### Zigbee2MQTT version

1.28.0-dev commit: 37aea8d

### Adapter firmware version

20220219

### Adapter

Sonoff 2.0 USB Dongle Plus (zStack3x0)

### Debug log

2022-10-05T18:51:16.363Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,13,69,201,172,101,212,237,202,254,255,114,2,92,0,0,0,150]
2022-10-05T18:51:16.363Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,13,69,201,172,101,212,237,202,254,255,114,2,92,0,0,0,150]
2022-10-05T18:51:16.364Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 13 - 2 - 5 - 201 - [172,101,212,237,202,254,255,114,2,92,0,0,0] - 150
2022-10-05T18:51:16.364Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - leaveInd - {"srcaddr":26028,"extaddr":"0x5c0272fffecaedd4","request":0,"removechildren":0,"rejoin":0}
2022-10-05T18:51:16.364Z zigbee-herdsman:controller:log Device leave '0x5c0272fffecaedd4'
2022-10-05T18:51:16.369Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-10-05T18:51:16.841Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,13,69,201,172,101,212,237,202,254,255,114,2,92,0,0,0,150]
2022-10-05T18:51:16.841Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,13,69,201,172,101,212,237,202,254,255,114,2,92,0,0,0,150]
2022-10-05T18:51:16.841Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 13 - 2 - 5 - 201 - [172,101,212,237,202,254,255,114,2,92,0,0,0] - 150
2022-10-05T18:51:16.841Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - leaveInd - {"srcaddr":26028,"extaddr":"0x5c0272fffecaedd4","request":0,"removechildren":0,"rejoin":0}
2022-10-05T18:51:16.841Z zigbee-herdsman:controller:log Device leave '0x5c0272fffecaedd4'
2022-10-05T18:51:16.847Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-10-05T18:51:17.311Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,13,69,201,172,101,212,237,202,254,255,114,2,92,0,0,0,150]
2022-10-05T18:51:17.311Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,13,69,201,172,101,212,237,202,254,255,114,2,92,0,0,0,150]
2022-10-05T18:51:17.311Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 13 - 2 - 5 - 201 - [172,101,212,237,202,254,255,114,2,92,0,0,0] - 150
2022-10-05T18:51:17.312Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - leaveInd - {"srcaddr":26028,"extaddr":"0x5c0272fffecaedd4","request":0,"removechildren":0,"rejoin":0}
2022-10-05T18:51:17.312Z zigbee-herdsman:controller:log Device leave '0x5c0272fffecaedd4'
2022-10-05T18:51:17.318Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-10-05T18:51:18.286Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,4,105,212,237,202,254,255,114,2,92,0,0,48]
2022-10-05T18:51:18.286Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,4,105,212,237,202,254,255,114,2,92,0,0,48]
2022-10-05T18:51:18.286Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [4,105,212,237,202,254,255,114,2,92,0,0] - 48
2022-10-05T18:51:18.286Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":26884,"extaddr":"0x5c0272fffecaedd4","parentaddr":0}
2022-10-05T18:51:18.287Z zigbee-herdsman:controller:log Device '0x5c0272fffecaedd4' joined
2022-10-05T18:51:18.288Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-10-05T18:51:18.288Z zigbee-herdsman:controller:log Device '0x5c0272fffecaedd4' accepted by handler
2022-10-05T18:51:18.288Z zigbee-herdsman:controller:log New device '0x5c0272fffecaedd4' joined
2022-10-05T18:51:18.288Z zigbee-herdsman:controller:log Creating device '0x5c0272fffecaedd4'
2022-10-05T18:51:18.333Z zigbee-herdsman:controller:log Interview '0x5c0272fffecaedd4' start
2022-10-05T18:51:18.339Z zigbee-herdsman:controller:device:log Interview - start device '0x5c0272fffecaedd4'
2022-10-05T18:51:18.342Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":26884,"nwkaddrofinterest":26884}
2022-10-05T18:51:18.342Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,4,105,4,105,35]
2022-10-05T18:51:18.353Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2022-10-05T18:51:18.353Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2022-10-05T18:51:18.353Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2022-10-05T18:51:18.353Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2022-10-05T18:51:18.353Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
[... 23 lines cut ...]
2022-10-05T18:51:19.428Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,4,105,0,239]
2022-10-05T18:51:19.428Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [4,105,0] - 239
2022-10-05T18:51:19.428Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":26884,"relaycount":0,"relaylist":[]}
2022-10-05T18:51:19.428Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-10-05T18:51:29.425Z zigbee-herdsman:controller:device:log Interview - active endpoints request failed for '0x5c0272fffecaedd4', attempt 1
2022-10-05T18:51:29.426Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - activeEpReq - {"dstaddr":26884,"nwkaddrofinterest":26884}
2022-10-05T18:51:29.426Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,5,4,105,4,105,36]
2022-10-05T18:51:29.441Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,5,0,97]
2022-10-05T18:51:29.441Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,5,0,97]
2022-10-05T18:51:29.441Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 5 - [0] - 97
2022-10-05T18:51:29.441Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - activeEpReq - {"status":0}
2022-10-05T18:51:29.442Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-10-05T18:51:29.459Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,4,105,0,239]
2022-10-05T18:51:29.459Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,4,105,0,239]
2022-10-05T18:51:29.460Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [4,105,0] - 239
2022-10-05T18:51:29.460Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":26884,"relaycount":0,"relaylist":[]}
2022-10-05T18:51:29.460Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-10-05T18:51:39.451Z zigbee-herdsman:controller:device:log Interview - active endpoints request failed for '0x5c0272fffecaedd4', attempt 2
2022-10-05T18:51:39.452Z zigbee-herdsman:controller:device:log Interview - quirks check for 'undefined'-'undefined'-'Router'
2022-10-05T18:51:39.452Z zigbee-herdsman:controller:device:log Interview - quirks did not match
2022-10-05T18:51:39.608Z zigbee-herdsman:controller:device:log Interview - failed for device '0x5c0272fffecaedd4' with error 'Error: Interview failed because can not get active endpoints ('0x5c0272fffecaedd4')
    at Device.interviewInternal (/app/node_modules/zigbee-herdsman/src/controller/model/device.ts:596:19)
    at Device.interview (/app/node_modules/zigbee-herdsman/src/controller/model/device.ts:445:13)
    at Controller.onDeviceJoined (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:567:17)'
2022-10-05T18:51:39.608Z zigbee-herdsman:controller:error Interview failed for '0x5c0272fffecaedd4 with error 'Error: Interview failed because can not get active endpoints ('0x5c0272fffecaedd4')'

akrauze, 2022-10-06:
Found a post somewhere in the internet about someone having issues pairing a Centralite 4200-C to a Hubitat with a suggestion to change the zigbee channel to 24. It resolved the issue and zigbee2mqtt was able to interview the device.

That resolved my problem.

closed 2022-10-06 by akrauze

dcarpio01, 2024-03-22:
This also resolved my issue with the centralite 4200-C. I'm going to try and change the zigbee channel to 25 and see if it stays connected.

dcarpio01, 2024-03-22:
Changing to Channel 15, 20, 25 did work after first pairing on Channel 24.

dcarpio01, 2024-03-23:
Spoke to soon. It stopped working shortly after I paired more 4200-C plugs.