Koenkk/zigbee2mqtt #14356: Silvercrest HG08673-FR: no value for 'energy'

EnerVision, 2022-10-05:
### What happened?

The value for energy is not being read
[image]

### What did you expect to happen?

Should have been read

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

0x26780700

### Adapter

Conbee2

### Debug log

2022-10-05 22:22:54Received Zigbee message from 'Droogkast', type 'readResponse', cluster 'haElectricalMeasurement', data '{"activePower":92,"rmsCurrent":545,"rmsVoltage":236}' from endpoint 1 with groupID null
Info 2022-10-05 22:22:54MQTT publish: topic 'zigbee2mqtt/Droogkast', payload '{"current":0.55,"linkquality":220,"power":92,"power_outage_memory":"off","state":"ON","voltage":236}'
Info 2022-10-05 22:22:54MQTT publish: topic 'zigbee2mqtt/Droogkast/state', payload 'ON'
Info 2022-10-05 22:22:54MQTT publish: topic 'zigbee2mqtt/Droogkast/power_outage_memory', payload 'off'
Info 2022-10-05 22:22:54MQTT publish: topic 'zigbee2mqtt/Droogkast/power', payload '92'
Info 2022-10-05 22:22:54MQTT publish: topic 'zigbee2mqtt/Droogkast/current', payload '0.55'
Info 2022-10-05 22:22:54MQTT publish: topic 'zigbee2mqtt/Droogkast/voltage', payload '236'
Info 2022-10-05 22:22:54MQTT publish: topic 'zigbee2mqtt/Droogkast/linkquality', payload '220'

blackie333, 2022-10-06:
I can confirm the problem in version 1.28.0-1.
[image]

Koenkk, 2022-10-07:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_ynmowqk2'}],
    model: 'HG08673-FR',
    vendor: 'Lidl',
    description: 'Silvercrest smart plug FR with power monitoring',
    fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report, fz.tuya_switch_power_outage_memory],
    toZigbee: [tz.on_off, tz.tuya_switch_power_outage_memory],
    exposes: [e.switch(), e.power(), e.current(), e.voltage().withAccess(ea.STATE),
        e.energy(), exposes.enum('power_outage_memory', ea.ALL, ['on', 'off', 'restore'])
            .withDescription('Recover state after power outage')],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
        endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
        endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {
            acVoltageMultiplier: 1, acVoltageDivisor: 1, acCurrentMultiplier: 1, acCurrentDivisor: 1000, acPowerMultiplier: 1,
            acPowerDivisor: 1,
        });
    },
    options: [exposes.options.measurement_poll_interval()],
    onEvent: tuya.onEventMeasurementPoll,
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

blackie333, 2022-10-07:
Thanks, but seems that the issue persists.
[image]

EnerVision, 2022-10-07:
I also confirm it not working (although I eventually pasted it into the lidl.js file, because apparently I did something wrong and zigbee2mqtt produced some errors)

blackie333, 2022-10-07:
Not sure if it helps but I realized that after Z2M restart the "energy" line is missing from "State" window (other properties are already there with values) and finally after cca 1 minute "energy" appears but with "null" value.

Koenkk, 2022-10-08:
@radaoggy  since you added this device; does `energy` work for you?

blackie333, 2022-10-08:
"Energy" property is reporting "null" value for the whole time since it appeared in the Z2M 1.28.
Adding the .js external_converter did not change the result.
Should I try to remove and add the device to my zigbee network again?

radaoggy, 2022-10-08:
> @radaoggy  since you added this device; does `energy` work for you?

No, it doesn't work all the time and shows “N/A”. But I didn't deal with it in any way, I don't use it.  I use Integration - Riemann sum integral instead. The rest works fine.

Koenkk, 2022-10-08:
I updated https://github.com/Koenkk/zigbee2mqtt/issues/14356#issuecomment-1271712889, can you check again?

After loading the new converter go to the z2m frontend -> device -> press yellow refresh (reconfigure) button. Does it say "configure successful"?

blackie333, 2022-10-08:
Have tried, no "configure successful" message and the same energy null behaviour.

Koenkk, 2022-10-09:
Then I would expect energy is not supported, to cross check, does anyone have a TuYa gateway to check with?

blackie333, 2022-10-09:
It's a pity that energy is the problematic property because it's the most important for me. 
It was the main reason for buying this plug. But I still hope that some kind of solution can be found.
Thanks to all involved and someone with the TuYa/Lidl gateway please check/confirm whether energy is working there as Mr. Koenkk asks.

EnerVision, 2022-10-09:
I don't have the TuYa gateway either. 
Are any of these files helpfull in fixing the problem (I'm not an IT expert)? I've found them somewhere on the internet specific for the energy smart plug.  I saw somewhere in these files something about "Power Profile".
[16-02802-002-0x0003-Identify-Cluster-XML-PICS.zip](https://github.com/Koenkk/zigbee2mqtt/files/9742446/16-02802-002-0x0003-Identify-Cluster-XML-PICS.zip)

Koenkk, 2022-10-10:
@EnerVision in the code I provided we already setup energy reporting via the standard Zigbee commands but the device doesn't adhere to it. So only way to find out is via the TuYa gateway.

EnerVision, 2022-10-10:
Koen, if that is the solution, I can buy a tuya zigbee gateway and check it out for you. But you'll have to talk me trough the proces of checking the behaviour on that gateway (what and how to check) as I'm not that clever to find it all out on my own.

EnerVision, 2022-10-10:
Just bought it, should be here tomorrow or the day after... and, Koen, of course I meant "would be much appreciated if you could..." instead of "have to..."  ;-)

EnerVision, 2022-10-11:
Got it today. Yes Tuya gateway shows energyconsumption. It says however "monthly" kWh. Perhaps you also need to send a time interval??
[image]

Koenkk, 2022-10-11:
That is interesting, can you find out the datapoints of this switch? 

https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#requirements-and-caveats

EnerVision, 2022-10-11:
[image]

EnerVision, 2022-10-11:
[image]

EnerVision, 2022-10-11:
For the requested payload: it states "17".

Is that what you needed?

EnerVision, 2022-10-11:
... for the "Add electricity" DP ID which states an event in kwh

Koenkk, 2022-10-12:
Interesting, seems this device can also work via the TuYa api. Can you update to the latest dev (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) and try the following external converter: https://gist.github.com/Koenkk/81b27922a58bb2922bb9e52cbcf7d685 . Only energy should work, if that is the case also provide me the datapoints ids of the other values (like the `17` you provided for energy)

EnerVision, 2022-10-12:
Perhaps I did something wrong (I added the code you provided in lidl.js because last time the external converter broke the fontend somehow).
It still get N/A for the energyreading.
[image]

EnerVision, 2022-10-12:
[image]

EnerVision, 2022-10-12:
If it in anyway helps: in the developmentconsole. Selecting endpoint 1, cluster "seMetering", attribute "currentSummDelivered" gives me kWh shown in Tuya-App devided by 10 . Tuya-App gave me 1,65 kWh. zigbee2mqtt gave me 0,165. Unless it was totally random luck...
[image]

Koenkk, 2022-10-12:
- Can you provide the debug log when connecting a load to the switch? (leave it running for a few minutes). See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.
- Do you have a sniffer, if so, can you provide the sniff when pairing the switch to the TuYa gateway and opening the app to see the energy values? https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

> Unless it was totally random luck...

This is not random luck. Actually this is the standard zigbee way to provide energy measurements. However this device doesn't seem to support reporting of these values so we have to rely on polling (not nice). It would be nicer if we could use the TuYa api such that we can rely on reporting.

EnerVision, 2022-10-12:
Unfortunately, I don't have a sniffer. I'm using a conbee2. Apparently that won't do sniffing as I have read somewhere.

... pff debugging is a pain: stopped the service in putty by "sudo systemctl stop zigbee2mqtt ", adapted the configuration.yaml file, went to /opt/zigbee2mqtt and typed in "DEBUG=zigbee-herdsman* npm start", got following:
> zigbee2mqtt@1.28.0-dev start
> node index.js

/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:183
  return new YAMLException(message, mark);
         ^
YAMLException: duplicated mapping key (16:1)

 13 |   adapter: deconz
 14 |   port: >-
 15 |     /dev/serial/by-id/usb-dresd ...
 16 | advanced:
------^
 17 |   network_key:
 18 |     - 64
    at generateError (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:183:10)
    at throwError (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:187:9)
    at storeMappingPair (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:358:7)
    at readBlockMapping (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1173:9)
    at composeNode (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1441:12)
    at readDocument (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1625:3)
    at loadDocuments (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1688:5)
    at Object.load (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1714:19)
    at Object.read (/opt/zigbee2mqtt/lib/util/yaml.ts:7:21)
    at read (/opt/zigbee2mqtt/lib/util/settings.ts:311:20)

Did a "sudo journalctl -u zigbee2mqtt.service -f" and got this:
-- Journal begins at Wed 2022-10-12 20:20:00 CEST. --
Oct 12 21:38:37 openhabian systemd[1]: Stopping Zigbee2MQTT...
Oct 12 21:38:37 openhabian npm[1089]: Zigbee2MQTT:debug 2022-10-12 21:38:37: Saving state to file /opt/zigbee2mqtt/data/state.json
Oct 12 21:38:37 openhabian npm[1089]: Zigbee2MQTT:info  2022-10-12 21:38:37: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
Oct 12 21:38:37 openhabian npm[1089]: Zigbee2MQTT:info  2022-10-12 21:38:37: Disconnecting from MQTT server
Oct 12 21:38:37 openhabian npm[1089]: Zigbee2MQTT:info  2022-10-12 21:38:37: Stopping zigbee-herdsman...
Oct 12 21:38:39 openhabian npm[1089]: Zigbee2MQTT:info  2022-10-12 21:38:39: Stopped zigbee-herdsman
Oct 12 21:38:39 openhabian npm[1089]: Zigbee2MQTT:info  2022-10-12 21:38:39: Stopped Zigbee2MQTT
Oct 12 21:38:39 openhabian systemd[1]: zigbee2mqtt.service: Succeeded.
Oct 12 21:38:39 openhabian systemd[1]: Stopped Zigbee2MQTT.
Oct 12 21:38:39 openhabian systemd[1]: zigbee2mqtt.service: Consumed 18min 4.327s CPU time.

Got confused because it said "stopping", so i did "sudo systemctl start zigbee2mqtt" and afterwards "sudo journalctl -u zigbee2mqtt.service -f" with folowing outcome:
Oct 12 21:43:56 openhabian systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 4.
Oct 12 21:43:56 openhabian systemd[1]: Stopped Zigbee2MQTT.
Oct 12 21:43:56 openhabian systemd[1]: zigbee2mqtt.service: Consumed 5.916s CPU time.
Oct 12 21:43:56 openhabian systemd[1]: Started Zigbee2MQTT.
Oct 12 21:43:58 openhabian npm[8479]: > zigbee2mqtt@1.28.0-dev start
Oct 12 21:43:58 openhabian npm[8479]: > node index.js
Oct 12 21:44:01 openhabian npm[8503]: /opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:183
Oct 12 21:44:01 openhabian npm[8503]:   return new YAMLException(message, mark);
Oct 12 21:44:01 openhabian npm[8503]:          ^
Oct 12 21:44:01 openhabian npm[8503]: YAMLException: duplicated mapping key (16:1)
Oct 12 21:44:01 openhabian npm[8503]:  13 |   adapter: deconz
Oct 12 21:44:01 openhabian npm[8503]:  14 |   port: >-
Oct 12 21:44:01 openhabian npm[8503]:  15 |     /dev/serial/by-id/usb-dresd ...
Oct 12 21:44:01 openhabian npm[8503]:  16 | advanced:
Oct 12 21:44:01 openhabian npm[8503]: ------^
Oct 12 21:44:01 openhabian npm[8503]:  17 |   network_key:
Oct 12 21:44:01 openhabian npm[8503]:  18 |     - 64
Oct 12 21:44:01 openhabian npm[8503]:     at generateError (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:183:10)
Oct 12 21:44:01 openhabian npm[8503]:     at throwError (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:187:9)
Oct 12 21:44:01 openhabian npm[8503]:     at storeMappingPair (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:358:7)
Oct 12 21:44:01 openhabian npm[8503]:     at readBlockMapping (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1173:9)
Oct 12 21:44:01 openhabian npm[8503]:     at composeNode (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1441:12)
Oct 12 21:44:01 openhabian npm[8503]:     at readDocument (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1625:3)
Oct 12 21:44:01 openhabian npm[8503]:     at loadDocuments (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1688:5)
Oct 12 21:44:01 openhabian npm[8503]:     at Object.load (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1714:19)
Oct 12 21:44:01 openhabian npm[8503]:     at Object.read (/opt/zigbee2mqtt/lib/util/yaml.ts:7:21)
Oct 12 21:44:01 openhabian npm[8503]:     at read (/opt/zigbee2mqtt/lib/util/settings.ts:311:20)
Oct 12 21:44:01 openhabian systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Oct 12 21:44:01 openhabian systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Oct 12 21:44:01 openhabian systemd[1]: zigbee2mqtt.service: Consumed 5.916s CPU time.

Don't know what to do. ONly have minimal knowledge of linux.

EnerVision, 2022-10-12:
...perhaps rebooting will help

EnerVision, 2022-10-12:
OK, found my mistake. double entry of advanced: in the yaml-file

EnerVision, 2022-10-14:
Ok, totally confused now. I just wanted to dive into problem again.
First wanted to see how it was originally coded. So I took the file lidl.js from the zigbee-converter devices again. Restarted Zigbee2MQTT, saw that there was an error in starting the service: Error: ENOSPC: no space left on device, mkdir '/var/log/zigbee2mqtt/2022-10-14.19-47-38al'.
Proceeded with deleting everything under '/var/log/zigbee2mqtt'
Total surprise: some of the energy readings came online! 

At this moment I've got 5 plugs active (koffiezet, BoilerToilet, Wasmachine, Droogkast and Printer). 2 of them states N/A kWh. the other three are displaying a total used energy reading.

Don't really know what wrong. I'll try to delete them an reconnect ... see what happens

EnerVision, 2022-10-14:
Curious, deleted 'Koffiezet', Reconnected the plug. It came online with an energy reading but without Voltage, Power and Curent went dead "N/A".
Restarted zigbee2mqtt service again. Lots of trouble with the 'no space error. In configuration.yaml, I then all commented the lines regarding the logging. 
After the restart I got also the Voltage, Power and Current reading for that plug.

EnerVision, 2022-10-14:
Just wildly guessing. Can it be that in first 'connection' to the plug it checks if some values are present  (and zero is being detected as not available)? And if -at first has no value or zero- it afterwards never checks again??

fluppie, 2022-10-17:
[image]
Also seeing the null N/A kWh

Is this issue related? https://github.com/Koenkk/zigbee-herdsman-converters/pull/4791

blackie333, 2022-10-19:
I have also connected both of the plugs to my Z2MQTT network right after unboxing so your theory may be correct.

blackie333, 2022-10-21:
I have tried to re-pair one of my plugs to the network after some energy was spent but it didn't helped.
Seems that some stronger magic is needed but I didn't understood the procedure how you convinced your plugs to start reporting Energy :) Hopefully you guys will find a solution. Thanks

EnerVision, 2022-10-22:
Koen, Is there somehow a way to trick the program to recognize the plug as a Tuya zigbee plug?
Perhaps that would help because the Tuya app with tuya bridge doesn't have any issues.

Koenkk, 2022-10-22:
I updated https://gist.github.com/Koenkk/81b27922a58bb2922bb9e52cbcf7d685, energy should now be polled every hour.

EnerVision, 2022-10-23:
Koen, 

Am I doing something wrong? I placed the file in /opt/zigbee2mqtt/data and added it to configuration.yaml as 
external_converters:
  - HG08673.js

It gives me following error:

Oct 23 08:39:11 openhabian npm[29066]: ReferenceError: globalStore is not defined
Oct 23 08:39:11 openhabian npm[29066]:     at onEventMeasurementPoll (/opt/zigbee2mqtt/dist/util/externally-loaded.js:19:12)
Oct 23 08:39:11 openhabian npm[29066]:     at Object.onEvent (/opt/zigbee2mqtt/dist/util/externally-loaded.js:59:47)
Oct 23 08:39:11 openhabian npm[29066]:     at OnEvent.callOnEvent (/opt/zigbee2mqtt/lib/extension/onEvent.ts:47:37)
Oct 23 08:39:11 openhabian npm[29066]:     at OnEvent.start (/opt/zigbee2mqtt/lib/extension/onEvent.ts:10:18)
Oct 23 08:39:11 openhabian npm[29066]:     at Controller.callExtensions (/opt/zigbee2mqtt/lib/controller.ts:314:40)
Oct 23 08:39:11 openhabian npm[29066]:     at processTicksAndRejections (node:internal/process/task_queues:96:5)
Oct 23 08:39:11 openhabian npm[29066]:     at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:153:9)
Oct 23 08:39:11 openhabian npm[29066]:     at start (/opt/zigbee2mqtt/index.js:109:5)
Oct 23 08:39:11 openhabian systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Oct 23 08:39:11 openhabian systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.

Koenkk, 2022-10-23:
No I was doing something wrong, updated https://gist.github.com/Koenkk/81b27922a58bb2922bb9e52cbcf7d685 😄

EnerVision, 2022-10-23:
Thanks a lot. Zigbee2MQTT started. And got some readings, I'll do some follow-up the next days.

radaoggy, 2022-10-23:
> I updated https://gist.github.com/Koenkk/81b27922a58bb2922bb9e52cbcf7d685, energy should now be polled every hour.

Good work! I will also test the external converter for a few days to see if it works. One more question, can I change the pool interval from 1 hour to a shorter time? For example 5 minutes? I know where to set it, just wondering if it affects performance or something.

Koenkk, 2022-10-23:
> . One more question, can I change the pool interval from 1 hour to a shorter time? For example 5 minutes? I know where to set it, just wondering if it affects performance or something.

Not with the current converter but I can add this as an option.

radaoggy, 2022-10-23:
Is it not enough to change the value in the external conventor? It would be nice if I could set a custom interval, like "measurement_poll_interval", an hour is quite a long time.

PetrP2B, 2022-10-24:
I managed to unpair and re-pair the outlet with the Z2M during the measurement and it started reporting the energy used, it even remembers it after disconnecting from the mains. I use without modifications Z2M:
Zigbee2MQTT version 1.28.0-dev commit: ceeaaba
Coordinator type zStack12
Coordinator revision 20201127
Frontend version 0.6.114
[image]

jorik41, 2022-11-04:
Any update on this issue ?
I can verify that i get energy value using
 Dev console cluster seMetering 
Attribute currentSummDeliverd 
But the value only updates every time I do a manual read

PetrP2B, 2022-11-04:
Yesterday I was doing some changes on the Z2M network and I noticed that the sockets are sending the power consumption status to the HA. I don't know if there has been any change, I'm using the latest version of Z2M-edge and the sockets got the new FW.

radaoggy, 2022-11-04:
I've been testing it for a few days now and it's working great. Yesterday I tried changing the pool interval from 1 hour to 1 minute (see image) and now it works even better. New data is updating every 1 minute (see graph before - hour, now - minute). I have no idea if this is causing a connection or performance issue, but everything seems fine so far.

But it would be great if the interval could be set directly in the device settings, as I wrote recently.

[image]

[image]

PetrP2B, 2022-11-04:
Is it possible to reset the accumulated energy?

Koenkk, 2022-11-05:
Could you do a final check with https://gist.github.com/Koenkk/81b27922a58bb2922bb9e52cbcf7d685 and z2m 1.28.2? This version should also allow you to set the measurement poll interval. Please double check if current/power/voltage reporting works.

EnerVision, 2022-11-06:
@PetrP2B  

Apparently you can reset the accumulated energy in the dev module by sending to endpoint 1, cluster "0x00", command "0"
[image]

PetrP2B, 2022-11-07:
@EnerVision 
It's working. Thak you :-)