Koenkk/zigbee2mqtt #14361: Inputs for blinds state are being reversed (20 inputs as 80 etc)

Brett-R-Turner, 2022-10-06:
### What happened?

Hi!

I recently got a really nice looking set of Yoolax blinds that show up as Yookee when connected to Z2MQTT (https://www.zigbee2mqtt.io/devices/D10110.html).

The issue is the blinds seem to be executing the opposite of the command that I input. 
If I set the blind state to 90 for example, it will instead move to the 10 position, and vice versa. Its somewhat difficult to explain how exactly this happens, so please see the attached loom video: https://www.loom.com/share/f3cc12e7d2ff4d699adfa23a3144551f.

The Yookee plugin does have the option to "invert" the close/open value which is a great use case, but that does not seem to have any effect on the issue here, still reversed inputs, it just changes how the tx from the blinds reports the status back to the device (have confirmed this on apple homekit).

In short I can still control the blinds, just the inputs and status are always inverted. 
So I can choose to have the blinds report to be closed when they are open, but have my "open blinds" and "close blinds" commands act as I would expect them to

Or I can have my "open" and "close" commands work opposite to how I'd expect , but have the status report properly. Initially I thought the issue was in the TX and how my end platform was interpreting it(z2m & HomeKit), but as I dig here and see the bouncing back of the slider to reflect the opposite value I input it seems the issue may be deeper.

I hope I'm providing enough info here, I'd love to help get this resolved as these blinds are otherwise fanatic. 

TIA for your time :)

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

 20210708

### Adapter

Sonoff 3.0 Stick -> pi 3

### Debug log

Debug Received MQTT message on 'zigbee2mqtt/0x385b44fffe2e1780/set' with data '{"position":90}'
Debug Publishing 'set' 'position' to '0x385b44fffe2e1780'
Info MQTT publish: topic 'zigbee2mqtt/0x385b44fffe2e1780', payload '{"linkquality":102,"position":90}'
Debug Received Zigbee message from '0x385b44fffe2e1780', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":86}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0x385b44fffe2e1780', payload '{"linkquality":102,"position":86}'
Debug Received Zigbee message from '0x385b44fffe2e1780', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":72}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0x385b44fffe2e1780', payload '{"linkquality":96,"position":72}'
Debug Received Zigbee message from '0x385b44fffe2e1780', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":61}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0x385b44fffe2e1780', payload '{"linkquality":66,"position":61}'
Debug Received Zigbee message from '0x385b44fffe2e1780', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":49}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0x385b44fffe2e1780', payload '{"linkquality":99,"position":49}'
Debug Received Zigbee message from '0x385b44fffe2e1780', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":37}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0x385b44fffe2e1780', payload '{"linkquality":57,"position":37}'
Debug Received Zigbee message from '0x385b44fffe2e1780', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":26}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0x385b44fffe2e1780', payload '{"linkquality":99,"position":26}'
Debug Received Zigbee message from '0x385b44fffe2e1780', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":14}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0x385b44fffe2e1780', payload '{"linkquality":93,"position":14}'
Debug Received Zigbee message from '0x385b44fffe2e1780', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":10}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0x385b44fffe2e1780', payload '{"linkquality":63,"position":10}'

Koenkk, 2022-10-06:
When you fully open the cover, is the position 0 or 100?

Brett-R-Turner, 2022-10-06:
> When you fully open the cover, is the position 0 or 100?

Shows as 100 when open in the default state; but again there is a toggle to invert that setting. I currently do have that off though.

Koenkk, 2022-10-07:
That is correct, so if you send `position: 100` the blind fully opens?

closed 2022-11-15 by github-actions[bot]

dejanzelic, 2022-12-11:
@Koenkk I have the same issue as is being reported here.

When I send:
Position: 100 = Blinds closed (shows up as position 0)
Position: 0 = Blinds Open (shows up as position 100)
Position 10 = Mostly open (Shows up as position 90)
Position 90 = Mostly closed (Shows up as position 10)

I see this issue was closed, should I open a new one or should this one be reopened?

FWIW, this is a Graywind motor which reports as the yooksmart ([Yookee](https://www.zigbee2mqtt.io/supported-devices/#v=Yookee)) [D10110](https://www.zigbee2mqtt.io/devices/D10110.html#yookee-d10110)

I have found others complaining about a similar issue in ZHA here:

https://github.com/zigpy/zha-device-handlers/issues/1306

Koenkk, 2022-12-12:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/f541c3ab99f6f567a067a63b9dcafd17
- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

dejanzelic, 2022-12-12:
Thanks for the quick response! I tried it but still the same issue. (There's a syntax error on  line 27 with an extra comma I had to fix) I was able to confirm in the settings that the file was loaded. 

Here is a video of the issue with this converter: https://drive.google.com/file/d/1nYksQYHZD98HtwvocjT4bqdjiMIuGRd-/view?usp=share_link

Koenkk, 2022-12-12:
Can you provide the debug log when setting position 0 and after that position 100

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

Koenkk, 2022-12-12:
I updated https://gist.github.com/Koenkk/f541c3ab99f6f567a067a63b9dcafd17, can you try with both with `coverInverted: true` and `coverInverted: false`?

dejanzelic, 2022-12-12:
That fixed it!!! Specifically `coverInverted: false`

Position 100 - Fully open, reports as open
Position 0 - Fully Closed, reports as closed
The transition position shows up properly as the blinds is moving
The Open Button opens the blinds
The Close Button closes the blinds

Koenkk, 2022-12-13:
Great, pushed the fix.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

dejanzelic, 2022-12-14:
Checked in dev, all working great!

Brett-R-Turner, 2022-12-14:
wooo! I had never solved this issue, I ended up just living with it, wasn't the worst thing. But this is so exciting that its fixed, thanks for reviving this and fixing guys, I appreciate y'all (:

dejanzelic, 2022-12-14:
Same for me, it wasn't super annoying to deal with. But recently I've been using scenes in HA and this issue made the scenes much harder to use.

matti125, 2023-10-22:
It seems like my blinds now work incorrectly, i.e. they report 100 when fully closed, and the position slider in ”zigbee2mqtt - devices- exposes” does not work logically. Open and Close work as they should. I guess the behaviour depends on firmware on the blinds. Anyone know whether firmware is available somewhere? Or could we have a per-device selection of whether the reporting is reversed or not?

night199uk, 2024-02-08:
hey folks .... i was the one that wrote the original quirk that was reverted here. i still have a bunch of these blinds.

older yookee/yooksmart firmwares definitely have a quirk still here, and this broke my blinds. perhaps we need a new optional quirk "yookee_invert: true:false" to enable this specific quirk as required for older firmwares.

i have found no way to upgrade firmware on my yookees - if anyone knows a way let me know.

    D10110_cover_position_tilt: {
        cluster: 'closuresWindowCovering',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.hasOwnProperty('currentPositionLiftPercentage') && msg.data['currentPositionLiftPercentage'] <= 100) {
                // The Yookee D10110 SENDs it's position reversed, relative to the spec.
                msg.data['currentPositionLiftPercentage'] = 100 - msg.data['currentPositionLiftPercentage'];
            }
            return converters.cover_position_tilt.convert(model, msg, publish, options, meta);
        },
    },

night199uk, 2024-02-08:
i was the original author of the yookee quirk. 

i just added some commentary to the zha thread, including the vendor. but the reality is early firmware on these blinds had a bug where sometimes the blinds would be inverted, sometimes not. you can see i even added the hack to z2m, reverted it, and then had to put it back as my own blinds would sometimes be inverted sometimes not.

my blinds currently report as inverted (and have been stable like this for some time) and i just found out after upgrading z2m for the first time in a while the hack was inverted. it is not as simple as just changing coverInverted (at least for my z2m->homebridge setup) as they report inverted but require "set" values in the correct way.

so i used the ext_converters.js logic to re-implement the hack. if you need it until yookee release their new firmware to the public, you can add it back using the ext_converter.js methodology koenkk mentioned above:

https://gist.github.com/night199uk/ab5687585716841cb4e078494d9da402
save this as file next to configuration.yaml as ext_converter.js
add it to configuration.yaml:
external_converters:
  - ext_converter.js
start z2m, check if issue is fixed