Koenkk/zigbee2mqtt #14363: git fetch - no dev branche

JacekKac, 2022-10-06:
### What happened?

on one of my hosts i cant switch to dev branch

pi@ismart-4:/opt/zigbee2mqtt $ sudo systemctl stop zigbee2mqtt
pi@ismart-4:/opt/zigbee2mqtt $ cd /opt/zigbee2mqtt
pi@ismart-4:/opt/zigbee2mqtt $ cp -R data data-backup
pi@ismart-4:/opt/zigbee2mqtt $ git fetch
pi@ismart-4:/opt/zigbee2mqtt $ git checkout dev
error: pathspec 'dev' did not match any file(s) known to git
pi@ismart-4:/opt/zigbee2mqtt $ git branch -a
  a
* master
  remotes/origin/HEAD -> origin/master
  remotes/origin/master
pi@ismart-4:/opt/zigbee2mqtt $

don't know why 

### What did you expect to happen?

switch to dev, update

### How to reproduce it (minimal and precise)

do the Switch to the dev branch

### Zigbee2MQTT version

1.27

### Adapter firmware version

all

### Adapter

all

JacekKac, 2022-10-06:
[in instal](https://www.zigbee2mqtt.io/guide/installation/01_linux.html#installing) 
there is git clone --depth 1 https://github.com/Koenkk/zigbee2mqtt.git /opt/zigbee2mqtt
should be git clone  https://github.com/Koenkk/zigbee2mqtt.git /opt/zigbee2mqtt

JacekKac, 2022-10-06:
not sure right  i clone the whole repository without the --depth 1. i Think the "how to switch to dev branche should be modified if its needed to do git fetch origin dev"

Koenkk, 2022-10-07:
I checked locally and with `git fetch origin dev` it works, updated the docs.

closed 2022-10-07 by Koenkk

ALaDoffe, 2022-10-26:
Hello 

Still an issue for me.
I can not switch to the dev branch 

(zigbee2mqtt) pi@raspberrypi:/srv/zigbee2mqtt $ cp -R data data-backup
(zigbee2mqtt) pi@raspberrypi:/srv/zigbee2mqtt $ git fetch origin dev
`remote: Enumerating objects: 154, done.`
`remote: Counting objects: 100% (154/154), done.`
`remote: Compressing objects: 100% (66/66), done.`
`remote: Total 138 (delta 106), reused 99 (delta 72), pack-reused 0`
`Receiving objects: 100% (138/138), 69.69 KiB | 3.17 MiB/s, done.`
`Resolving deltas: 100% (106/106), completed with 15 local objects.`
`From https://github.com/Koenkk/zigbee2mqtt`
` * branch            dev        -> FETCH_HEAD`
(zigbee2mqtt) pi@raspberrypi:/srv/zigbee2mqtt $ git checkout dev
`error: pathspec 'dev' did not match any file(s) known to git`
(zigbee2mqtt) pi@raspberrypi:/srv/zigbee2mqtt $ git branch -a
`* master`
`  remotes/origin/HEAD -> origin/master`
`  remotes/origin/master`

For information it is a newly Z2M fresh install.

JacekKac, 2022-10-26:
i can confirm that the issue still persist even with the workarround

ALaDoffe, 2022-10-30:
Maybe you can reopen this issue.

ALaDoffe, 2022-11-01:
Resolved by following this link.

https://dev.to/mariayudina/git-errors-cannot-checkout-branch-error-pathspec-branchname-did-not-match-any-files-known-to-git-59kg

JacekKac, 2023-03-05:
@Koenkk  it's still not working can we reopen this ?

Koenkk, 2023-03-06:
@JacekKac what if you do `git checkout origin/dev` ?

mrmaximas, 2023-12-11:
> `git checkout origin/dev`
trying to switch to dev branch.... mac os catalina
max@NVR zigbee2mqtt % git fetch origin dev
From https://github.com/Koenkk/zigbee2mqtt
 * branch              dev        -> FETCH_HEAD
max@NVR zigbee2mqtt % git checkout dev
error: pathspec 'dev' did not match any file(s) known to git
max@NVR zigbee2mqtt % git checkout master
M	data/configuration.yaml
Already on 'master'
Your branch is up to date with 'origin/master'.
max@NVR zigbee2mqtt %