Koenkk/zigbee2mqtt #14369: [New device support]:

Tuttle13, 2022-10-06:
### Link

https://agmelectrical.com.au/home-automation-wifi-intercoms-smart-switches-and-power-points/sparkelec-touch-zigbee-wifi-switches-and-power-points/5-gang-smart-zigbee-touch-glass-wall-switch-off-white-sparkelec-sws5tz-white.html

### Database entry

Model TS0601 IEEE: 0xa4c13883f126fce2 (0xC543)

### Comments

Hi, I can’t work out what to do either the data I have discovered on this device. It’s showing as “unsupported” even though it’s model is stated as a TS0601.
From my hours googling, I can’t find any previous support for this or any similar 5 gang switch. 
I took screenshots of the data produced after creating a logger, first with the 5 switches off, then each of them all turned on, one by one. But I can’t recognise any real patters (maybe the DP value goes to 1-5?).
Have I missed a trick, or have I genuinely got a new and unsupported device?

I’m at a standstill and just have zero idea what to do next. Any help with making this device accessible in HA would be most appreciated! Thanks

Tuttle13, 2022-10-08:
That worked a treat! Tried all 5 individually, and all 5 values triggered the 5 separate lights. 
My next obvious question (I'm sure you saw coming), is how to I combine all 5 of these, and "enable" the device and have all 5 switches available to work with in HA?
(not sure if it will be an issue, but I'm still getting red popup messages saying "datapoint_21/22/23 not available" periodically while this is running?)
Thanks again! I thought i was going to have to give up on the whole 5 switch gang.

Koenkk, 2022-10-08:
@Tuttle13 what are the datapoints for all the swithces?

I'm not sure what the other datapoints mean, to find out check https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#requirements-and-caveats

Tuttle13, 2022-10-08:
So I need to do the process as described in your link with the Tuya app?

All I did was as suggested, I changed the data point from "1", to "2", then to "3", "4","5", and each time I ran that js in HA, it triggered rach of the different lights on and off? Do I need more data than that?

Koenkk, 2022-10-08:
>  I changed the data point from "1", to "2", then to "3", "4","5", and each time I ran that js in HA, it successfully triggered each of the 5 different lights on and off when clicking on/off in expose.

That's all I needed to know.

Please check with the updated https://gist.github.com/Koenkk/159904dbc760446e39091c4b14f10a9a

Tuttle13, 2022-10-08:
Thank you for that - though when I run that code, I'm getting errors for all 5 switches when trying any of them: "Publish set state to lights failed. error: no data point defined for 'state'?
Here's what they look like (including weird formatting inn different ways):
[image]

Tuttle13, 2022-10-09:
Solved: found a similar device you've previously supported and the missing ingredient was:
multiEndpoint: true,

Once I added that to first line of meta, all 5 seperate switches became functional (they were showing correct switch status, but weren't responding prior to this change)

Not sure if this is going the right way about it, but the working code I am now using is:

Koenkk, 2022-10-09:
Well found, added the device!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-10-09 by Koenkk

Tuttle13, 2022-10-09:
thanks again for your assistance with this, and hopefully it helps anyone else looking at these type of plugs here is Aus (or wlsewhere)

Tuttle13, 2022-11-17:
Hi @Koenkk this has been working amazingly! The only left over issue is that I constantly get 3 DataPoint alerts, which I don’t care about, since all the switches are working.
“Datapoint 21/22/23 not defined…” 

It doesn’t impact the gang switch operation, but it’s bloody annoying when using Z2M as it constantly pop’s up, and all 3 have to be cleared repeatedly.

Is there any way to disable these, or add a line to ignore these datapoints? Screenshot attached of them

thanks for your help! (again)
[image]

Koenkk, 2022-11-17:
Done!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)