Koenkk/zigbee2mqtt #14378: Problem Xiaomi ZNCZ04LM

satelitman, 2022-10-06:
### What happened?

Hello - I am asking for your help. I have 24 sockets in my house - Xiaomi ZNCZ04LM. These sockets were in the list of supported devices - ((switch (state), power, energy, device_temperature, voltage, current, consumer_connected, led_disabled_night, power_outage_memory, auto_off, overload_protection, linkquality)). On this basis, I decided to buy them. I gradually bought 24 pieces and gradually placed them in my house. Everything worked perfectly. I can control the outlet (on/off) - I used information about consumption, current total, etc. I try to keep my consumption under control. This information is very important to me. After updating the sockets, all 24 sockets stopped displaying important information - hardly any data is displayed. Please, can we help you somehow. Thank you in advance for any answer and help. I'm sorry for my english.

### Zigbee2MQTT version

1.28.0-1 

### Adapter firmware version

01-23-2022

### Adapter

Xiaomi - ZNCZ04LM

### Debug log

info  2022-09-28 11:44:36: Logging to console and directory: '/config/zigbee2mqtt/log/2022-09-28.11-44-36' filename: log.txt
info  2022-09-28 11:44:36: Starting Zigbee2MQTT version 1.27.2-dev (commit #1e24734)
info  2022-09-28 11:44:36: Starting zigbee-herdsman (0.14.55)
info  2022-09-28 11:44:36: zigbee-herdsman started (resumed)
info  2022-09-28 11:44:36: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210319,"transportrev":2},"type":"zStack3x0"}'
info  2022-09-28 11:44:36: Set transmit power to '20'
info  2022-09-28 11:44:36: Currently 72 devices are joined:
info  2022-09-28 11:44:36: zasuvka Veľká chodba (0x04cf8cdf3c7e3cd7): ZNCZ04LM - Xiaomi Mi power plug ZigBee EU (Router)
info  2022-09-28 11:44:36: zasuvka Spálňa za posteľou (0x04cf8cdf3c8e442e): ZNCZ04LM - Xiaomi Mi power plug ZigBee EU (Router)
info  2022-09-28 11:44:36: zasuvka Spálňa DECO-GOOGLE MAX (0x04cf8cdf3c8e416a): ZNCZ04LM - Xiaomi Mi power plug ZigBee EU (Router)
info  2022-09-28 11:44:36: zasuvka Spálňa televízor LG (0x04cf8cdf3c8e2f49): ZNCZ04LM - Xiaomi Mi power plug ZigBee EU (Router)
info  2022-09-28 11:44:36: zasuvka spalna satelit (0x04cf8cdf3c8e4379): ZNCZ04LM - Xiaomi Mi power plug ZigBee EU (Router)
info  2022-09-28 11:44:36: zasuvka Technická miestnosť KOTOL (0x04cf8cdf3c8a2bf8): ZNCZ04LM - Xiaomi Mi power plug ZigBee EU (Router)
info  2022-09-28 11:44:36: zasuvka Technická miestnosť MRAZNIČKA (0x04cf8cdf3c8a2966): ZNCZ04LM - Xiaomi Mi power plug ZigBee EU (Router)
info  2022-09-28 11:44:36: zasuvka Technická miestnosť PRÁČKA (0x04cf8cdf3c8a2b7c): ZNCZ04LM - Xiaomi Mi power plug ZigBee EU (Router)
info  2022-09-28 11:44:36: zasuvka Technická miestnosť SERVER (0x04cf8cdf3c8a2dd0): ZNCZ04LM - Xiaomi Mi power plug ZigBee EU (Router)
info  2022-09-28 11:44:36: vlhkost Kúpeľňa rodičia (0x00158d000587c866): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor (EndDevice)
info  2022-09-28 11:44:36: pohyb Kúpeľňa rodičia (0x00158d0006c51200): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
info  2022-09-28 11:44:36: dvere Kupelna rodičia (0x00158d0003c7a3b8): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
info  2022-09-28 11:44:36: dvere Spálňa šatník (0x00158d0003c7a3ca): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
info  2022-09-28 11:44:36: pohyb Spálňa komoda smer kúpeľňa (0x00158d00068a00e5): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
info  2022-09-28 11:44:36: pohyb Spálňa posteľ (0x00158d0006a3d0d9): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
info  2022-09-28 11:44:36: dvere Vstup do domu (0x00158d00068a1e81): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
info  2022-09-28 11:44:36: dvere Vstupná chodba (0x00158d00067cf6b3): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
info  2022-09-28 11:44:36: dvere Mala prechodova chodbicka do garáže (0x00158d00068a048a): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
info  2022-09-28 11:44:36: dvere Vstup do garaze mala chodba (0x00158d0003981f17): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
info  2022-09-28 11:44:36: Garáž ĽAVÁ otvorené (0x00158d0007e4c770): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
info  2022-09-28 11:44:36: Garáž PRAVÁ otvorená (0x00158d0007ed12e3): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
info  2022-09-28 11:44:36: Garáž PRAVÁ zatvorené (0x00158d0007ef9926): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
info  2022-09-28 11:44:36: Garáž ĽAVÁ zatvorené (0x00158d0007e61f86): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
info  2022-09-28 11:44:36: zasuvka Kuchyna KaVOVAR (0x04cf8cdf3c759783): ZNCZ04LM - Xiaomi Mi power plug ZigBee EU (Router)
info  2022-09-28 11:44:36: zasuvka Kuchyna GOOGLE-ROBOT (0x04cf8cdf3c8a2d1f): ZNCZ04LM - Xiaomi Mi power plug ZigBee EU (Router)
info  2022-09-28 11:44:36: Kuchyňa chodba 02 (0x00158d00024f5467): 07004D/07005L - Immax Neo SMART LED E27/E14 color, dimmable, Zigbee 3.0 (Router)
info  2022-09-28 11:44:36: Kuchyňa chodba 01 (0x00158d00024f4c14): 07004D/07005L - Immax Neo SMART LED E27/E14 color, dimmable, Zigbee 3.0 (Router)
info  2022-09-28 11:44:36: zasuvka Obývačka PC SERVER (0x04cf8cdf3c8a319a): ZNCZ04LM - Xiaomi Mi power plug ZigBee EU (Router)
info  2022-09-28 11:44:36: zasuvka Obyvacka HIFI CLASSE (0x04cf8cdf3c8a30ba): ZNCZ04LM - Xiaomi Mi power plug ZigBee EU (Router)
info  2022-09-28 11:44:36: zasuvka Obývačka TV LG (0x04cf8cdf3c8a36cb): ZNCZ04LM - Xiaomi Mi power plug ZigBee EU (Router)
info  2022-09-28 11:44:36: zasuvka Kancelaria PC (0x04cf8cdf3c7598ca): ZNCZ04LM - Xiaomi Mi power plug ZigBee EU (Router)
info  2022-09-28 11:44:36: zasuvka kancelaria siet (0x04cf8cdf3c867bcb): ZNCZ04LM - Xiaomi Mi power plug ZigBee EU (Router)
info  2022-09-28 11:44:36: zasuvka chlapci led (0x04cf8cdf3c76414f): ZNCZ04LM - Xiaomi Mi power plug ZigBee EU (Router)
info  2022-09-28 11:44:36: zasuvka chlapci jakub pc (0x04cf8cdf3c7c11b4): ZNCZ04LM - Xiaomi Mi power plug ZigBee EU (Router)
info  2022-09-28 11:44:36: zasuvka Detská izba ANNA LED PÁSIK (0x04cf8cdf3c763a71): ZNCZ04LM - Xiaomi Mi power plug ZigBee EU (Router)
info  2022-09-28 11:44:36: zasuvka Detska izba ANNA PC (0x04cf8cdf3c8a2f31): ZNCZ04LM - Xiaomi Mi power plug ZigBee EU (Router)
info  2022-09-28 11:44:36: Obývačka RGB HIFI (0x00158d00024f57c0): 07004D/07005L - Immax Neo SMART LED E27/E14 color, dimmable, Zigbee 3.0 (Router)
info  2022-09-28 11:44:36: Obývačka RGB knižnica (0x00158d00024f530c): 07004D/07005L - Immax Neo SMART LED E27/E14 color, dimmable, Zigbee 3.0 (Router)
info  2022-09-28 11:44:36: zasuvka obyvacka satelit (0x04cf8cdf3c8e3196): ZNCZ04LM - Xiaomi Mi power plug ZigBee EU (Router)
info  2022-09-28 11:44:36: zasuvka Obývačka video (0x04cf8cdf3c8a38ce): ZNCZ04LM - Xiaomi Mi power plug ZigBee EU (Router)
info  2022-09-28 11:44:36: zasuvka obyvacka netatmo ir (0x04cf8cdf3c8e2a9d): ZNCZ04LM - Xiaomi Mi power plug ZigBee EU (Router)
info  2022-09-28 11:44:36: dvere Kuchyna spajza (0x00158d00067cf5b2): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
info  2022-09-28 11:44:36: lux Hlavný (0x54ef4410000c20dd): GZCGQ01LM - Xiaomi MiJia light intensity sensor (EndDevice)
info  2022-09-28 11:44:36: pohyb Kuchyna linka chladnicka (0x00158d0006b22941): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
info  2022-09-28 11:44:36: pohyb Kuchyna skrinka nad stolom (0x00158d0006a2629d): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
[... 17 lines cut ...]
info  2022-09-28 11:44:36: IKEA obývačka stôl (0xbc33acfffe8f7124): E1524/E1810 - IKEA TRADFRI remote control (EndDevice)
info  2022-09-28 11:44:36: IKEA kuchyňa linka (0xbc33acfffed4503f): E1524/E1810 - IKEA TRADFRI remote control (EndDevice)
info  2022-09-28 11:44:36: IKEA chodba skriňa (0x60a423fffececeb4): E1524/E1810 - IKEA TRADFRI remote control (EndDevice)
info  2022-09-28 11:44:36: IKEA spálňa JANETTE (0x804b50fffeef7a11): E1743 - IKEA TRADFRI ON/OFF switch (EndDevice)
info  2022-09-28 11:44:36: IKEA spálňa MILAN (0x84fd27fffe025174): E1743 - IKEA TRADFRI ON/OFF switch (EndDevice)
info  2022-09-28 11:44:36: IKEA ofiice (0x84fd27fffe025188): E1743 - IKEA TRADFRI ON/OFF switch (EndDevice)
[... cut ...]
error 2022-09-28 11:44:40: MQTT error: Connection refused: Not authorized
error 2022-09-28 11:44:41: MQTT error: Connection refused: Not authorized
error 2022-09-28 11:44:42: MQTT error: Connection refused: Not authorized
error 2022-09-28 11:44:43: MQTT error: Connection refused: Not authorized
error 2022-09-28 11:44:44: MQTT error: Connection refused: Not authorized
error 2022-09-28 11:44:45: MQTT error: Connection refused: Not authorized
error 2022-09-28 11:44:46: MQTT error: Connection refused: Not authorized
info  2022-09-28 11:44:46: Stopped zigbee-herdsman

satelitman, 2022-10-08:
[image]
[image]

satelitman, 2022-10-08:
Please, is it possible to downgrade the ZNCZ04LM socket.
I found an alternative, but I don't know if it's feasible.
I lost information on 24 outlets that were important to me.
Thank you for any help and support - I sent and will definitely send for coffee (Satelitman)
[image]
[image]

Whippie72, 2022-10-09:
I'm having the same issue. I did the update (01-23-2022) to 3 Xiaomi ZNCZ04LM plugs. 2 of them are working pretty well after the update. But one plug does not deliver any values. I only can switch it on/off.

{
    "linkquality": 48,
    "state": "ON",
    "update": {
        "state": "idle"
    }
}

Koenkk, 2022-10-09:
@satelitman you need to update your coordinator to the latest firmware and repair the device, that should solve the issue

satelitman, 2022-10-09:
Thank you very much for your help and guidance - is there any way to upgrade the coordinator??? thank you very much I have a chance thank you thank you

Koenkk, 2022-10-09:
Instructions and firmware for each adapter can be found on this page: https://www.zigbee2mqtt.io/guide/adapters/#recommended

satelitman, 2022-10-09:
it works, thank you very much
[image]

satelitman, 2022-10-09:
thank you very much

satelitman, 2022-10-09:
thank you very much

Koenkk, 2022-10-09:
Great! 🥳

closed 2022-10-09 by Koenkk