Koenkk/zigbee2mqtt #14379: [New device support]: Hue Dimmer Remote 929002398601 (close to '02' but not the same)

johnsturgeon, 2022-10-06:
### Link

https://www.philips-hue.com/en-us/p/hue-dimmer-switch--latest-model-/046677562779

### Database entry

{"id":10,"type":"EndDevice","ieeeAddr":"0x001788010b5985c5","nwkAddr":14252,"manufId":4107,"manufName":"Signify Netherlands B.V.","powerSource":"Battery","modelId":"RWL022","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":2096,"inClusterList":[0,1,3,64512,4096],"outClusterList":[25,0,3,4,6,8,5,4096],"clusters":{"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b7ac71","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b7ac71","endpointID":1},{"cluster":64512,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b7ac71","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b7ac71","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20201125","swBuildId":"2.44.0_hBB3C188","zclVersion":2,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1665088130538,"defaultSendRequestWhen":"immediate"}

### Comments

The device is detected as a different hue dimmer (929002398602) and as such the functionality is indeed quite different.  I'm starting to go through the process to add the device now.

johnsturgeon, 2022-10-06:
Some more info.  It is identified by zigbee2mqtt as:
[image]

Which has the following button layout:
Top (on/off toggle)
Middle two buttons (brightness up - brightness down)
Bottom (scene change)

But the remote buttons are acting like:
https://zigbee.blakadder.com/Philips_RWL020.html

danieledwardgeorgehitchcock, 2022-10-13:
This one might be a difficult one to resolve.

The differences between the two devices may be determined using fingerprinting instead of Zigbee model lookup but, we'd need to see a database.db entry for the other remote to see if there is any difference in the device fingerprints and if it is therefore feasible.

In the meantime, you could copy the sections that apply to you and modify / use them in an external converter.

External converters take precedence over other converters in Z2M startup

danieledwardgeorgehitchcock, 2022-10-13:
Failing the above, the button actions could be 'tweaked' slightly to be more generic for both devices...

johnsturgeon, 2022-10-14:
> In the meantime, you could copy the sections that apply to you from the existing device's code and modify / use it in an external converter.
> 
> External converters take precedence over other converters in Z2M startup

That is exactly what I'm doing, and it's working

closed 2022-11-21 by github-actions[bot]