Koenkk/zigbee2mqtt #14380: Hue dimmer remote (v2) is not defined correctly

johnsturgeon, 2022-10-06:
### What happened?

I purchased the Hue dimmer remove (929002398601) here: https://www.philips-hue.com/en-us/p/hue-dimmer-switch--latest-model-/046677562779

and it was found in the DB as 929002398602, model `RWL022`.  The problem is that while the 'model' seems to be identified correctly, the button layout is completely wrong

Instead of a layout that's similar to:
https://zigbee.blakadder.com/Philips_RWL020.html

It's distinctly different:
[image]

- Top (on/off toggle)
- Middle two buttons (brightness up - brightness down)
- Bottom (scene change)

I filed a new product support ticket #14379 , but when I tried to add the model config .js file things didn't work, I suspect it's because the model is already there.

I'm going to close that issue

### What did you expect to happen?

Top button should toggle light on / off
 - I get discrete 'on'
bottom button should give me a 'next_scene' command (or something), but I get 'discrete off'

### Zigbee2MQTT version

1.28.0-1

### Adapter firmware version

20210708

### Adapter

Sonoff

keachi, 2022-11-07:
What's the status here? I'm affected too.

closed 2022-12-16 by github-actions[bot]

jeeftor, 2023-04-22:
What are you seeing that's not working correctly? I've noticed with the Hue Remotes that they work differently in Z2M vs in the Hue Software.

Specifically with the V1 the "scene change" features seem to be implemented in software