Koenkk/zigbee2mqtt #14382: [New device support]: Eglo 33956 - square ceiling light with color temperature

aalaesar, 2022-10-06:
### Link

https://www.castorama.fr/panneau-led-eglo-60-x-60-cm-4200-lm-ip20/9002759797775_CAFR.prd

### Database entry

{"id":24,"type":"Router","ieeeAddr":"0xa4c1383c58dac55f","nwkAddr":51721,"manufId":4417,"manufName":"AwoX","powerSource":"Mains (single phase)","modelId":"TLSR82xx","epList":[1,3,242],"endpoints":{"1":{"profId":260,"epId":1,"inClusterList":[0,3,4,5,6,8,768,4096,64599],"outClusterList":[6],"clusters":{"genBasic":{"attributes":{"modelId":"TLSR82xx","manufacturerName":"AwoX","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":4751,"epId":3,"inClusterList":[65360,65361],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1664495370579,"defaultSendRequestWhen":"immediate"}

### Comments

A square version of https://www.zigbee2mqtt.io/devices/33955.html with same features  
but recognized as https://www.zigbee2mqtt.io/devices/33951_33948.html in dev latest versions

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TLSR82xx'],
    model: '33956',
    vendor: 'Awox',
    description: 'LED ceiling light with color temperature',
    extend: extend.light_onoff_brightness_colortemp({colorTempRange: [153, 370]}),
    fingerprint: [
      { type: "Router",
        manufacturerName: "AwoX",
        modelID: "TLSR82xx",
        endpoints: [
          {ID: 1, profileID: 260, inputClusters: [0, 3, 4, 5, 6, 8, 768, 4096, 64599], outputClusters: [6]},
          {ID: 3, profileID: 4751, inputClusters: [65360, 65361], outputClusters: [65360, 65361]},
          {ID: 242, profileID: 41440, inputClusters: [], outputClusters: [33]},
        ],
      },
    ],
};

module.exports = definition;
```

### Supported color modes

color temperature

### Color temperature range

{"colorTempPhysicalMin":153, "colorTempPhysicalMax":370}

Koenkk, 2022-10-07:
Without the external converter, is it already recognized by z2m? (with z2m 1.28.0)

aalaesar, 2022-10-08:
Hello,
nope. 😢 
just re-pulled latest-dev (just to be sure), disabled the converter in config then rebooted
```
Zigbee2MQTT:info  2022-10-08 20:12:03: Starting Zigbee2MQTT version 1.28.0-dev (commit #5b39884)
Zigbee2MQTT:info  2022-10-08 20:12:03: Starting zigbee-herdsman (0.14.64)
```
devices are still recognized as https://www.zigbee2mqtt.io/devices/33951_33948.html

regards

Koenkk, 2022-10-09:
The `devId` is missing that is probably the reason it is not detected correctly.

Can you force remove the device through the z2m frontend, restart z2m and pair the device again? The `devId` should now appear in the data/database.db. If not, repeat this with herdsman debug logging activated and provide the logging here.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

aalaesar, 2022-10-12:
Hello
OK thank you for the info.
I'll try it and report the result ASAP
regards

aalaesar, 2022-11-08:
Hello back.
I have'nt yet succeeded to get the devId when peering one of my ceilling ligth with all in debug Z2M rebooted many times 
and of course all converters disabled.
I still get the same line in the database.db, no devId

here are the logs I got with debug enabled for herdsman.

```
Zigbee2MQTT:info  2022-11-08 23:16:43: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1383c58dac55f","ieee_address":"0xa4c1383c58dac55f"},"type":"device_joined"}'
Zigbee2MQTT:info  2022-11-08 23:16:43: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xa4c1383c58dac55f"},"type":"device_connected"}'
2022-11-08T22:16:43.846Z zigbee-herdsman:controller:log Interview '0xa4c1383c58dac55f' start
Zigbee2MQTT:info  2022-11-08 23:16:43: Starting interview of '0xa4c1383c58dac55f'
Zigbee2MQTT:info  2022-11-08 23:16:43: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1383c58dac55f","ieee_address":"0xa4c1383c58dac55f","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2022-11-08 23:16:43: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xa4c1383c58dac55f"},"type":"pairing"}'
2022-11-08T22:16:43.940Z zigbee-herdsman:controller:device:log Interview - start device '0xa4c1383c58dac55f'
2022-11-08T22:16:43.944Z zigbee-herdsman:adapter:ezsp:debg Requesting 'Node Descriptor' for '3475'
2022-11-08T22:16:43.944Z zigbee-herdsman:adapter:ezsp:driv ZDO Node_Desc_req params: {"dstaddr":3475}
2022-11-08T22:16:43.945Z zigbee-herdsman:adapter:ezsp:ezsp ==> setExtendedTimeout: {"remoteEui64":{"_value":{"type":"Buffer","data":[164,193,56,60,88,218,197,95]}},"extendedTimeout":true}
2022-11-08T22:16:43.947Z zigbee-herdsman:adapter:ezsp:uart <-- [3308b1a97d312a7d31b259874a25aa5592499cc3d8757d5ec39874f1c6f084a3bbe5ffd7f51fcb017d38df7e4409a1a9212aea388d7e3c08b1a97d312a7d31b259874a25aa5592499cc3d8757d5ec39874f1c6f084a3bbe5ffd7f51fcb0127d37e4c09a1a9212aeaab207e]
2022-11-08T22:16:43.948Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (3,3,0): 3308b1a9112a11b259874a25aa5592499cc3d8757ec39874f1c6f084a3bbe5ffd7f51fcb0118df7e
2022-11-08T22:16:43.948Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (4)
2022-11-08T22:16:43.948Z zigbee-herdsman:adapter:ezsp:uart --> [8430fc7e]
2022-11-08T22:16:43.949Z zigbee-herdsman:adapter:ezsp:uart Unexpected packet sequence 3 | 4
2022-11-08T22:16:43.949Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 4a9001450004000013000000000000008dffde930dffff0c00930d5fc5da583c38c1a48e
2022-11-08T22:16:43.949Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x45: {"_id_":69,"_cls_":"incomingMessageHandler","_isRequest_":false,"type":4,"apsFrame":{"profileId":0,"clusterId":19,"sourceEndpoint":0,"destinationEndpoint":0,"options":0,"groupId":0,"sequence":141},"lastHopLqi":255,"lastHopRssi":-34,"sender":3475,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[0,147,13,95,197,218,88,60,56,193,164,142]}}
2022-11-08T22:16:43.950Z zigbee-herdsman:adapter:ezsp:debg processMessage: {"messageType":4,"apsFrame":{"profileId":0,"clusterId":19,"sourceEndpoint":0,"destinationEndpoint":0,"options":0,"groupId":0,"sequence":141},"lqi":255,"rssi":-34,"sender":3475,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[0,147,13,95,197,218,88,60,56,193,164,142]}}
2022-11-08T22:16:43.950Z zigbee-herdsman:adapter:ezsp:debg ZDO Device announce: 3475, a4c1383c58dac55f
2022-11-08T22:16:43.950Z zigbee-herdsman:adapter:ezsp:debg Device join request received: 3475 a4c1383c58dac55f
2022-11-08T22:16:43.951Z zigbee-herdsman:controller:log Device '0xa4c1383c58dac55f' joined
2022-11-08T22:16:43.951Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (4,4,0): 4409a1a9212aea388d7e
2022-11-08T22:16:43.952Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (5)
2022-11-08T22:16:43.952Z zigbee-herdsman:adapter:ezsp:uart --> [8520dd7e]
2022-11-08T22:16:43.953Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 4b80017500ff
2022-11-08T22:16:43.953Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x75: {"_id_":117,"_cls_":"findKeyTableEntry","_isRequest_":false,"index":255}
2022-11-08T22:16:43.954Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (3,4,1): 3c08b1a9112a11b259874a25aa5592499cc3d8757ec39874f1c6f084a3bbe5ffd7f51fcb0127d37e
2022-11-08T22:16:43.954Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (4)
2022-11-08T22:16:43.954Z zigbee-herdsman:adapter:ezsp:uart --> [8430fc7e]
2022-11-08T22:16:43.955Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 4a9001450004000013000000000000008dffde930dffff0c00930d5fc5da583c38c1a48e
2022-11-08T22:16:43.955Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x45: {"_id_":69,"_cls_":"incomingMessageHandler","_isRequest_":false,"type":4,"apsFrame":{"profileId":0,"clusterId":19,"sourceEndpoint":0,"destinationEndpoint":0,"options":0,"groupId":0,"sequence":141},"lastHopLqi":255,"lastHopRssi":-34,"sender":3475,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[0,147,13,95,197,218,88,60,56,193,164,142]}}
2022-11-08T22:16:43.956Z zigbee-herdsman:adapter:ezsp:debg processMessage: {"messageType":4,"apsFrame":{"profileId":0,"clusterId":19,"sourceEndpoint":0,"destinationEndpoint":0,"options":0,"groupId":0,"sequence":141},"lqi":255,"rssi":-34,"sender":3475,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[0,147,13,95,197,218,88,60,56,193,164,142]}}
2022-11-08T22:16:43.956Z zigbee-herdsman:adapter:ezsp:debg ZDO Device announce: 3475, a4c1383c58dac55f
[... cut ...]
2022-11-08T22:16:43.959Z zigbee-herdsman:controller:log Not interviewing '0xa4c1383c58dac55f', completed 'false', in progress 'true'
2022-11-08T22:16:43.959Z zigbee-herdsman:controller:log Device '0xa4c1383c58dac55f' accepted by handler
2022-11-08T22:16:43.960Z zigbee-herdsman:controller:log Not interviewing '0xa4c1383c58dac55f', completed 'false', in progress 'true'
2022-11-08T22:16:43.960Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (4) success
```

Is this `3475` the information you are looking for ?

Regards.

Koenkk, 2022-11-09:
The retrieval of the deviceID comes a bit after this logging, before it you will see a line like: `Interview - got active endpoints for device`

miksen, 2022-11-28:
I just filed https://github.com/Koenkk/zigbee2mqtt/issues/15223 for the EGLO 12251 -- it seems like that has the same fingerprint?

aalaesar, 2023-01-01:
Hello there
some update for the bot
I've tried many times to reset and debug my ceiling ligths (I have 6 that can't be reset individually  but I still haven't got  the deviceId to appear.
I'll try a complete reset of the app in the next weeks to get the fix for the EZSP adapter bug.

closed 2023-02-09 by github-actions[bot]