Koenkk/zigbee2mqtt #14384: [Feature request]: Hue device Power recovery

Loic691, 2022-10-07:
### Is your feature request related to a problem? Please describe

no

### Describe the solution you'd like

Is it possible to add in web GUI the aim to configure the power mode behavior ?
As supported in few outdoor devices
https://www.zigbee2mqtt.io/devices/1741530P7.html#power-on-behavior

### Describe alternatives you've considered

No other alternatives. We have actually to use the manual steps to do this with mqtt commands

### Additional context

A simple list to choose the power on behaviour
- LAST STATE
- ON
- OFF

danieledwardgeorgehitchcock, 2022-10-13:
The power on behaviour is exposed in the frontend for converters (most, if not all of the Hue devices) - it is labelled as power_on_behaviour in the devices Exposes tab

lucek8123, 2022-10-15:
Example: Philips Hue Beyond white and color ambiance suspension lamp (7121131PU) or Philips Hue white ambiance (8718696548738)

danieledwardgeorgehitchcock, 2022-10-15:
Both of those are old fixtures and I actually created the converter for the 7121131PU.

I Asked Philips at the time if power on behaviour was going to be supported for the device and they said it wasn't.

I don't think it's received a feature update in quite some time (just security and stability patches)

Not sure on the 8718696548738.

Best thing to do would be to hook them back up to the Hue bridge and see if you can set the power on behaviour via the Hue app.

If you can, please screenshot it for each of the devices you mention above and I'll try and tweak the converters.

Loic691, 2022-10-16:
I'm not sure to understand...
Do you talk about the frontend ? What is the converters ?
Actually ther is a process with mqtt described in z2m documentation for each hue devices which supported this features.
How act the z2m request with the device ?

danieledwardgeorgehitchcock, 2022-10-16:
Converters are what Z2M uses to support devices.

If the power on behaviour isn't supported by the firmware of the device, it won't work no matter how you try to do it in any software.

lucek8123, 2022-10-16:
> Both of those are old fixtures and I actually created the converter for the 7121131PU.
>
>
>
> I Asked Philips at the time if power on behaviour was going to be supported for the device and they said it [...]

Both of these devices support hue power on behavior. This is described in documentation. I checked this too by sending commands to mosquitto. Lights behaved as they should. But there's no toggle for this in frontend, so I had to do it manually.

danieledwardgeorgehitchcock, 2022-10-16:
Could you provide a screenshot of the Exposes tab in the Z2M frontend for your 7121131PU device?

lucek8123, 2022-10-16:
[image]

danieledwardgeorgehitchcock, 2022-10-16:
You have color_temp_startup which is the Hue device power recovery

lucek8123, 2022-10-16:
I have to run `mosquitto_pub -t "zigbee2mqtt/FRIENDLY_NAME/set" -m '{ "hue_power_on_behavior": "recover" }'` to set hue device power recovery (https://www.zigbee2mqtt.io/devices/7121131PU.html#power-on-behavior). Changing `color_temp_startup` does nothing. Controls for `hue_power_on_behavior`, `hue_power_on_color`, etc. doesn't appear in frontend.

closed 2022-11-24 by github-actions[bot]