Koenkk/zigbee2mqtt #14387: Devices unavailable in HA after HA restart

Inrego, 2022-10-07:
### What happened?

After restarting my HA, all z2m devices were unavailable. I used MQTT Explorer to check if the auto discovery topics were there. And they weren't. It is likely due to my MQTT broker having been restarted as well, probably losing retained messages.

In Zigbee2MQTT, the Home Assistant status topic is set to `hass/status` (the default value). But even though I manually sent messages on this topic to offline and back to online, the autodiscovery messages were never repeated.
What is this HA status topic used for?

Is there any way to resend the autodiscovery messages from Z2M without restarting it?
Either by using the HA status topic, or a seperate topic to resend them, and/or a button in the GUI to do it.

I've previously seen other issues with Z2M devices unavailable in HA, but that's been fixed by manually setting the following topic to online: `zigbee2mqtt/bridge/state`

### Zigbee2MQTT version

1.28.0

### Adapter firmware version

0x26680700

### Adapter

ConBee2/RaspBee2

Miguelin96, 2022-10-08:
The same thing happens to me with the same coordinator and with the same version of zigbee2mqqt. Has several disconnections a day all devices.

[image]

Xeevis, 2022-10-23:
Same problem here, also with ConBee II. After HA is restarted all entities are `Unavailable` and never will be on their own. Currently best way is to restart `Zigbee2MQTT` with automation by publishing to `zigbee2mqtt/bridge/request/restart` topic when HA is started. 

Birth message is sent correctly, Zigbee2MQTT responds with status update of all devices, but that doesn't do a thing, it appears HA doesn't remember what to do with it, that entity is no longer there.
```
MQTT publish: topic 'zigbee2mqtt/Patro Obyvák Lampa', payload '{"brightness":255,"color":{"x":0.14912280701754385,"y":0.06140350877192982},"color_mode":"color_temp","color_temp":150,"linkquality":255,"state":"OFF"}'
```
[image]

I've noticed particular entity is recognized by going to Device > Settings > Submit. I'm no expert but it appears Zigbee2MQTT might need to respond with more than just device status update when `homeassistant/status` is set to `online`?

Inrego, 2022-10-23:
Yes, i believe that Z2M should respond with the auto discovery messages as well

Zegorax, 2022-11-06:
I have the same problem currently. I'm trying out Z2M and it does not respond when sending the payload "online" to homeassistant/status. All devices in homeassistant/* are not re-updated and HA will show all devices as unavailable.

mastermc0, 2022-11-13:
I have noticed the same behaviour and latest and latest-dev tags on Docker.

Inrego, 2022-12-15:
Bad bot

ruifung, 2023-01-15:
I've noticed this too right now. In the meantime, I'm setting up my HA instance to poke z2m at zigbee2mqtt/bridge/request/restart to make it resend the things...

Xeevis, 2023-01-15:
My issue was most likely related to RabbitMQ not sending retained messages to wildcard subscribers [#2556](https://github.com/rabbitmq/rabbitmq-server/issues/2556), switching to [Mosquitto broker](https://mosquitto.org/) solved it and now I don't need to needlessly restart Z2M.

ruifung, 2023-01-16:
And in my case, I was using EMQX.

closed 2023-02-24 by github-actions[bot]