Koenkk/zigbee2mqtt #14392: After firmware upgrade Ikea button binds to GroupID 0 and can’t be changed

MyKEms, 2022-10-07:
### What happened?

After firmware upgrade of `IKEA E1743` from `2.2.010` to `2.3.079` button binds to GroupID 0 and can’t be changed. I reproduced same behavior on another button.

I tried to reset device manytimes - rebinded from groups to no-groups + default group but it didn’t help. Only possible solution is to bind button directly to ligh. With this it will start to control light but still reporting group 0.

Could be related issue: https://github.com/Koenkk/zigbee2mqtt/issues/4495

### What did you expect to happen?

When I reconfigure binding it should report group ID I tied device with.

### How to reproduce it (minimal and precise)

After firmware upgrade of `IKEA E1743` from `2.2.010` to `2.3.079` button binds to GroupID 0 and can’t be changed. I reproduced same behavior on another button.

### Zigbee2MQTT version

 1.28.0

### Adapter firmware version

20220219

### Adapter

Slae.sh CC2652RB

### Debug log

```
Debug 2022-10-07 13:57:32Saving state to file /share/zigbee2mqtt/state.json
Info 2022-10-07 13:57:32Device '0x70ac08fffe650d0f' joined
Info 2022-10-07 13:57:32MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x70ac08fffe650d0f","ieee_address":"0x70ac08fffe650d0f"},"type":"device_joined"}'
Info 2022-10-07 13:57:32MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x70ac08fffe650d0f"},"type":"device_connected"}'
Info 2022-10-07 13:57:32MQTT publish: topic 'homeassistant/sensor/0x70ac08fffe650d0f/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x70ac08fffe650d0f"],"manufacturer":"IKEA","model":"TRADFRI ON/OFF switch (E1743)","name":"0x70ac08fffe650d0f","sw_version":"2.3.079"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0x70ac08fffe650d0f","name":"0x70ac08fffe650d0f battery","state_class":"measurement","state_topic":"zigbee2mqtt/0x70ac08fffe650d0f","unique_id":"0x70ac08fffe650d0f_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
Info 2022-10-07 13:57:32MQTT publish: topic 'homeassistant/sensor/0x70ac08fffe650d0f/action/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x70ac08fffe650d0f"],"manufacturer":"IKEA","model":"TRADFRI ON/OFF switch (E1743)","name":"0x70ac08fffe650d0f","sw_version":"2.3.079"},"enabled_by_default":true,"icon":"mdi:gesture-double-tap","json_attributes_topic":"zigbee2mqtt/0x70ac08fffe650d0f","name":"0x70ac08fffe650d0f action","state_topic":"zigbee2mqtt/0x70ac08fffe650d0f","unique_id":"0x70ac08fffe650d0f_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'
Info 2022-10-07 13:57:32MQTT publish: topic 'homeassistant/sensor/0x70ac08fffe650d0f/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x70ac08fffe650d0f"],"manufacturer":"IKEA","model":"TRADFRI ON/OFF switch (E1743)","name":"0x70ac08fffe650d0f","sw_version":"2.3.079"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x70ac08fffe650d0f","name":"0x70ac08fffe650d0f linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x70ac08fffe650d0f","unique_id":"0x70ac08fffe650d0f_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Info 2022-10-07 13:57:32MQTT publish: topic 'homeassistant/sensor/0x70ac08fffe650d0f/click/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x70ac08fffe650d0f"],"manufacturer":"IKEA","model":"TRADFRI ON/OFF switch (E1743)","name":"0x70ac08fffe650d0f","sw_version":"2.3.079"},"icon":"mdi:toggle-switch","json_attributes_topic":"zigbee2mqtt/0x70ac08fffe650d0f","name":"0x70ac08fffe650d0f click","state_topic":"zigbee2mqtt/0x70ac08fffe650d0f","unique_id":"0x70ac08fffe650d0f_click_zigbee2mqtt","value_template":"{{ value_json.click }}"}'
Info 2022-10-07 13:57:32MQTT publish: topic 'homeassistant/sensor/0x70ac08fffe650d0f/update_state/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x70ac08fffe650d0f"],"manufacturer":"IKEA","model":"TRADFRI ON/OFF switch (E1743)","name":"0x70ac08fffe650d0f","sw_version":"2.3.079"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:update","json_attributes_topic":"zigbee2mqtt/0x70ac08fffe650d0f","name":"0x70ac08fffe650d0f update state","state_topic":"zigbee2mqtt/0x70ac08fffe650d0f","unique_id":"0x70ac08fffe650d0f_update_state_zigbee2mqtt","value_template":"{{ value_json['update']['state'] }}"}'
Info 2022-10-07 13:57:32MQTT publish: topic 'homeassistant/binary_sensor/0x70ac08fffe650d0f/update_available/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x70ac08fffe650d0f"],"manufacturer":"IKEA","model":"TRADFRI ON/OFF switch (E1743)","name":"0x70ac08fffe650d0f","sw_version":"2.3.079"},"device_class":"update","enabled_by_default":true,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0x70ac08fffe650d0f","name":"0x70ac08fffe650d0f update available","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0x70ac08fffe650d0f","unique_id":"0x70ac08fffe650d0f_update_available_zigbee2mqtt","value_template":"{{ value_json['update']['state'] == \"available\" }}"}'
Info 2022-10-07 13:57:32Starting interview of '0x70ac08fffe650d0f'
Info 2022-10-07 13:57:32MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x70ac08fffe650d0f","ieee_address":"0x70ac08fffe650d0f","status":"started"},"type":"device_interview"}'
Info 2022-10-07 13:57:32MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x70ac08fffe650d0f"},"type":"pairing"}'
Debug 2022-10-07 13:57:32Received MQTT message on 'homeassistant/sensor/0x70ac08fffe650d0f/battery/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x70ac08fffe650d0f"],"manufacturer":"IKEA","model":"TRADFRI ON/OFF switch (E1743)","name":"0x70ac08fffe650d0f","sw_version":"2.3.079"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0x70ac08fffe650d0f","name":"0x70ac08fffe650d0f battery","state_class":"measurement","state_topic":"zigbee2mqtt/0x70ac08fffe650d0f","unique_id":"0x70ac08fffe650d0f_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
Debug 2022-10-07 13:57:32Received MQTT message on 'homeassistant/sensor/0x70ac08fffe650d0f/action/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x70ac08fffe650d0f"],"manufacturer":"IKEA","model":"TRADFRI ON/OFF switch (E1743)","name":"0x70ac08fffe650d0f","sw_version":"2.3.079"},"enabled_by_default":true,"icon":"mdi:gesture-double-tap","json_attributes_topic":"zigbee2mqtt/0x70ac08fffe650d0f","name":"0x70ac08fffe650d0f action","state_topic":"zigbee2mqtt/0x70ac08fffe650d0f","unique_id":"0x70ac08fffe650d0f_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'
Debug 2022-10-07 13:57:32Received MQTT message on 'homeassistant/sensor/0x70ac08fffe650d0f/linkquality/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x70ac08fffe650d0f"],"manufacturer":"IKEA","model":"TRADFRI ON/OFF switch (E1743)","name":"0x70ac08fffe650d0f","sw_version":"2.3.079"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x70ac08fffe650d0f","name":"0x70ac08fffe650d0f linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x70ac08fffe650d0f","unique_id":"0x70ac08fffe650d0f_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Debug 2022-10-07 13:57:32Received MQTT message on 'homeassistant/sensor/0x70ac08fffe650d0f/click/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x70ac08fffe650d0f"],"manufacturer":"IKEA","model":"TRADFRI ON/OFF switch (E1743)","name":"0x70ac08fffe650d0f","sw_version":"2.3.079"},"icon":"mdi:toggle-switch","json_attributes_topic":"zigbee2mqtt/0x70ac08fffe650 [... cut ...]
Info 2022-10-07 13:57:55MQTT publish: topic 'zigbee2mqtt/0x70ac08fffe650d0f', payload '{"action":null,"battery":87,"click":"","linkquality":116,"update":{"state":null},"update_available":null}'
Info 2022-10-07 13:57:55MQTT publish: topic 'zigbee2mqtt/0x70ac08fffe650d0f/action', payload 'off'
Info 2022-10-07 13:57:55MQTT publish: topic 'zigbee2mqtt/0x70ac08fffe650d0f/click', payload 'off'
Debug 2022-10-07 13:57:56Received Zigbee message from '0x70ac08fffe650d0f', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
```

sjorge, 2022-10-07:
After upgrading the ikea remotes can no longer be bound/unbound from groups and they are bound to devices instead.
I think these are now paired to the coordinator by default after interviewing. (Not sure I, refuse to update mine cause I use groups a lot)

michael-sabrnak-swi, 2022-10-09:
Do you know if there's way how to downgrade firmware via z2m?

MyKEms, 2022-10-09:
Do you know if there's any way how to downgrade firmware via z2m?

sjorge, 2022-10-09:
AFAIK, there is no easy way to downgrade

closed 2022-11-17 by github-actions[bot]

MattWestb, 2022-11-17:
@sjorge If you is getting one OTA upgrade by mistake is all IKEA controllers blocked for firmware downgrade so can being dome with OTA.
But if you is having one WSD probe you can using extracted firmware from the _last good with working groups_ to flashing them to the good ones.
You need little hardware and some free software and many brave Z2M user have doing it.

The firmware and instruction is near the end of this discussion https://github.com/zigpy/zigpy/discussions/660.