Koenkk/zigbee2mqtt #14399: Aqara Smoke Detector JY-GZ-01AQ

elmasryahmed, 2022-10-08:
### What happened?

Linkage alarm is not working, as well temperature value is stuck att 25 Celsius.

### What did you expect to happen?

when a smoke alarm got triggered, alarm siren should start on other linked smoke detectors

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

20220219

### Adapter

SONOFF usb 3.0

Danit2, 2022-10-08:
Please update to the current version.

elmasryahmed, 2022-10-13:
@Danit2 thanks for your reply, updated to the current version as well checked #13548, but can’t figure out how to trigger the alarm if smoke is detected within other sensor.

elmasryahmed, 2022-10-19:
Thanks @Danit2, I'm using HomeKit via Homebridge, unfortunately can't find a way to achieve it, i guess will have to wait for a future update.

closed 2022-11-27 by github-actions[bot]

ciB89, 2023-01-12:
I have created a blueprint in home assistant to achieve this functionality:

https://community.home-assistant.io/t/aqara-smoke-detector-linkage-alarm/517656