Koenkk/zigbee2mqtt #14400: SNZB-03 stopped reporting battery values

elgerg, 2022-10-08:
### What happened?

Since the latest update my SNZB-03 devices have stopped reporting their batter status.

One of my SNZB-03 stopped working a day or so ago with battery value of 54% (2800mv).

Having put a new battery in the value is still reported as  54% (2800mv).

### What did you expect to happen?

I would have expected the battery level to go up to 100% after inserting a new battery.

### How to reproduce it (minimal and precise)

Use a SNZB-03 device with a low battery level and add a new battery.

### Zigbee2MQTT version

1.28.0 commit: 03ba647d

### Adapter firmware version

20220219

### Adapter

I cant remember, where can I find this?

callacomp, 2022-10-12:
I've got one in a spare room, when there's no motion it doesn't send anything and I get offline notifications, not sure if this is similar to yours?

elgerg, 2022-10-24:
The one in question is my landing PIR and I walk past it multiple times a day. Having said that none are now reporting any battery change whatsoever :(

Even after inserting a new battery it's still reporting 54% and thats 2 weeks since the new battery. Something has definitely broken in the last release or 2..

closed 2022-12-01 by github-actions[bot]