Koenkk/zigbee2mqtt #14412: [New device support]: Namron Heater - Model 5401395 (Panelovn)

GMTrevis, 2022-10-09:
### Link

https://www.elektroimportoren.no/namron-panelovn-zigbee-1000w-matt-sort/5401399/Product.html

### Database entry

{"id":6,"type":"Router","ieeeAddr":"0x3fff0004000007ef","nwkAddr":49619,"manufId":4644,"manufName":"NAMRON AS","powerSource":"Mains (single phase)","modelId":"5401395","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[0,3,4,5,9,10,513,516,1794,2820],"outClusterList":[1,3,25],"clusters":{"genBasic":{"attributes":{"modelId":"5401395","manufacturerName":"NAMRON AS","powerSource":1,"zclVersion":3,"appVersion":14,"stackVersion":0,"hwVersion":1,"dateCode":"NULL","swBuildId":"6.9.1.0_r4"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":14,"stackVersion":0,"hwVersion":1,"dateCode":"NULL","swBuildId":"6.9.1.0_r4","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1665302834903,"defaultSendRequestWhen":"immediate"}

### Comments

Hi,
Tried adding this device with different existing converters over and over without luck (probably because im too noob in this area).
It seems like the existing converter (from the library) for the Namron termostat looks at lot like the Namron heater (Model 5401395) now requested.

### External converter

```shell
// #region Non-generic converters
    namron_thermostat: {
        cluster: 'hvacThermostat',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            const data = msg.data;
            if (data.hasOwnProperty(0x1000)) { // Display brightness
                const lookup = {0: 'low', 1: 'mid', 2: 'high'};
                result.lcd_brightness = lookup[data[0x1000]];
            }
            if (data.hasOwnProperty(0x1001)) { // Button vibration level
                const lookup = {0: 'off', 1: 'low', 2: 'high'};
                result.button_vibration_level = lookup[data[0x1001]];
            }
            if (data.hasOwnProperty(0x1002)) { // Floor sensor type
                const lookup = {1: '10k', 2: '15k', 3: '50k', 4: '100k', 5: '12k'};
                result.floor_sensor_type = lookup[data[0x1002]];
            }
            if (data.hasOwnProperty(0x1003)) { // Sensor
                const lookup = {0: 'air', 1: 'floor', 2: 'both'};
                result.sensor = lookup[data[0x1003]];
            }
            if (data.hasOwnProperty(0x1004)) { // PowerUpStatus
                const lookup = {0: 'default', 1: 'last_status'};
                result.powerup_status = lookup[data[0x1004]];
            }
            if (data.hasOwnProperty(0x1005)) { // FloorSensorCalibration
                result.floor_sensor_calibration = precisionRound(data[0x1005], 2) / 10;
            }
            if (data.hasOwnProperty(0x1006)) { // DryTime
                result.dry_time = data[0x1006];
            }
            if (data.hasOwnProperty(0x1007)) { // ModeAfterDry
                const lookup = {0: 'off', 1: 'manual', 2: 'auto', 3: 'away'};
                result.mode_after_dry = lookup[data[0x1007]];
            }
            if (data.hasOwnProperty(0x1008)) { // TemperatureDisplay
                const lookup = {0: 'room', 1: 'floor'};
                result.temperature_display = lookup[data[0x1008]];
            }
            if (data.hasOwnProperty(0x1009)) { // WindowOpenCheck
                result.window_open_check = data[0x1009] / 2;
            }
            if (data.hasOwnProperty(0x100A)) { // Hysterersis
                result.hysterersis = precisionRound(data[0x100A], 2) / 10;
            }
            if (data.hasOwnProperty(0x100B)) { // DisplayAutoOffEnable
                const lookup = {0: 'enabled', 1: 'disabled'};
                result.display_auto_off_enabled = lookup[data[0x100B]];
            }
            if (data.hasOwnProperty(0x2001)) { // AlarmAirTempOverValue
                result.alarm_airtemp_overvalue = data[0x2001];
            }
            if (data.hasOwnProperty(0x2002)) { // Away Mode Set
                result.away_mode = data[0x2002] ? 'ON' : 'OFF';
            }

            return result;
        },
    },
```

Koenkk, 2022-10-09:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-10-09 by Koenkk

JonnaI89, 2022-10-22:
I have to be in dev branch to get this heater to work with this? noob tryn to get this panel heater to work a little better then the normal support new devices from zigbee2mqtt setup.

GMTrevis, 2022-10-22:
> 
Yes, works perfect using the Zigbee2mqtt Edge!
[image]

JonnaI89, 2022-10-22:
> > 
> 
> Yes, works perfect using the Zigbee2mqtt Edge! [image]

Nice! but i cant run edge and normal at the same time?

GMTrevis, 2022-10-22:
@JonnaI89 as far i as iknow you have to choose.
1. At first i made a backup. 
2. Then installed the edge version 
3. Stopped the normal version. 
4. Copied the config from the normal version to the edge version and started the edge version. If i recall correctly the steps i'm pretty sure that's what i did.
Hope it helps. Rememer to disable autostart etc. from the the normal version
[image]

JonnaI89, 2022-10-22:
> @JonnaI89 as far i as iknow you have to choose.
>
> 1. At first i made a backup.
> 2. Then installed the edge version
> 3. Stopped the normal version.
> 4. Copied the config from the normal version to the [...]

Worked like a charm! :) awsome!

JonnaI89, 2022-10-22:
Any fix how to let the heater i had made a converter on for the other Zigbee2MQTT part to join now? It looks like it pairs up on the same spot with the same info as before. I deleted the .js file and factory reset on the heater.. it comes back with the same setup i made.

GMTrevis, 2022-10-24:
@JonnaI89, sure all your previous references to the converter you added were removed?
I removed all references to the converter i used and did not experience any issues.
Double check old converter references (remove if any) and restart addon?

JonnaI89, 2022-10-24:
> @JonnaI89, sure all your previous references to the converter you added were removed? I removed all references to the converter i used and did not experience any issues. Double check old converter references (remove if any) and restart addon?

Yes, from what i can find its all gone. Frustrating!

JonnaI89, 2022-10-24:
Fixed! deleted logs also 👍

auduny, 2022-11-07:
Hm. I have this oven. Thanks for the integration. But it seems like hysteresis er not correct. It says that unit is 0.1 degrees. But in practice it seems to be 1 degrees. And the minimum of 5 degrees is actually 5 degrees, not 0.5 At least that is how my own works now. To turn it off i have to set it to several degrees lower than current.

TomasDJo, 2022-11-30:
I have the same issue as @auduny

5 degrees hysteresis, really messes up the functionality of the oven.

auduny, 2022-12-04:
Unfortunatly it`s not hysteris. There is a "bug" as in the value should be 0.5 instead of 5, but that is only visual. The main problem with the oven is that it is preprogrammed to add 3-5 degrees to measured temparature to compensate for the closeness to the heating element. But the end result is it eventually will be 3-5 degrees warmer than you set it to. This is not a issue with zigbee2mqtt, but with the firmware itself. It is acctually mentioned in the manual.

auduny, 2022-12-05:
Yes it does. I sort of understand why they do it, but the result is that it will eventually be 3 to 8(!) degrees warmer in the room than you set it to.

vodkapmp, 2022-12-05:
> Yes it does. I sort of understand why they do it, but the result is that it will eventually be 3 to 8(!) degrees warmer in the room than what you set it to.

This probably means that if I compensate by setting my target temperature 3 degrees lower.. It'll not turn on again until it is 3.5 degrees lower than my target temperature...
Say I want 17c, so I set it to 14c so that it turns off at 17.5c, it would then not turn back on again until the temp is down all the way to 13.5c again right?

TomasDJo, 2022-12-05:
I think that is incorrect. What you are talking about there is the hysteresis, which is usually 0.5 degrees. The manual specifies that if you set the temperature to 20 degrees, then it will turn on heating at 19.5 and off at 20.5. So I guess setting the oven 3 degrees lower than your actual setpoint is a way of getting around this when in the range of 10-21 degrees celcius. 

Knowing how this oven works is a start at least. I guess one can program this functionality away, either in the z2m logic or in home assistant. A temperature-sensor located a few meters away from the oven would be preferable anyway.

vodkapmp, 2022-12-05:
> I think that is incorrect. What you are talking about there is the hysteresis, which is usually 0.5 degrees. The manual specifies that if you set the temperature to 20 degrees, then it will turn on [...]

I tested, temperature set to 14c, it turns on at 13.5c and doesn't turn off until 17.5c because of the 3c offset + the 0.5c hysteresis.

TomasDJo, 2022-12-05:
I have an oven in the coldest part of the room (below the only window) and I set the temperature to 18 degrees.

(blue = current temperature, orange indicates that the oven is heating, purple is the setpoint)
[image]

Seems to be operating as it should on my part. The oven turned off at about 21.4 degrees, and turns on again at somewhere between 20.8 and 20.3 (the sampling rate in my home assistant graph isn't very high for some reason).

vodkapmp, 2022-12-05:
> I have an oven in the coldest part of the room (below the only window) and I set the temperature to 18 degrees.
>
> (blue = current temperature, orange indicates that the oven is heating, purple is the [...]

Mine is definitely not operating the same, this requires more testing.

To fix your low sampling rate, change the localTemp attribute under the reporting tab for the device in Zigbee2MQTT. I set mine to min rep interval 60 and min rep change to 0.1
My graphs in HA are now updated usually every minute, unless the degree doesn't change more than 0.1c.

TomasDJo, 2022-12-05:
Did you update to the latest z2m (1.28.4)? There is a bugfix there for the hysteresis setting of the oven. 

https://github.com/Koenkk/zigbee-herdsman-converters/pull/4918

And thank you for the advice, here is a graph where the oven turns on and off again. Seems alright to me (except for the fact that I have the oven set to 18 degrees, hehe)

[image]

vodkapmp, 2022-12-05:
> Did you update to the latest z2m (1.28.4)? There is a bugfix there for the hysteresis setting of the oven.
> 
> [Koenkk/zigbee-herdsman-converters#4918](https://github.com/Koenkk/zigbee-herdsman-converters/pull/4918)
> 

Yeah I am on this version. Turns out there was a significant flaw in my first test.
I had opened my window above the heater, it is -10c out, by the time the heater had time to react and turn back on again its measurement had dropped to 13.3c.
My heater works the same as yours, it turns off and on at the set hysteris range, at target temperature +3.

TomasDJo, 2023-02-24:
There is a thread regarding this on https://www.hjemmeautomasjon.no/forums/topic/11043-namron-panelovn-med-zigbee-innebygget-erfaring/ (norwegian home automation forum).

According to responses in this thread Namron are revisioning this oven where they've fixed this issue and added OTA updates.  Seems we are out of luck unless the seller will allow us to replace the ovens when the fix has arrived.

patrikron, 2023-12-28:
Would it be possible to add OTA for this panel heater?

I bought 6 of these, one of the panel heater was delivered with firmware version 6.10.3.0 but the rest had firmware version 6.9.1.0_r4. I noticed that the one with the newest firmware had correct thermostat value, the older one a have to set to 3 degrees lower to get the correct temperature. 

I quess that with Athom Homey it would be possible to OTA upgrade, but I dont wont to buy a Athom Homey just for that..

vodkapmp, 2023-12-31:
> Would it be possible to add OTA for this panel heater?
>
> I bought 6 of these, one of the panel heater was delivered with firmware version 6.10.3.0 but the rest had firmware version 6.9.1.0_r4. I [...]

I recently bought some more units, and wish I could downgrade from 6.10.3.0 to 6.9.1.0_r4 personally.
The 6.9 firmware will happily report temperature every 5 minutes as I have it configured, but the 6.10 firmware will stop reporting any changes lower than +/- 0.5c after 5 minutes.

patrikron, 2024-01-01:
Okay that's a pity. I hadn't noticed that yet. I quess that chaning reporting settings i z2m does not help?
Do you have any more problems than not being able to see the temperature as accurately? Does the thermostat still work as usual?

vodkapmp, 2024-01-19:
> Okay that's a pity. I hadn't noticed that yet. I quess that chaning reporting settings i z2m does not help? Do you have any more problems than not being able to see the temperature as accurately? Does the thermostat still work as usual?

Sorry for the late reply.
Thermostat works as it should, to me it just seems like the reporting goes to "sleep" and has to hit a certain treshold before it wakes up. 
I see a lot of my ovens reporting 19c exactly for large parts of the daytime while my heat pump is running, and then when the temperature of my heat pump is lowered at night, they start reporting due to the colder temperatures making the ovens kick in every now and then, causing larger changes in temperature.