Koenkk/zigbee2mqtt #14415: [New device support]: ZemiSmart Smart Ceiling light Z4E

niceddes, 2022-10-09:
### Link

https://homekitnews.com/2022/10/08/zemismart-smart-ceiling-light-review/

### Database entry

{"id":46,"type":"Router","ieeeAddr":"0xa4c138852a91fcdd","nwkAddr":4799,"manufId":4417,"manufName":"_TZ3210_e020aaaj","powerSource":"Mains (single phase)","modelId":"TS0505B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[3,4,5,6,4096,8,768,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0505B","manufacturerName":"_TZ3210_e020aaaj","powerSource":1,"zclVersion":3,"appVersion":100,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":100,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1665235174053,"defaultSendRequestWhen":"immediate"}

### Comments

Hi, i hope someone can help to add support for this device? Thanks a lot in advance! BR.

### Supported color modes

RGBCW

Koenkk, 2022-10-09:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-10-09 by Koenkk

niceddes, 2022-10-09:
Thank you very much!

niceddes, 2022-10-14:
Sorry, but i have an additional question regarding this issue. 

I am using the iobroker.zigbee adapter. Therefore i added the manufracture name manually in the tuya.js file. Restarted the adapter but it is still not working. Pairing will work and i get the alive status but i cannot change parameters. I hope you can help me or give me a hint to solve this issue for me? 

Thanks a lot in advance....

niceddes, 2022-10-17:
@Koenkk : Can you please help? Thanks a lot in advance & BR

Koenkk, 2022-10-18:
I don't maintain iobroker.zigbee, you should create an issue here: https://github.com/ioBroker/ioBroker.zigbee