Koenkk/zigbee2mqtt #14417: Failed to bind using Sonoff

jiquem, 2022-10-09:
### What happened?

I am unable to bind any device, even to a virtual empty group

Here I tried to bind a Legrand [067773](https://www.zigbee2mqtt.io/devices/067773.html#legrand-067773) to an empty group, but I've got following error

```
error 2022-10-09 23:00:31: Failed to bind cluster 'genOnOff' from 'legrand_switch_cagibi' to 'test' (Error: Bind 0x0004740000c43729/1 genOnOff from '13' failed (AREQ - ZDO - bindRsp after 10000ms))
error 2022-10-09 23:00:41: Failed to bind cluster 'genLevelCtrl' from 'legrand_switch_cagibi' to 'test' (Error: Bind 0x0004740000c43729/1 genLevelCtrl from '13' failed (AREQ - ZDO - bindRsp after 10000ms))
info  2022-10-09 23:00:41: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/bind', payload '{"data":{"clusters":[],"failed":["genOnOff","genLevelCtrl"],"from":"legrand_switch_cagibi/1","to":"test"},"error":"Failed to bind","status":"error","transaction":"ywk6c-1"}'
```

### What did you expect to happen?

To have the device binded

### Zigbee2MQTT version

1.28.0

### Adapter firmware version

20220219

### Adapter

SONOFF ZigBee 3.0 USB Dongle Plus, TI CC2652P

rotilho, 2022-10-13:
Same thing. I can't bind nor unbind. The list of binds also seems wrong (Ikea E1743 is bindto my Aqara SP-EUC01 but I don't see the bind in the UI).

jiquem, 2022-11-05:
Thanks to @celestianx
 Binding iequires to have the device recently announced on the network, otherwise I would see it fail. So when you bind it, you need to have pressed the switch a couple of seconds ago.

closed 2022-11-05 by jiquem