Koenkk/zigbee2mqtt #14418: Xiaomi RTCGQ11LM no longer updates temperature

ky3mr2, 2022-10-09:
### What happened?

After updating past 1.25.2 and the temperature was renamed to device_temperature it is no longer updating the temperature (stuck showing temperature before upgrade)

### What did you expect to happen?

Temperature to change as expected

### Zigbee2MQTT version

1.28.0

### Adapter firmware version

26660700

### Adapter

Conbee II

Koenkk, 2022-10-10:
This has been renamed and thus `temperature` will not update anymore. To remove the stale value stop z2m and remove the value manually from `data/state.json`.

closed 2022-10-10 by Koenkk

ky3mr2, 2022-10-10:
Sorry, I meant temperature as in the value not the key, device_temperature (under exposes\state) does not change value. I've checked the state.json and it had already removed reference to just "temperature", however I renamed the file anyway to generate a new state.json and the values are still not updating. I'm not even sure where its getting the value its chosen from?! as the initial value is not the temperate of the room (I have an analogue thermometer to check, also the sensors were accurate before the update).

Koenkk, 2022-10-10:
These values are not accurate, usually they are just `20` or `25`.

ky3mr2, 2022-10-10:
They seem random, I have 6 of these sensors, one is saying 32 degrees while others are stating 23-25, however my house is actually 20 degrees. When i first set the sensors up (before the update) they gave the correct temperature value, its only since the update the value doesnt change at all, i put one near to my fire and it stayed at 23 degrees, didn't change a single degree.