Koenkk/zigbee2mqtt #14422: Philips Hue White Ambience randomly go offline

amilanov75, 2022-10-10:
### What happened?

After 6-8 weeks of stability, all Philips Hue White Ambience GU10's go offline. Only remedy is to restart zigbee service in Liunx.

Philips bulbs have different firmware e.g. 1.93.11 (latest version) and 1.65.11_hB798F2B older version, however all bulbs are showing offline. https://www.philips-hue.com/en-us/support/release-notes/lamps)

I have an Aqara FP1 that I added to the network recently, after the last issue was seen about 6 weeks ago, and it remains online. Before having the Aqara FP1 on the network, I assume it was a total network failure, but now I can see it's related to the Philips Hue bulbs.

I am not sure what to do next, other than restart the service. 

### How to reproduce it (minimal and precise)

I can't reproduce it. It seems to happen randomly after a period of time, around 6-8 weeks.

### Zigbee2MQTT version

1.25.2 commit: a252914e

### Adapter firmware version

20220302

### Adapter

cc2652P2 USB stick zigbee2mqtt ZigBee Zigstar v4

### Debug log

[Log.txt.txt](https://github.com/Koenkk/zigbee2mqtt/files/9745225/Log.txt.txt)

sygys, 2022-10-11:
I also have philips hue lights go offline form time to time. For me its the normal bulbs with white ambiance. the strange thing is that these bulbs after they go offline are very hard to reset or connect back to the network. they seem to have crashed so hard that even after a hard reset with a dimmer switch and repair it in z2m they keep giving errors. the only fix i found is to force remove the light from z2m. restart z2m then do a hard reset on the light and repair it in z2m. its kind of annoying to have to do this every so many weeks indeed. there doesnt seem to be any reason why these lights fall off the network. they just seem to crash or something.

amilanov75, 2022-10-11:
Interesting, my issue is slightly different. If I just unplug the Z2M stick and plug it back in, it works again. That said, every time I have tried this, I previously restarted the service and rebooted the server, so maybe it requires more than just unplugging and re-plugging. 

The issue also happened with Ikea bulbs, so I tend to think it's a broader problem.

amilanov75, 2022-11-11:
Can anyone else comment on this?

schovanec, 2022-12-09:
I have been having issues similar to this with my Hue motion sensors. They will randomly become unresponsive. When I go to investigate they usually have show a little warning triangle. If you hover over the triangle it says "interview failed" and if I look at the bindings table the coordinator bindings are missing. Clicking reconfigure fails and deleting and re-joining them also seems to not re-create the missing bindings. The only thing that seems to work is to restart Z2M, then delete and re-pair. The reconfigure button also sometimes seems to work immediately after restarting, but it seems to be inconsistent.

RubenKelevra, 2023-01-14:
Does this also happen on 1.29.2? There were some db issues fixed which can lead to Z2M become unresponsive/laggy.

sygys, 2023-01-14:
I experience some strange behaviour with one of my groups it seems that I cannot seem to fix it. And it effectst the stability of all my other 120+ devices in the network. It seems that lights in this group randomly become unresponsive. Before I had 5 innr gu10 spots that dropped off the network or better said got unresponsive randomly. So i thought it was the fault of the innr spots. After I replaced them with hue gu10 spots. But they have the same problem. The funny thing is that when controlled from a group all lights keep responding but individually they don't. So i think this must be a problem with z2mqtt. When one of these spots from the group is not responding the whole network starts to lag. Devices don't post their states or very late. 

I also have a feeling that z2mqtt doesn't reroute the network when things change atleast not until devices are rebooted. I have around 70 hue lights in our home at the moment and nothing works responsive. Lights randomly lag. Our 12 motion. Sensors now and they don't post their states or it takes around 15 seconds for lights to turn on. There are some serious problems with z2mqtt with large networks. It just doesn't work. I would like to invite @Koenkk to visit our home sometime as he also lives in the same city as I do. To see what happens. Maybe we can trouble shoot this mess

amilanov75, 2023-01-15:
I don't recall now which versions I installed, however I am pretty sure I tried the latest at the time plus two others. 
In the end I have bought a new zigbee stick which uses another version of the driver and so far, so good.
In another month or two I will move my other network to the new stick as well.

Koenkk, 2023-01-15:
A firmware which has been confirmed to improve stability for larger networks will be released soon: https://github.com/Koenkk/Z-Stack-firmware/tree/6.10.01.01/coordinator/Z-Stack_3.x.0/bin

closed 2023-02-22 by github-actions[bot]