Koenkk/zigbee2mqtt #14424: Heiman HS2WD-E siren not really compatible ?

marcgtp, 2022-10-10:
### What happened?

The siren device from Heiman (HS2WD-E) doesn't work properly. There was already exchanges on this topic but it seems that the problem is not solved.

The problem is that, when sending MQTT commands like 
'{"warning":{"duration":2,"level":"low","mode":"burglar","strobe":true,"strobe_duty_cycle":1,"strobe_level":"low"}}', all parameters are ignored except:

-   'duration" 

-  'mode' with only 2 possibilities: 'stop'  => siren Off and all other parameters ( burglar, fire ...) lead to siren On without recognition of the type and level.

All relative to strobe is not taken into account.
There is no information coming back via MQTT giving the state of the siren.

The same behavior is observed when using the Front-end and this device doesn't appear on the dashboard.

Is there a way to solve this issue ? 

### What did you expect to happen?

I expect a normal behavior with all parameters taken into account when sending commands.

### How to reproduce it (minimal and precise)

It is the observed behavior when just adding this device in the network.

### Zigbee2MQTT version

1.28.0 

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Debug log

debug 2022-10-10 12:42:15Succesfully pinged 'Si_Heiman_1' (attempt 1/2)
debug 2022-10-10 12:42:31Saving state to file /opt/zigbee2mqtt/data/state.json
debug 2022-10-10 12:44:58Received MQTT message on 'zigbee2mqtt_1/Si_Heiman_1/set' with data '{"warning":{"duration":2,"level":"low","mode":"burglar","strobe":true,"strobe_duty_cycle":1,"strobe_level":"low"}}'
debug 2022-10-10 12:44:58Publishing 'set' 'warning' to 'Si_Heiman_1'
debug 2022-10-10 12:47:31Saving state to file /opt/zigbee2mqtt/data/state.json
debug 2022-10-10 12:50:47Received Zigbee message from 'Le_Tuya_1', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
debug 2022-10-10 12:50:47Succesfully pinged 'Le_Tuya_1' (attempt 1/2)
debug 2022-10-10 12:52:16Received Zigbee message from 'Si_Heiman_1', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
debug 2022-10-10 12:52:16Succesfully pinged 'Si_Heiman_1' (attempt 1/2)
debug 2022-10-10 12:52:31Received Zigbee message from 'Si_Heiman_1', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200}' from endpoint 1 with groupID 0
info 2022-10-10 12:52:31MQTT publish: topic 'zigbee2mqtt_1/Si_Heiman_1', payload '{"battery":100,"linkquality":43}'
debug 2022-10-10 12:52:31Received Zigbee message from 'Si_Heiman_1', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":16}' from endpoint 1 with groupID 0
debug 2022-10-10 12:52:31No converter available for 'HS2WD-E' with cluster 'ssIasZone' and type 'commandStatusChangeNotification' and data '{"extendedstatus":0,"zonestatus":16}'
debug 2022-10-10 12:52:31Saving state to file /opt/zigbee2mqtt/data/state.json
debug 2022-10-10 12:57:31Saving state to file /opt/zigbee2mqtt/data/state.json

marcgtp, 2022-10-21:
In fact, after investigating on this device, the reason for limited control is that this siren doesn't have all functions as described  in the "Zigbee2MQTT Devices information". The only controls are the mode (On-Off), the strobe (On-Off) and the duration.
The warning (composite) is not exposed as it doesn't exist.
The remaining question is: Why nothing is exposed except LinkQuality ?

Maybe the Zigbee2MQTT-Devices information should be updated ?

vincentBesseau, 2022-11-16:
Hi,
How do you turn the alarm on or off using MQTT?
In my case the alarm does not even react
[image]
[image]
[image]

can you explain what happened ?

marcgtp, 2022-11-16:
Hi,

I control the siren by using:

-  either the Front End
   the only valid parameters are **mode** (Off = stop, On = all other choices but with same "melody") and **duration**.

- or by using my own automation system based on Mosquito and Python MQTT Client (Paho MQTT) for zigbee devices.
  in this case the payload is **{"warning": {"duration": 5, "strobe": true,"mode": "emergency"}}**
  (I observe no effect from strobe attribute)

In all the cases there is nothing exposed in return.

marcgtp, 2022-11-17:
Yes, I have the sound (probably corresponding to very high level) during the right duration. However, the strobe doesn't work.

marcgtp, 2022-12-21:
It seems that this topic doesn't generate high interest.
I would propose, at least, to update the z2m  information page in order to make the potential user aware that the functionality is very limited.
Is it acceptable ?
If yes how to do it ?

ASNNetworks, 2023-01-08:
I have this device for a year now. Indeed, nothing is exposed to Home Assistant sadly. However, you can use it fine with mqtt calls through HA automation and Node-RED. You could also create an input_boolean helper that sends the alarm state through mqtt when turned on and send the off state when input_boolean turned off.

marcgtp, 2023-01-10:
This is what I do with a different mqtt client. 
The only parameters taken into account are:
- mode: 
      - all except "stop" leads to siren ON with the same "melody"
      - "stop" leads to siren stop
- duration

All the other parameters are ignored ("level" and "strobe")

Do you observe the same ?

ASNNetworks, 2023-01-10:
> This is what I do with a different mqtt client. The only parameters taken into account are:
> 
> * mode:
>   - all except "stop" leads to siren ON with the same "melody"
>   - "stop" leads to siren stop
> * duration
> 
> All the other parameters are ignored ("level" and "strobe")
> 
> Do you observe the same ?

Yes correct, I basically all it does is either send a fixed alarm sound with a predefined duration, or stop it.

fehgab, 2023-02-04:
With ZHA the duration is working correctly, however I can not change the volume level and only one melody is available. 

device firmware: 2020.8.20 (attribute: date_code)

action:
  - service: zha.warning_device_warn
    data:
      mode: 3
      strobe: 1
      level: 2
      #Siren is on for 1sec.
      duration: 1
      #Strobe pulsing in 50-50%, it can be changed too
      duty_cycle: 50
      intensity: 2
      ieee: your_device_id

luixal, 2023-02-10:
Same problem in my case.

Really annoying as I bought this siren instead of a cheaper one because of z2m support shown in docs.

onedr0p, 2023-03-13:
Not stale.

onedr0p, 2023-04-14:
Not stale.

J-Bot87, 2023-04-15:
Same for me ! Level don't work 🙄

marcgtp, 2023-04-17:
I tried again with the last versions of z2m (1.30.3) and adapter firmware (zStack3x0 - 20221226) and the behaviour is the same.
In summary, this device proposes only On-Off capability (one melody and one level) without exposing any parameters except battery and link quality.
Personally, I am not able, at the moment, to modify converters so, if this topic has quite low interest I think we should modify the documentation by adding a note on this problem, this would help before bying this device.

Vanntag, 2023-04-19:
Hello, i have the same issue. 
Thank you for this informations.
I get this siren because in Z2M documentation, it was possible to set the level sound or melody. 
Hope, i can send it back….

chicknlil, 2023-04-19:
The lack of official response to this issue is disheartening.

Koenkk, 2023-04-20:
My expectation is that the siren does not implement all the features, can anyone test it with the original gateway?

marcgtp, 2023-04-20:
You are right. As I explained in the discussion, after investigation with the seller, this siren has only sound On-Off, strobe On-Off and duration capabilities. The remaining problems are:

- the strobe doesn't work
- nothing is exposed except battery and link quality
- the z2m documentation is misleading (except if there are several versions with same reference ?)

Koenkk, 2023-04-21:
If it indeed supports strobe, then it should be able to be activated through a certain mode, did you play with different paramter values and check what happend? If no combination triggers the strobe someone should sniff the traffic with the original gateway such that we can reverse engineer the command that has to be send.

marcgtp, 2023-04-24:
I tried several combinations and I succeeded to switch On the strobe by adding "strobe_duty_cycle" on the payload.
Instead of 
{"warning": {"duration": 5, "mode": "emergency", "strobe": true}}
the payload should be:
{"warning": {"duration": 5, "mode": "emergency", "strobe": true, "strobe_duty_cycle": 1}}

The duty cycle default value seems to lead to infinite time where the strobe is Off ! 

However, whatever the value for strobe_duty-cycle, the blinking period is 1 second ... but it is better than nothing !
The exposed parameters are still limited to battery and linkquality

cesarvoggithub, 2023-06-01:
not stale

solarmill, 2023-07-06:
still not working.  I made a script to send direct commands, but noticed that since my Z2M converter is not properly addressing it that it doesn't show up properly in MQTT explorer, which made hacking a bit more difficult.  

The below script for home assistant will make the device "chirp."  It sends an on command, wait for 120ms, then send an off command. Since sometimes the messages get missed, there's another brief delay followed by another off. As noted in another thread, the on payload must include strobe_duty_cycle or it will not fire.

```
alias: Heiman Chirp
sequence:
  - service: mqtt.publish
    data:
      retain: false
      payload: >-
        {"warning": {"duration": 1, "mode": "emergency", "level": "low",
        "strobe": true, "strobe_duty_cycle": 1}}
      topic: zigbee2mqtt/0x000d6f0014f82293/set
  - delay:
      hours: 0
      minutes: 0
      seconds: 0
      milliseconds: 150
  - service: mqtt.publish
    data:
      retain: false
      topic: zigbee2mqtt/0x000d6f0014f82293/set
      payload: "{\"warning\": {\"mode\": \"stop\", \"strobe\": false}}"
      qos: "0"
  - delay:
      hours: 0
      minutes: 0
      seconds: 0
      milliseconds: 30
  - service: mqtt.publish
    data:
      retain: false
      topic: zigbee2mqtt/0x000d6f0014f82293/set
      payload: "{\"warning\": {\"mode\": \"stop\", \"strobe\": false}}"
      qos: "0"
```

sesardelaisla, 2023-07-16:
Any way to light the strobe without the siren sound? I would like to do some sort of light “chirp” (blink) but with no sound.

solarmill, 2023-07-19:
Yes, this is trivial to do.  Just change the payload sent in MQTT.

🔊 with _sound_ but no strobe:

`{"warning": {"duration": 1, "mode": "emergency", "level": "low", "strobe": false, "strobe_duty_cycle": 5}}`

🚨with _strobe_ but no sound:

`{"warning": {"duration": 1, "mode": "off", "level": "low", "strobe": true, "strobe_duty_cycle": 5}}`

the variables of _duration_ and _strobe_duty_cycle_ will be useful to adjust the effect. The duty cycle determines amount of time the light is on or off out of a second.  

"10" is on the entire duration. 
"8" is on for 0.8 seconds and off for 0.1s.

"5" will be on for 50% of the time and off for 50% of the time. Extend the duration and you have a steady on/off pulsing light. 

`{"warning": {"duration": 8, "mode": "off", "level": "low", "strobe": true, "strobe_duty_cycle": 5}}`

Shorten the _duration_ to the minimum (1s) and lower the _strobe_duty_cycle_ to "1" and you get a nice blink that is 100ms long. 

You can't make it blink faster or slower, all you can do is adjust the on/off ratio. The frequency is fixed at 1Hz (one cycle per second).

When I use this for an audible chip an entire second is way too much and there's no way to control the duty cycle or level of the siren, so I quickly (150ms) follow up with another payload to turn it off so it doesn't complete the 1s cycle. If you're using it to flash the light, there's less need for that, especially since you can control the duty cycle.  The only reason I can see to send an interrupting off payload is if for some reason you wanted a blink less than 0.1s, but this method of double-tap payloads is not consistent and sometimes the effect is a little longer than expected so I'd keep it simple and just use the single payload with duty_cycle giving consistent results in 100ms increments. 

Overall this little device is *so close* to being useful but disappointing.  The sound levels don't work and it has no other tone or chirp options built in.  It could have been a really useful multi-purpose sound and light effect, from doorbells to alarms, but instead we have to use workarounds to get anything more than a crap-your-pants siren out of it. 

> Any way to light the strobe without the siren sound? I would like to do some sort of light “chirp” (blink) but with no sound.

sesardelaisla, 2023-07-19:
Hey! Thank you so much! Works like a charm. And also thank you very much for your detailed explanation. It is clear what the parameters do.

I do agree that it is a pity that this device has not a better Zigbee implementation. Sound is pretty loud and light is enough for the purpose. However, firmware is not very fine tuned so it is a bit challenging to perform very simple tasks.

fredericf73, 2023-08-06:
> Yes, this is trivial to do. Just change the payload sent in MQTT.
> 
> 🔊 with _sound_ but no strobe:
> 
> `{"warning": {"duration": 1, "mode": "emergency", "level": "low", "strobe": false, "strobe_duty_cycle": 5}}`
> 
> 🚨with _strobe_ but no sound:
> 
> `{"warning": {"duration": 1, "mode": "off", "level": "low", "strobe": true, "strobe_duty_cycle": 5}}`
> 
> the variables of _duration_ and _strobe_duty_cycle_ will be useful to adjust the effect. The duty cycle determines amount of time the light is on or off out of a second.
> 
> "10" is on the entire duration. "8" is on for 0.8 seconds and off for 0.2s.
> 
> "5" will be on for 50% of the time and off for 50% of the time. Extend the duration and you have a steady on/off pulsing light.
> 
> `{"warning": {"duration": 8, "mode": "off", "level": "low", "strobe": true, "strobe_duty_cycle": 5}}`
> 
> Shorten the _duration_ to the minimum (1s) and lower the _strobe_duty_cycle_ to "1" and you get a nice blink that is 100ms long.
> 
> `{"warning": {"duration": 1, "mode": "off", "level": "low", "strobe": true, "strobe_duty_cycle": 1}}`
> 
> You can't make it blink faster or slower, all you can do is adjust the on/off ratio. The frequency is fixed at 1Hz (one cycle per second).
> 
> When I use this device for an audible chip an entire second is way too much and there's no way to control the duty cycle or level of the siren, so I quickly (150ms) follow up with another payload to turn it off so it doesn't complete the otherwise minimum 1s cycle. If you're using it to flash the light, there's less need for that, especially since you can control the duty cycle. The only reason I can see to send an interrupting off payload is if for some reason you wanted a blink less than 0.1s, but this method of double-tap payloads is not consistent and sometimes the effect is a little longer than expected so I'd keep it simple and just use the single payload with duty_cycle giving consistent results in 100ms increments.
> 
> Overall this little device is _so close_ to being useful but disappointing. The sound levels don't work and it has no other tone or chirp options built in. It could have been a really useful multi-purpose sound and light effect, from doorbells to alarms, but instead we have to use workarounds to get anything more than a crap-your-pants siren out of it.
> 
> > Any way to light the strobe without the siren sound? I would like to do some sort of light “chirp” (blink) but with no sound.

I'm trying to do the same thing.

Can you please tell me how can i achieve using zigbee2mqtt?
I use HomeSeer

marcgtp, 2023-09-01:
I think we can close this issue. The problems remain (limited interest for investigations ?) but at least we know better the limits of this device.

**In summary:**

**The typical payload to be sent is:**
{"warning": {"duration": 1, "mode": "emergency", "strobe": true, "strobe_duty_cycle": 5}}

**What we can control:**
- the alarm switching (mode) ON-OFF (ON = all melodies, OFF = stop)
- the alarm duration
- the strobe ON-OFF 
- the strobe duration for a fix period of 1s (typically from 1 to 10: 1 => 0.1s ON and 0.9s OFF, 10 => ON full time)

**What we cannot control:**
- the type of melody
- the sound level
- nothing is exposed (no feed-back)

**In conclusion**
- this device can be used as it is in automation systems but with limitations and specific additional code to take into account the problems
- the z2m documentation should be updated

solarmill, 2023-09-06:
I concur.  Excellent summarization @marcgtp

closed 2023-10-14 by github-actions[bot]

SamuelCarson, 2024-01-25:
not stale

ttrueten, 2024-02-24:
not stale

orcad-fr, 2024-03-06:
not stale

gvillo, 2024-03-13:
this is not stale

ginkel, 2024-03-24:
I also just got the HS2WD-E and hit the limitations documented in this issue. I'd be willing to sniff the traffic with the original gateway (that I would have to order first), but am wondering how I'd figure out the network key. Any ideas?

LMQTLV, 2024-03-30:
Hello
I got 3, none of them are able to work with Z2M :-(
Two are in version "18 | 2018-03-21" and the third is "23 | 2021-02-19".

ginkel, 2024-03-30:
> Hello I got 3, none of them are able to work with Z2M :-( Two are in version "18 | 2018-03-21" and the third is "23 | 2021-02-19".

Did you press the recessed button to start pairing?

LMQTLV, 2024-03-31:
> > Hello I got 3, none of them are able to work with Z2M :-( Two are in version "18 | 2018-03-21" and the third is "23 | 2021-02-19".
> 
> Did you press the recessed button to start pairing?

sorry, I expressed myself badly: they are well paired, but I meant that they do not respond to commands, like the other people on this topic

nimrodolev, 2024-04-13:
Question for the entire audience here - did any of you ever try running the siren for more than 60 seconds and have it work?

I have one and it seems no matter how long I set the `duration`, the siren stops after a minute of sounding the alarm, and I'm wondering if it might be a common issue that people just didn't notice because it's freaking loud and people just turn it off quickly...

Wrap it on a big sweatshirt so it's not crazy loud, and hit it with 3 minutes of alarm sound - does it work for you for more than a minute? I it just mine that's behaving like that?

@ginkel @solarmill, etc.

marcgtp, 2024-04-14:
Up to now I didn't observe such problem but with a value lower than 1mn. I will try.

gvillo, 2024-04-29:
I ran the siren for 300 seconds and it worked correctly (set up in Z2M). I am facing the same lack of features that everyone here. No melody/volume/sensors.

ginkel, 2024-05-06:
I got in touch with Heiman regarding the inability to control sound and volume and they requested a pcap of the Zigbee traffic when activating the alarm. I'll keep you posted...

LouisAugry, 2024-06-13:
Hello ! Any news ?

I have another problem with the Alarm. When I reboot Z2M server the alarm ring for 1s, do you have the same thing ?

J-Bot87, 2024-06-13:
> Hello ! Any news ?
> 
> 
> 
> I have another problem with the Alarm. When I reboot Z2M server the alarm ring for 1s, do you have the same thing ?

Not on my side ! 🙂‍↔️

evanwon, 2024-06-13:
@ginkel Just curious - any update from Heiman? Thanks!

ginkel, 2024-06-13:
They replied and asked for a pcap. Unfortunately, I didn't yet have the time to capture one as the Zigbee pcap solution I had previously working no longer works...

solarmill, 2024-06-13:
Having followed this discussion for almost two years, it's clear this device is not going to be supported any time soon.

Normally this would be off topic, but considering the circumstances, does anyone have alternative recomendations for a zigbee device that serve as a multifunction alarm/chime/sound effect?  I had high hopes this would fit my needs, but they dropped the ball big time.

leonshadow15, 2024-06-18:
I'm joining the discussion as I've been experiencing the exact same limitations. I've also contacted the support via email, let's see if we can push this for a resolution.

Can I help any of you in the meanwhile? I have everything set up and configured if someone could make anything out of the. logs

ginkel, 2024-06-18:
@leonshadow15 Can you capture a Zigbee pcap of the bad interaction? That's what Heiman requested from me...

leonshadow15, 2024-06-18:
> @leonshadow15 Can you capture a Zigbee pcap of the bad interaction? That's what Heiman requested from me...

I have never sniffed the traffic, aren't the logs enough? if yes, can you specify from where I should get them? (mqtt explorer, z2m…)

leonshadow15, 2024-06-20:
> @leonshadow15 Can you capture a Zigbee pcap of the bad interaction? That's what Heiman requested from me...

I found a guide, will look for the 2nd usb stick. So far they haven't replied to me via email.

jnule1a, 2024-07-08:
I just bought and receive the Meian siren (Tuya TS0224 in zigbee2mqtt supported device list). It has the same limitations as described here while using it with zigbee2mqtt.

thogens, 2024-12-08:
HI folks!

Seems here are a few guys having this siren. I have 4 of them since years now and like to share some experiences:
After a while the devices start to get "noisy". - Could be a capacitor or something else.
Soon after it seems the devices get unresponsive (no alarm anymore) and more importantly:
- they still report excellent lqi
- they seem to slow the whole network down
- you can't reset and re-pair them anymore.
The last issue is the hardest one.
For that I found 2 solutions:
1: You wait until the battery is down. (takes a couple of days). - Then re-pair. - works
2: Press the reset button like crazy. (Haven't found a specific number) - After some minutes of pressing over and over again, the device blinks red and seems to completely reset. - Re-pair works again.

Does anybody encounter the same issues? Especially the case that the whole network seems impacted is a bad one.

evanwon, 2024-12-08:
@thogens I've had that buzzing issue several times, I also assumed it was a capacitor issue or something, however it was not correlated to any functional issues with the device like you described. When it started buzzing, I was able to unplug it and plug it back in and that generally resolve the issue. 

I have however had the connectivity issue you described one time, it was unrelated to the buzzing issue, and in that case I did the technique you described where I had to spam the reset button like crazy until it finally somehow started working again. It only happened to be one time, and hasn't happened since. But it was pretty annoying in that all the devices that were routing through that alarm were temporarily disconnected from the network.

thogens, 2024-12-09:
@evanwon you could be right, that there is no correlation between the 2 issues. I somehow "felt" that after deleting the noisy devices, the network worked better. But that could very well coincidence. 
Maybe I should try to re-add the unplugged devices...

kgara, 2025-01-20:
> I just bought and receive the Meian siren (Tuya TS0224 in zigbee2mqtt supported device list). It has the same limitations as described here while using it with zigbee2mqtt.

Confirm same case with Tuya TS0224 both in zha and z2m configurations.

gvillo, 2025-02-10:
is anyone experiencing this device to be offline running 2.1.1? I've tried to pair it, and it's not working :/