Koenkk/zigbee2mqtt #14425: Legrand netatmo Valena Next roller shutter reports 'OPEN" when closed and vice versa

stefanuytterhoeven, 2022-10-10:
### What happened?

Legrand netatmo Valena Next roller shutter reports 'OPEN" when closed and vice versa
invert_cover : true or false, the STATE of the shutter roller is always the wrong state.
invert_cover only related to the percentage the roller shutter is opened.
Very annoying as lot of checks are done on the state of the roller shutter and not on the percentage.
Maybe add possibility to invert the state too?

### What did you expect to happen?

When the roller shutter is opened, the state will be "OPEN" and vice versa.

### How to reproduce it (minimal and precise)

Simple to reproduce with this device: the state is always wrong

### Zigbee2MQTT version

 1.28.0-1

### Adapter firmware version

 20220219

### Adapter

ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus

Koenkk, 2022-10-10:
What is the model number of this device?

stefanuytterhoeven, 2022-10-10:
This is the info Z2m gives
[Z2M@8sdkbu5wqu2v3z36c92qbeegf1r26v4s.ui.nabu.casa](https://8sdkbu5wqu2v3z36c92qbeegf1r26v4s.ui.nabu.casa/api/hassio_ingress/JJfYPB2CRDgztK9Wq5jexmDYKY4F7CbKhwRYTPz-uzg/#/)

Woonkamer Rolluik
[Over](https://8sdkbu5wqu2v3z36c92qbeegf1r26v4s.ui.nabu.casa/api/hassio_ingress/JJfYPB2CRDgztK9Wq5jexmDYKY4F7CbKhwRYTPz-uzg/#/device/0x0004740000a167b4/info)
[Toont](https://8sdkbu5wqu2v3z36c92qbeegf1r26v4s.ui.nabu.casa/api/hassio_ingress/JJfYPB2CRDgztK9Wq5jexmDYKY4F7CbKhwRYTPz-uzg/#/device/0x0004740000a167b4/exposes)
[Koppelen](https://8sdkbu5wqu2v3z36c92qbeegf1r26v4s.ui.nabu.casa/api/hassio_ingress/JJfYPB2CRDgztK9Wq5jexmDYKY4F7CbKhwRYTPz-uzg/#/device/0x0004740000a167b4/bind)
[Rapportage](https://8sdkbu5wqu2v3z36c92qbeegf1r26v4s.ui.nabu.casa/api/hassio_ingress/JJfYPB2CRDgztK9Wq5jexmDYKY4F7CbKhwRYTPz-uzg/#/device/0x0004740000a167b4/reporting)
[Instellingen](https://8sdkbu5wqu2v3z36c92qbeegf1r26v4s.ui.nabu.casa/api/hassio_ingress/JJfYPB2CRDgztK9Wq5jexmDYKY4F7CbKhwRYTPz-uzg/#/device/0x0004740000a167b4/settings)
[Instellingen (specifiek)](https://8sdkbu5wqu2v3z36c92qbeegf1r26v4s.ui.nabu.casa/api/hassio_ingress/JJfYPB2CRDgztK9Wq5jexmDYKY4F7CbKhwRYTPz-uzg/#/device/0x0004740000a167b4/settings-specific)
[Status](https://8sdkbu5wqu2v3z36c92qbeegf1r26v4s.ui.nabu.casa/api/hassio_ingress/JJfYPB2CRDgztK9Wq5jexmDYKY4F7CbKhwRYTPz-uzg/#/device/0x0004740000a167b4/state)
[Clusters](https://8sdkbu5wqu2v3z36c92qbeegf1r26v4s.ui.nabu.casa/api/hassio_ingress/JJfYPB2CRDgztK9Wq5jexmDYKY4F7CbKhwRYTPz-uzg/#/device/0x0004740000a167b4/clusters)
[Scène](https://8sdkbu5wqu2v3z36c92qbeegf1r26v4s.ui.nabu.casa/api/hassio_ingress/JJfYPB2CRDgztK9Wq5jexmDYKY4F7CbKhwRYTPz-uzg/#/device/0x0004740000a167b4/scene)
[Ontwikkelaarsconsole](https://8sdkbu5wqu2v3z36c92qbeegf1r26v4s.ui.nabu.casa/api/hassio_ingress/JJfYPB2CRDgztK9Wq5jexmDYKY4F7CbKhwRYTPz-uzg/#/device/0x0004740000a167b4/dev-console)

Eenvoudige naam
Woonkamer Rolluik

Beschikbaar Online
Apparaat type Router
Zigbee model Shutter switch with neutral
Zigbee Fabrikant Legrand
Omschrijving Netatmo wired shutter switch
Ondersteuningsstatus Ondersteund

Firmware build date 20191105
Firmwareversie 001a
Fabrikant Legrand 
Model 06777

Tonio16, 2022-10-11:
Are you sur of your wiring?

Antoine

stefanuytterhoeven, 2022-10-11:
Yes. Tested it with another member here.

stefanuytterhoeven, 2022-10-17:
067776 ! 
My mistake!

stefanuytterhoeven, 2022-11-02:
I just installed version: 1.28.1-1 of Z2M
Restarted HA en restarted Z2M addon. Result: Problem solved! 
Thanks !

closed 2022-11-02 by Koenkk