Koenkk/zigbee2mqtt #14427: Namron 4512737/4512738 - dry mode missing

ctvedt, 2022-10-10:
### What happened?

Dry mode is missing for Namron 4512737/4512738. From dev console it is possible to activate it, but mode is not exposed to  GUI (or Home Assistant).

### What did you expect to happen?

Be able to activate dry mode from GUI (or Home Assistant)

### How to reproduce it (minimal and precise)

Open dev console, set systemMode to 8 for cluster hvacThermostat.

[image]

### Zigbee2MQTT version

1.28.0

### Adapter firmware version

20210708

### Adapter

ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus

owangen, 2022-11-03:
@ctvedt I have this issue on my radar now and will implement support for it when time permits.

owangen, 2022-11-22:
This is fixed with mentioned PR and this issue should be closed.

closed 2022-11-22 by ctvedt