Koenkk/zigbee2mqtt #14429: Develco (Frient) AQSZB-110 is detected as Develco HMSZB-110

probus, 2022-10-10:
### What happened?

A Frient (Develco) AQSZB-110 is identified as Develco HMSZB-110, and is therefore missing the air quality measurement. Also, for some reason the device address is different from the sticker placed on the back of the device (0x0015bc00360015f1 vs 0x0015bc0035000ee7) if that makes a difference.

### Zigbee2MQTT version

1.28.0

### Adapter firmware version

20220219

### Adapter

ZigStar LilyZig

### Debug log

debug 2022-10-10 15:45:35: Received Zigbee message from '0x0015bc0035000ee7', type 'read', cluster 'genTime', data '["time"]' from endpoint 38 with groupID 0
debug 2022-10-10 15:45:36: Received Zigbee message from '0x0015bc0035000ee7', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 38 with groupID 0
debug 2022-10-10 15:45:36: Received Zigbee message from '0x0015bc0035000ee7', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 38 with groupID 0
debug 2022-10-10 15:45:36: Received Zigbee message from '0x0015bc0035000ee7', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 38 with groupID 0
debug 2022-10-10 15:45:37: Received Zigbee message from '0x0015bc0035000ee7', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 38 with groupID 0
debug 2022-10-10 15:45:37: Received Zigbee message from '0x0015bc0035000ee7', type 'readResponse', cluster 'genPollCtrl', data '{"checkinInterval":14400}' from endpoint 38 with groupID 0
info  2022-10-10 15:45:37: Successfully interviewed '0x0015bc0035000ee7', device has successfully been paired
info  2022-10-10 15:45:37: Device '0x0015bc0035000ee7' is supported, identified as: Develco Temperature & humidity sensor (HMSZB-110)
info  2022-10-10 15:45:37: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Temperature & humidity sensor","exposes":[{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Indicates if the battery of this device is almost empty","name":"battery_low","property":"battery_low","type":"binary","value_off":false,"value_on":true},{"access":1,"description":"Measured temperature value","name":"temperature","property":"temperature","type":"numeric","unit":"°C"},{"access":1,"description":"Measured relative humidity","name":"humidity","property":"humidity","type":"numeric","unit":"%"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"HMSZB-110","options":[{"access":2,"description":"Number of digits after decimal point for temperature, takes into effect on next report of device.","name":"temperature_precision","property":"temperature_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the temperature value (absolute offset), takes into effect on next report of device.","name":"temperature_calibration","property":"temperature_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for humidity, takes into effect on next report of device.","name":"humidity_precision","property":"humidity_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the humidity value (absolute offset), takes into effect on next report of device.","name":"humidity_calibration","property":"humidity_calibration","type":"numeric"}],"supports_ota":false,"vendor":"Develco"},"friendly_name":"0x0015bc0035000ee7","ieee_address":"0x0015bc0035000ee7","status":"successful","supported":true},"type":"device_interview"}'
info  2022-10-10 15:45:37: Configuring '0x0015bc0035000ee7'
debug 2022-10-10 15:45:39: Received Zigbee message from '0x0015bc0035000ee7', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2510,"tolerance":50}' from endpoint 38 with groupID 0
debug 2022-10-10 15:45:39: Received Zigbee message from '0x0015bc0035000ee7', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":5100,"tolerance":350}' from endpoint 38 with groupID 0
info  2022-10-10 15:45:39: MQTT publish: topic 'zigbee2mqtt/0x0015bc0035000ee7', payload '{"battery":null,"battery_low":null,"humidity":53,"linkquality":131,"temperature":25.1}'
info  2022-10-10 15:45:39: MQTT publish: topic 'zigbee2mqtt/0x0015bc0035000ee7', payload '{"battery":null,"battery_low":null,"humidity":51,"linkquality":131,"temperature":25.1}'
debug 2022-10-10 15:45:39: Received Zigbee message from '0x0015bc0035000ee7', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryAlarmState":0,"batteryVoltage":30}' from endpoint 38 with groupID 0
info  2022-10-10 15:45:39: MQTT publish: topic 'zigbee2mqtt/0x0015bc0035000ee7', payload '{"battery":71,"battery_low":false,"humidity":51,"linkquality":120,"temperature":25.1,"voltage":3000}'
debug 2022-10-10 15:45:41: Received Zigbee message from '0x0015bc0035000ee7', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2510,"tolerance":50}' from endpoint 38 with groupID 0
info  2022-10-10 15:45:41: MQTT publish: topic 'zigbee2mqtt/0x0015bc0035000ee7', payload '{"battery":71,"battery_low":false,"humidity":51,"linkquality":167,"temperature":25.1,"voltage":3000}'
debug 2022-10-10 15:45:41: Received Zigbee message from '0x0015bc0035000ee7', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":5100,"tolerance":350}' from endpoint 38 with groupID 0
info  2022-10-10 15:45:41: MQTT publish: topic 'zigbee2mqtt/0x0015bc0035000ee7', payload '{"battery":71,"battery_low":false,"humidity":51,"linkquality":167,"temperature":25.1,"voltage":3000}'
debug 2022-10-10 15:45:41: Received Zigbee message from '0x0015bc0035000ee7', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryAlarmState":0,"batteryVoltage":30}' from endpoint 38 with groupID 0
info  2022-10-10 15:45:41: MQTT publish: topic 'zigbee2mqtt/0x0015bc0035000ee7', payload '{"battery":71,"battery_low":false,"humidity":51,"linkquality":171,"temperature":25.1,"voltage":3000}'
debug 2022-10-10 15:45:41: Received Zigbee message from '0x0015bc0035000ee7', type 'readResponse', cluster 'genPowerCfg', data '{"batteryVoltage":30}' from endpoint 38 with groupID 0
info  2022-10-10 15:45:41: MQTT publish: topic 'zigbee2mqtt/0x0015bc0035000ee7', payload '{"battery":71,"battery_low":false,"humidity":51,"linkquality":174,"temperature":25.1,"voltage":3000}'
info  2022-10-10 15:45:41: Successfully configured '0x0015bc0035000ee7'

sjorge, 2022-10-11:
> Also, for some reason the device address is different from the sticker placed on the back of the device

Do you have the database entry for the device? Since the sticker doesn match, I wonder if the device somehow got a wrong stucker and is actually a HMSZB-110 instead :s

probus, 2022-10-11:
> Do you have the database entry for the device? Since the sticker doesn match, I wonder if the device somehow got a wrong stucker and is actually a HMSZB-110 instead :s

How can I get the database entry? That's what I'm starting to suspect as well, but since they look the same it's difficult to say. There are actually two stickers, one inside the enclosure, both with the same (wrong) address. Any other way to verify?

sjorge, 2022-10-11:
> How can I get the database entry? That's what I'm starting to suspect as well, but since they look the same it's difficult to say. There are actually two stickers, one inside the enclosure, both [...]

The database entry would tell us what the device firmware itself reported as modelId, assuming there is not some weird broken firmware on it, should match with the actual features of the device. You can grab it by looking for the device id (e.g. 0x0015bc00360015f1) in `data/database.json`.

I'm not sure how the HMSZB-110 looks onder the cover, but my AQSZB-110 all look like this:
[image]

probus, 2022-10-12:
Thanks for the help. Here's the database entry and a photo of the device (note the wrong address here as well). If I'm reading this right, it's reporting itself as HMSZB-110. I'll open a discussion with the seller.
```
{
    "id": 32,
    "type": "EndDevice",
    "ieeeAddr": "0x0015bc0035000ee7",
    "nwkAddr": 57821,
    "manufId": 4117,
    "manufName": "frient A/S",
    "powerSource": "Battery",
    "modelId": "HMSZB-110",
    "epList":
    [
        1,
        38
    ],
    "endpoints":
    {
        "1":
        {
            "profId": 49353,
            "epId": 1,
            "devId": 1,
            "inClusterList":
            [
                3,
                5,
                6
            ],
            "outClusterList": [],
            "clusters": {},
            "binds": [],
            "configuredReportings": [],
            "meta": {}
        },
        "38":
        {
            "profId": 260,
            "epId": 38,
            "devId": 770,
            "inClusterList":
            [
                0,
                1,
                3,
                32,
                1026,
                1029
            ],
            "outClusterList":
            [
                3,
                10,
                25
            ],
            "clusters":
            {
                "msTemperatureMeasurement":
                {
                    "attributes":
                    {
                        "measuredValue": 2220,
                        "tolerance": 50
                    }
                },
                "msRelativeHumidity":
                {
                    "attributes":
                    {
                        "measuredValue": 5100,
                        "tolerance": 350
                    }
                },
                "genBasic":
                {
                    "attributes": {}
                },
                "genPollCtrl":
                {
                    "attributes":
                    {
                        "checkinInterval": 14400
                    }
                },
                "genPowerCfg":
                {
                    "attributes":
                    {
                        "batteryVoltage": 30,
                        "batteryAlarmState": 0
                    }
                }
            },
            "binds":
            [
                {
                    "cluster": 32,
                    "type": "endpoint",
                    "deviceIeeeAddress": "0x00124b00258c52f3",
                    "endpointID": 1
                },
                {
                    "cluster": 1026,
                    "type": "endpoint",
                    "deviceIeeeAddress": "0x00124b00258c52f3",
                    "endpointID": 1
                },
                {
                    "cluster": 1029,
                    "type": "endpoint",
                    "deviceIeeeAddress": "0x00124b00258c52f3",
                    "endpointID": 1
                },
                {
                    "cluster": 1,
                    "type": "endpoint",
                    "deviceIeeeAddress": "0x00124b00258c52f3",
                    "endpointID": 1
                }
            ],
            "configuredReportings":
            [
                {
                    "cluster": 1026,
                    "attrId": 0,
                    "minRepIntval": 60,
                    "maxRepIntval": 600,
                    "repChange": 10
                },
                {
                    "cluster": 1029,
                    "attrId": 0,
                    "minRepIntval": 60,
                    "maxRepIntval": 600,
                    "repChange": 300
                },
                {
                    "cluster": 1,
                    "attrId": 32,
                    "minRepIntval": 3600,
                    "maxRepIntval": 62000,
                    "repChange": 0
                }
            ],
            "meta": {}
        }
    },
    "dateCode": "2021-09-14 09:43",
    "zclVersion": 1,
    "interviewCompleted": true,
    "meta":
    {
        "configured": 984527238
    },
    "lastSeen": 1665570580904,
    "defaultSendRequestWhen": "active"
}
```
[image]

sjorge, 2022-10-12:
According to the db it is indeed reporting itself as a HMSZB-110, it also looks different than the AQSZB’s I have. 

I’m guessing this somehow got misslabeled

closed 2022-11-20 by github-actions[bot]