Koenkk/zigbee2mqtt #14432: [Feature request]: Please add exposes.presets.switch_type_2() in SONOFF ZBMINI-L

mced76, 2022-10-10:
### Is your feature request related to a problem? Please describe

Switch type is note defined in SONOFF ZBMINI-L. You cannot use push button

### Describe the solution you'd like

add exposes.presets.switch_type_2()

### Describe alternatives you've considered

non alternative

### Additional context

Can not use several button with this toggle swith.
https://www.domotique-store.fr/domotique/modules-domotiques/modules-encastrables/micromodules-switch-domotique-sans-fil/1681-sonoff-zbmini-l-module-zigbee-pour-interrupteur-sans-neutre-compatible-ewelink.html

molusk, 2022-10-11:
@mced76 where did you find this solution ?
I have a ZBMINI-L in rocker switch mode and the only "switch_type" I have found is a read-only attribute
`info 2022-10-11 12:47:24Read result of 'genOnOffSwitchCfg': {"switchType":0}`

Anyway, there is a guide to support new device (or extend functions of existing) by adding external converter to your configuration. https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#instructions
The file you can play with to begin is sonoff.js in herdsman-converters project : https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/

And you should open this issue in that project as I think there is no need to change anything in zigbee2mqtt to add this feature.

mced76, 2022-10-11:
> @mced76 as a workaround, you can press the device button three times to switch from rocker to push button mode.

Yes it works fine ! thanks a lot !

closed 2022-11-19 by github-actions[bot]