Koenkk/zigbee2mqtt #14439: Livarno smart LED ceiling light not switching color temperature

nekocentral, 2022-10-11:
### What happened?

The color temperature of the Livarno smart LED ceiling light does not respond to the color temperature slider in Z2MQTT and HA

### What did you expect to happen?

Color temperature changes to what the slider is set to like 370 mired

### How to reproduce it (minimal and precise)

Connect a Livarno smart LED ceiling light your coordinator.
After the interview try to set color temperature 

### Zigbee2MQTT version

1.28.0

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Debug log

```
Zigbee2MQTT:info  2022-10-11 14:25:04: MQTT publish: topic 'zigbee2mqtt/Living room Light ', payload '{"brightness":33,"color":{"x":0.3131,"y":0.3232},"color_mode":"color_temp","color_temp":153,"linkquality":116,"state":"ON"}'
2022-10-11T12:25:04.757Z zigbee-herdsman:controller:endpoint Command 0x2c1165fffe6d682a/1 lightingColorCtrl.moveToColorTemp({"colortemp":153,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2022-10-11T12:25:04.757Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x2c1165fffe6d682a:48561/1 (0,0,1)
2022-10-11T12:25:04.759Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":48561,"destendpoint":1,"srcendpoint":1,"clusterid":768,"transid":5,"options":0,"radius":30,"len":7,"data":{"type":"Buffer","data":[1,6,10,153,0,0,0]}}
2022-10-11T12:25:04.760Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,17,36,1,177,189,1,1,0,3,5,0,30,7,1,6,10,153,0,0,0,179]
2022-10-11T12:25:04.773Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2022-10-11T12:25:04.773Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2022-10-11T12:25:04.774Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2022-10-11T12:25:04.774Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2022-10-11T12:25:04.774Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-10-11T12:25:04.777Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,5,195]
2022-10-11T12:25:04.780Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,5,195]
2022-10-11T12:25:04.780Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,5] - 195
2022-10-11T12:25:04.781Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":5}
2022-10-11T12:25:04.781Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-10-11T12:25:04.800Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,25,68,129,0,0,0,3,177,189,1,1,0,120,0,222,88,43,0,0,5,8,6,11,10,0,177,189,29,29]
2022-10-11T12:25:04.800Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,25,68,129,0,0,0,3,177,189,1,1,0,120,0,222,88,43,0,0,5,8,6,11,10,0,177,189,29,29]
2022-10-11T12:25:04.801Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 25 - 2 - 4 - 129 - [0,0,0,3,177,189,1,1,0,120,0,222,88,43,0,0,5,8,6,11,10,0,177,189,29] - 29
2022-10-11T12:25:04.801Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":768,"srcaddr":48561,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":120,"securityuse":0,"timestamp":2840798,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[8,6,11,10,0]}}
2022-10-11T12:25:04.807Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":6,"manufacturerCode":null,"commandIdentifier":11},"Payload":{"cmdId":10,"statusCode":0},"Command":{"ID":11,"name":"defaultRsp","parameters":[{"name":"cmdId","type":32},{"name":"statusCode","type":32}]}},"address":48561,"endpoint":1,"linkquality":120,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2022-10-11T12:25:04.808Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:info  2022-10-11 14:25:04: MQTT publish: topic 'zigbee2mqtt/Living room Light ', payload '{"brightness":33,"color":{"x":0.3131,"y":0.3232},"color_mode":"color_temp","color_temp":153,"linkquality":120,"state":"ON"}'
2022-10-11T12:25:05.655Z zigbee-herdsman:controller:endpoint Command 0x2c1165fffe6d682a/1 lightingColorCtrl.moveToColorTemp({"colortemp":370,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2022-10-11T12:25:05.656Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x2c1165fffe6d682a:48561/1 (0,0,1)
2022-10-11T12:25:05.657Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":48561,"destendpoint":1,"srcendpoint":1,"clusterid":768,"transid":6,"options":0,"radius":30,"len":7,"data":{"type":"Buffer","data":[1,7,10,114,1,0,0]}}
2022-10-11T12:25:05.657Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,17,36,1,177,189,1,1,0,3,6,0,30,7,1,7,10,114,1,0,0,91]
2022-10-11T12:25:05.671Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2022-10-11T12:25:05.672Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2022-10-11T12:25:05.672Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2022-10-11T12:25:05.673Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2022-10-11T12:25:05.673Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-10-11T12:25:05.676Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,6,192]
2022-10-11T12:25:05.676Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,6,192]
2022-10-11T12:25:05.676Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,6] - 192
2022-10-11T12:25:05.677Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":6}
2022-10-11T12:25:05.677Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-10-11T12:25:05.697Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,25,68,129,0,0,0,3,177,189,1,1,0,112,0,203,51,44,0,0,5,8,7,11,10,0,177,189,29,109]
2022-10-11T12:25:05.698Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,25,68,129,0,0,0,3,177,189,1,1,0,112,0,203,51,44,0,0,5,8,7,11,10,0,177,189,29,109]
2022-10-11T12:25:05.698Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 25 - 2 - 4 - 129 - [0,0,0,3,177,189,1,1,0,112,0,203,51,44,0,0,5,8,7,11,10,0,177,189,29] - 109
2022-10-11T12:25:05.699Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":768,"srcaddr":48561,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":112,"securityuse":0,"timestamp":2896843,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[8,7,11,10,0]}}
2022-10-11T12:25:05.702Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":7,"manufacturerCode":null,"commandIdentifier":11},"Payload":{"cmdId":10,"statusCode":0},"Command":{"ID":11,"name":"defaultRsp","parameters":[{"name":"cmdId","type":32},{"name":"statusCode","type":32}]}},"address":48561,"endpoint":1,"linkquality":112,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2022-10-11T12:25:05.703Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
[... 19 lines cut ...]
2022-10-11T12:25:06.662Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":8,"manufacturerCode":null,"commandIdentifier":11},"Payload":{"cmdId":10,"statusCode":0},"Command":{"ID":11,"name":"defaultRsp","parameters":[{"name":"cmdId","type":32},{"name":"statusCode","type":32}]}},"address":48561,"endpoint":1,"linkquality":116,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2022-10-11T12:25:06.663Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
[... cut ...] p","parameters":[{"name":"cmdId","type":32},{"name":"statusCode","type":32}]}},"address":48561,"endpoint":1,"linkquality":112,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2022-10-11T12:25:07.483Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:info  2022-10-11 14:25:07: MQTT publish: topic 'zigbee2mqtt/Living room Light ', payload '{"brightness":33,"color":{"x":0.3804,"y":0.3767},"color_mode":"color_temp","color_temp":250,"linkquality":112,"state":"ON"}'
```

nekocentral, 2022-10-11:
Some more info about the device
Zigbee Model
TS0504B
Zigbee Manufacturer
_TZ3210_sroezl0s
Description
Livarno smart LED ceiling light
Support status
Supported
IEEE Address
0x2c1165fffe6d682a
Network address
0xBDB1
Manufacturer
[Lidl](https://www.zigbee2mqtt.io/supported-devices/#v=Lidl)
Model
[14153806L](https://www.zigbee2mqtt.io/devices/14153806L.html#lidl-14153806l)

Mosher23, 2022-10-15:
I have same light and can confirm the issue. Temperatures cannot be changed.

nekocentral, 2022-11-15:
with updated z2m still an issue

chris400, 2022-11-30:
count me in as well.
Got 3 of these

Zigbee2MQTT version
1.28.2

Adapter firmware version
0x26780700 (latest from 15th May 2022)

Adapter
ConBee2

log

> 2022-11-30 12:44:36Received MQTT message on 'zigbee2mqtt/Light Staircase FF/set' with data '{"color_temp":153}'
> 2022-11-30 12:44:36Publishing 'set' 'color_temp' to 'Light Staircase FF'
> 2022-11-30 12:44:36Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x50325ffffeb0c8e8!
> 2022-11-30 12:44:36Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x50325ffffeb0c8e8!
> 2022-11-30 12:45:05Received MQTT message on 'zigbee2mqtt/Light Staircase FF/set' with data '{"brightness":1}'
> 2022-11-30 12:45:05Publishing 'set' 'brightness' to 'Light Staircase FF'
> 2022-11-30 12:45:09Received MQTT message on 'zigbee2mqtt/Light Staircase FF/set' with data '{"brightness":233}'
> 2022-11-30 12:45:09Publishing 'set' 'brightness' to 'Light Staircase FF'
> 2022-11-30 12:45:10Received MQTT message on 'zigbee2mqtt/Light Staircase FF/set' with data '{"color":{"hex":"#FFA52C"}}'
> 2022-11-30 12:45:10Skipping state because of Home Assistant
> 2022-11-30 12:45:10Publishing 'set' 'color' to 'Light Staircase FF'
> 2022-11-30 12:45:11Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x50325ffffeb0c8e8!
> 2022-11-30 12:45:11Received MQTT message on 'zigbee2mqtt/Light Staircase FF/set' with data '{"color":{"hex":"#0000F9"}}'
> 2022-11-30 12:45:11Skipping state because of Home Assistant
> 2022-11-30 12:45:11Publishing 'set' 'color' to 'Light Staircase FF'
> 2022-11-30 12:45:11Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x50325ffffeb0c8e8!
> 2022-11-30 12:45:12Received MQTT message on 'zigbee2mqtt/Light Staircase FF/set' with data '{"color_temp":250}'
> 2022-11-30 12:45:12Skipping state because of Home Assistant
> 2022-11-30 12:45:12Publishing 'set' 'color_temp' to 'Light Staircase FF'
> 2022-11-30 12:45:12Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x50325ffffeb0c8e8!
> 2022-11-30 12:45:12Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x50325ffffeb0c8e8!
> 2022-11-30 12:45:14Received MQTT message on 'zigbee2mqtt/Light Staircase FF/set' with data '{"color_temp":370}'
> 2022-11-30 12:45:14Skipping state because of Home Assistant
> 2022-11-30 12:45:14Publishing 'set' 'color_temp' to 'Light Staircase FF'
> 2022-11-30 12:45:14Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x50325ffffeb0c8e8!
> 2022-11-30 12:45:14Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x50325ffffeb0c8e8!
> 2022-11-30 12:45:16Received MQTT message on 'zigbee2mqtt/Light Staircase FF/set' with data '{"color_temp":500}'
> 2022-11-30 12:45:16Skipping state because of Home Assistant
> 2022-11-30 12:45:16Publishing 'set' 'color_temp' to 'Light Staircase FF'
> 2022-11-30 12:45:16Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x50325ffffeb0c8e8!
> 2022-11-30 12:45:16Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x50325ffffeb0c8e8!

Mauricevb, 2022-12-01:
I have the same issue unfortunately. This might be caused by [#5932](https://github.com/Koenkk/zigbee2mqtt/issues/5932#issuecomment-768522354). 

@Koenkk Could it be possible that this is caused by your change related to #5932?

Koenkk, 2022-12-01:
this is not related to my change in https://github.com/Koenkk/zigbee-herdsman-converters/commit/ba60e89738c85b2985b44537467955e3a4483007

Can you:
- provide the herdsman debug log when setting the color temperature
- provide the data/database.db entry of this device.

chris400, 2022-12-01:
I am running zigbee2mqtt on Truenas Scale, so I am not sure if the [herdsman debug log  is enabled](https://www.zigbee2mqtt.io/guide/usage/debug.html#enabling-logging) correctly.

[mynewzigbee2mqtt_mynewzigbee2mqtt-59b595ffd4-qtns5_mynewzigbee2mqtt.log](https://github.com/Koenkk/zigbee2mqtt/files/10135740/mynewzigbee2mqtt_mynewzigbee2mqtt-59b595ffd4-qtns5_mynewzigbee2mqtt.log)

> /data # grep -i TS0504B database.db 
> {"id":18,"type":"Router","ieeeAddr":"0x2c1165fffe47fa92","nwkAddr":42766,"manufId":4098,"manufName":"_TZ3210_sroezl0s","powerSource":"Mains (single phase)","modelId":"TS0504B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,4096,8,768,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0504B","manufacturerName":"_TZ3210_sroezl0s","powerSource":1,"zclVersion":3,"appVersion":89,"stackVersion":0,"hwVersion":1,"dateCode":""}},"lightingColorCtrl":{"attributes":{"colorCapabilities":29}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":89,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1669923530627,"defaultSendRequestWhen":"immediate"}
> {"id":20,"type":"Router","ieeeAddr":"0x50325ffffeb0c8e8","nwkAddr":41240,"manufId":4098,"manufName":"_TZ3210_sroezl0s","powerSource":"Mains (single phase)","modelId":"TS0504B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,4096,8,768,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0504B","manufacturerName":"_TZ3210_sroezl0s","powerSource":1,"zclVersion":3,"appVersion":89,"stackVersion":0,"hwVersion":1,"dateCode":""}},"lightingColorCtrl":{"attributes":{"colorCapabilities":29,"colorMode":2,"currentX":8880,"currentY":2615,"currentHue":23,"currentSaturation":229,"colorTemperature":250,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}},"genLevelCtrl":{"attributes":{"currentLevel":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":89,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1669923744443,"defaultSendRequestWhen":"immediate"}
> {"id":21,"type":"Router","ieeeAddr":"0x2c1165fffe56605b","nwkAddr":4288,"manufId":4098,"manufName":"_TZ3210_sroezl0s","powerSource":"Mains (single phase)","modelId":"TS0504B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,4096,8,768,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0504B","manufacturerName":"_TZ3210_sroezl0s","powerSource":1,"zclVersion":3,"appVersion":89,"stackVersion":0,"hwVersion":1,"dateCode":""}},"lightingColorCtrl":{"attributes":{"57344":1000,"57600":[12,65536020],"61447":[],"61448":[],"61449":[],"61453":0,"colorCapabilities":29,"tuyaRgbMode":0,"currentHue":null}},"genOnOff":{"attributes":{"61440":0,"onOff":1}},"genLevelCtrl":{"attributes":{"61440":12}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":89,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1669923596890,"defaultSendRequestWhen":"immediate"}
>

Koenkk, 2022-12-02:
Seems the light does not support the standard Zigbee commands. Can someone make a sniff when controlling this device via the TuYa gateway?  https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

chris400, 2022-12-02:
Sorry can't do, don't own the gateway.
Anybody else? 🙏

flosch-dev, 2022-12-18:
Hello, 
I'm having the same device and also noticed that temp cannot be changed.
If I'm reading the description, I assume the lamp simply does not support color temp changes as datasheet is saying 4000k temp.

chris400, 2022-12-18:
> Hello, I'm having the same device and also noticed that temp cannot be changed. If I'm reading the description, I assume the lamp simply does not support color temp changes as datasheet is saying 4000k temp.

oh really? That would be really really sad.

@flosch-dev 
Do you own the vendor's bridge and app an can verify it? Just to be 100% sure.

flosch-dev, 2022-12-18:
No unfortunately I dont have such bridge. But if you compare the description with the bulb from lidl (https://www.lidl.de/p/livarno-home-led-lampe-16-millionen-farben/p100357153) there they do claim color temp.
Hence I doubt this Livarno LED do support color temp for white.

I was coming to this issue because I'm getting frustrated with another issue I have with such Lidl device: https://github.com/Koenkk/zigbee2mqtt/issues/14714
I'm wondering if anyone else is also having the problem with light going OFF when switching ON?

chris400, 2022-12-23:
I've reached out to the actual vendor. flosch-dev is right: You cannot change the color temp of the light :(
Thx for pointing this out!

Vendor's answer:
> Bei diesem Model ist eine Änderung der Farbtemperatur nicht vorgesehen und tatsächlich auch nicht möglich. 
> Neben der RGB-Ausführung hat die Leuchte wie beschrieben eine feste Farbtemperatur von 4000K, das heißt neutralweis.

@Koenkk So I guess removing the capability to change color temp is the consequence?

Koenkk, 2022-12-23:
Good to know. 

The IKEA LED1624G9 has the same issues, support color but not color temperature. There we have a solution so set the color temperature via the color (translation is done by z2m). I've added the same for this device now. Let me know if it works.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

flosch-dev, 2022-12-23:
thanks a lot for your fast update on this topic.
I just updated to dev branch to check the result and yes, the color temp is now changing. However, now the inbuilt white LEDs ( with it's 4000k) are not used anymore. They are much more brighter than the RGB calculated white temperature. 
So now I can change the color temp but I can't use the original white LED which I don't think is an advantage.

flosch-dev, 2022-12-23:
@Koenkk  - do you think there could be a relation to https://github.com/Koenkk/zigbee2mqtt/issues/14714 when sending a color temp value even it's not supported by the device?
Can you simply remove the color temp functionality for that device?

GsakuL, 2022-12-23:
@flosch-dev 
the HG08008 (TS0505B) has working CCT+RGB. As to say, yes RGBs there are also "weaker" like here. But the color_temp is using real C/W white LEDs.
It says so on the box "2000K - 6500K" ..."16 Millionen Farben"

Koenkk, 2022-12-23:
> However, now the inbuilt white LEDs ( with it's 4000k) are not used anymore. 

When were these used?

flosch-dev, 2022-12-23:
> > However, now the inbuilt white LEDs ( with it's 4000k) are not used anymore.
> 
> When were these used?

Whenever the attribute color_mode = color_temp, than the white LEDs were used. Just the temp never changed.
The attribute color_mode is from home assistant, not sure what is the corresponding name in zigbee.

Koenkk, 2022-12-24:
@flosch-dev I see, then we should leave in the color temp just as a way to enable the white leds. I've reverted my change and added a note about it in the docs.

closed 2023-01-31 by github-actions[bot]

elfensky, 2024-10-31:
Either you guys updated the config to work properly (props if so), or these issues were seemingly caused by first connecting it to the Lidl hub. 

I bought 2 of these, and directly connected them to HA. 

Able to control brightness, color(temp), on/off just fine. 

[youtube of the light in action](https://www.youtube.com/shorts/tqgh895mwwM)

chris400, 2024-11-02:
@elfensky pls share the exact product number of your device.

Seems a bit odd that you are able to change the colour temp although the vendor did tell me it's not possible to do so.

elfensky, 2024-11-02:
[image]

chris400, 2024-11-04:
@elfensky you are using [Lidl 399629_2110](https://www.zigbee2mqtt.io/devices/399629_2110.html#lidl-399629-2110) while this topic is about [Lidl 14153806L](https://www.zigbee2mqtt.io/devices/14153806L.html#lidl-14153806l) 😃