Koenkk/zigbee2mqtt #14441: Device TS0601 not recognized

zsoltty91, 2022-10-11:
### What happened?

I would like to add my new TRV (Tuya) device to zigbee2mqtt, but unfortunately it is not recognized:
Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_bvu2wnxz'

I found a discussion with the same problem but different manufacturer end name:
https://github.com/Koenkk/zigbee2mqtt/discussions/10873

Device link:
https://www.aliexpress.com/item/1005003952150617.html?spm=a2g0o.order_list.0.0.acae1802L0ePX4

Thank you!

### Zigbee2MQTT version

newest

### Adapter firmware version

202110

### Adapter

Slaesh

Koenkk, 2022-10-11:
Added!

Koenkk, 2022-10-11:
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-10-11 by Koenkk

zsoltty91, 2022-10-19:
Dear Koenkk,

I tried the thermostat with developer branch, and it does not work.

I see it in z2m now, but the type is not correct.

Z2M add as a device TV02:
https://www.zigbee2mqtt.io/devices/TV02-Zigbee.html#tuya-tv02-zigbee

But the correct type should be TS601:
https://www.zigbee2mqtt.io/devices/TS0601_thermostat_1.html

Can you change the type in the config?

Thank you!

deveylder, 2024-07-13:
Is it anyway possible to add the latest Tuya trv TS0601   _TZE284_p3dbf6qs to the list. 
Thank you

gdistasi, 2024-09-02:
I have also these devices at hand and need help, thanks