Koenkk/zigbee2mqtt #14446: WHD02 paired as TS0001

ciberknibal, 2022-10-11:
### What happened?

i buy whd02 but is paired as ts0001

### How to reproduce it (minimal and precise)

just paired

### Zigbee2MQTT version

1.28.0-1

### Adapter firmware version

20211217

### Adapter

sonoff

### Debug log

{"id":37,"type":"Router","ieeeAddr":"0xa4c138f8ca5dbfc2","nwkAddr":52282,"manufId":4417,"manufName":"_TZ3000_xkap8wtb","powerSource":"Mains (single phase)","modelId":"TS000F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,10,57344,1794,2820,4096,6280],"outClusterList":[25],"clusters":{"57344":{"attributes":{"53251":"AAAA"}},"genOnOff":{"attributes":{"32768":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2,"onOff":0,"onTime":0,"offWaitTime":0}},"genBasic":{"attributes":{"hwVersion":0}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":230,"rmsCurrent":78,"activePower":0}},"seMetering":{"attributes":{"currentSummDelivered":[0,0]}},"manuSpecificTuya_3":{"attributes":{"switchType":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1665484790846,"defaultSendRequestWhen":"immediate"}

Koenkk, 2022-10-12:
Added!

Changes will be available in the dev branch in a 8 hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-10-12 by Koenkk

m4tek, 2022-10-23:
I'm afraid that something went wrong here.

I have device: https://zigbee.blakadder.com/Aubess_AP-SMT-Breaker02-1CH.html

reporting as follows:
```
"TS000F" "_TZ3000_xkap8wtb"
```

with entry in the database.db as below:
```
{"id":65,"type":"Router","ieeeAddr":"0xa4c13810fac1d7fc","nwkAddr":60406,"manufId":4417,"manufName":"_TZ3000_xkap8wtb","powerSource":"Mains (single phase)","modelId":"TS000F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,10,57344,1794,2820,4096,6280],"outClusterList":[25],"clusters":{"57344":{"attributes":{"53251":"AAAA"}},"genOnOff":{"attributes":{"32768":0,"tuyaBacklightMode":2,"moesStartUpOnOff":0,"onOff":0,"onTime":0,"offWaitTime":0}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":244,"acVoltageMultiplier":1,"acVoltageDivisor":1,"acCurrentMultiplier":1,"acCurrentDivisor":1000,"acPowerMultiplier":1,"acPowerDivisor":1,"rmsCurrent":95,"activePower":0}},"genBasic":{"attributes":{"modelId":"TS000F","manufacturerName":"_TZ3000_xkap8wtb","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017"}},"seMetering":{"attributes":{"currentSummDelivered":[0,0],"divisor":100,"multiplier":1}},"manuSpecificTuya_3":{"attributes":{"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0021b7813a","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0021b7813a","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0021b7813a","endpointID":1}],"configuredReportings":[{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-2046940934},"lastSeen":1666548303658,"defaultSendRequestWhen":"immediate"}
```

that works almost perfectly (`current` is off by a factor of 3, but `power` is very accurate) using BHP13 profile `model: 'TS0121_plug'`

When detected as `WHD02` there is no power monitoring available.