Koenkk/zigbee2mqtt #14449: [New device support]: TuYa TS0601_smart_human_presense_sensor new manufacturerName: '_TZE200_holel4dk'

callacomp, 2022-10-12:
### Link

https://www.aliexpress.com/item/1005004148507209.html

### Database entry

{"id":61,"type":"Router","ieeeAddr":"0x385b44fffe332e6c","nwkAddr":51983,"manufId":4098,"manufName":"_TZE200_holel4dk","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u0012","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_holel4dk","powerSource":1,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1665532448248,"defaultSendRequestWhen":"immediate"}

### Comments

bought this MOES Human presence radar detection sensor model as reviews says it worked with HA and Z2M, but it didn't, has a new manufacturerName: '_TZE200_holel4dk'. Looked at existing Tuya convertor, copied it and added the manufacturerName: '_TZE200_holel4dk' and it worked. So can we just add this manufacturerName: '_TZE200_holel4dk' to the existing entry in the Tuya section

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_holel4dk'}],
    model: 'TS0601_smart_human_presense_sensor_test',
    vendor: 'TuYa',
    model: 'moes',
    description: 'Smart Human presence sensor test',
    fromZigbee: [fz.tuya_smart_human_presense_sensor],
    toZigbee: [tz.tuya_smart_human_presense_sensor],
    exposes: [e.illuminance_lux(), e.presence(),
        exposes.numeric('target_distance', ea.STATE).withDescription('Distance to target').withUnit('m'),
        exposes.numeric('radar_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(9).withValueStep(1)
            .withDescription('sensitivity of the radar'),
        exposes.numeric('minimum_range', ea.STATE_SET).withValueMin(0).withValueMax(9.5).withValueStep(0.15)
            .withDescription('Minimum range').withUnit('m'),
        exposes.numeric('maximum_range', ea.STATE_SET).withValueMin(0).withValueMax(9.5).withValueStep(0.15)
            .withDescription('Maximum range').withUnit('m'),
        exposes.numeric('detection_delay', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(0.1)
            .withDescription('Detection delay').withUnit('s'),
        exposes.numeric('fading_time', ea.STATE_SET).withValueMin(0).withValueMax(1500).withValueStep(1)
            .withDescription('Fading time').withUnit('s'),
        // exposes.text('cli', ea.STATE).withDescription('not recognize'),
        exposes.enum('self_test', ea.STATE, Object.values(tuya.tuyaHPSCheckingResult))
            .withDescription('Self_test, possible resuts: checking, check_success, check_failure, others, comm_fault, radar_fault.')],
};

module.exports = definition;
```

closed 2022-10-12 by Koenkk

callacomp, 2022-10-12:
Great, thanks. One thing they are very chatty. I've seen another one have this in the config

options: [exposes.binary('quiet', ea.SET, true, false).withDescription('When enabled the device will only broadcast changed values. Recommended to set this to true since this device is really chatty. Default = true. Also recommended to set "Debounce" to 1 and filter out the attributes you don\'t need')],

Is this needed

Koenkk, 2022-10-12:
Not needed, it just means more MQTT publishes will be done.