Koenkk/zigbee2mqtt #14452: TS011F_plug_3 not reporting energy use

thve25, 2022-10-12:
### What happened?

Hi,

I have a "_TZ3000_r6buo8ba" since several days that does not report energy use.

{
"child_lock": "LOCK",
"current": 0.98,
"indicator_mode": "on",
"linkquality": 189,
"power": 200,
"power_outage_memory": "restore",
"state": "ON",
"voltage": 217,
"device": {
"applicationVersion": 160,
"dateCode": "20210625",
"friendlyName": "smartplug-PC",
"hardwareVersion": 3,
"ieeeAddr": "0x70b3d52b6001415a",
"manufacturerID": 4660,
"manufacturerName": "_TZ3000_r6buo8ba",
"model": "TS011F_plug_3",
"networkAddress": 14959,
"powerSource": "Mains (single phase)",
"softwareBuildID": "20B+TZSKT41BS101",
"stackVersion": 1,
"type": "Router",
"zclVersion": 2
},
"energy": null,
"update": {
"state": null
},
"update_available": null
}

Strange thing is that I have the same plug (_TZ3000_r6buo8ba), but one that I ordered a few weeks earlier, that does report energy use:

{
"child_lock": "LOCK",
"current": 0.54,
"energy": 27.66,
"linkquality": 87,
"power": 87,
"power_outage_memory": "on",
"state": "ON",
"voltage": 231,
"device": {
"applicationVersion": 160,
"dateCode": "20210720",
"friendlyName": "smartplug-extra",
"hardwareVersion": 2,
"ieeeAddr": "0x70b3d52b6000953e",
"manufacturerID": 0,
"manufacturerName": "_TZ3000_r6buo8ba",
"model": "TS011F_plug_3",
"networkAddress": 53012,
"powerSource": "Mains (single phase)",
"softwareBuildID": "20B-TZSKT11BS3.4",
"stackVersion": 1,
"type": "Router",
"zclVersion": 2
},
"indicator_mode": null,
"update": {
"state": null
},
"update_available": null
}

### Zigbee2MQTT version

1.28.0-dev commit: c49f5b1

### Adapter firmware version

20210319

### Adapter

СС2652P

thve25, 2022-10-12:
I found this in the debug log:

Debug 2022-10-12 10:53:06Received Zigbee message from 'smartplug-PC', type 'raw', cluster 'seMetering', data '{"data":[8,9,10,0,0,0,37,84,0,0,0,0,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Debug 2022-10-12 10:53:06No converter available for 'TS011F_plug_3' with cluster 'seMetering' and type 'raw' and data '{"data":[8,9,10,0,0,0,37,84,0,0,0,0,0],"type":"Buffer"}'

forty, 2022-10-12:
Several same reports in another ticket:

https://github.com/Koenkk/zigbee2mqtt/issues/7405#issuecomment-1249648121
https://github.com/Koenkk/zigbee2mqtt/issues/7405#issuecomment-1268579986
https://github.com/Koenkk/zigbee2mqtt/issues/7405#issuecomment-1271558685
https://github.com/Koenkk/zigbee2mqtt/issues/7405#issuecomment-1275792690
https://github.com/Koenkk/zigbee2mqtt/issues/7405#issuecomment-1275829647

Both power reporting and OTA are broken

NickBarkerTV, 2022-10-12:
I realise you didn't write this for my plug. I have _TZ3000_okaz9tjs which also doesn't report power or allow OTA.
I tied the code (replacing _TZ3000_r6buo8ba with _TZ3000_okaz9tjs) hoping it would work.
I got the error shown in the PNG.
I am using Zigbee2MQTT Edge on Homeassistant.
Not sure if this helps or not. If not, please disregard.
[image]

thve25, 2022-10-12:
> Can you update to the latest dev (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) and check if just the energy reporting works with the following external converter: https://gist.github.com/Koenkk/81b27922a58bb2922bb9e52cbcf7d685
> 
> * save this as file next to `configuration.yaml` as `ext_converter.js`
> * add it to `configuration.yaml`:
> 
> ```yaml
> external_converters:
>   - ext_converter.js
> ```
> 
> * start z2m, check if issue is fixed

I updated to the latest dev docker and added that file:

Zigbee2MQTT version
1.28.0-dev commit: [c49f5b1](https://github.com/Koenkk/zigbee2mqtt/commit/c49f5b1)
Coordinator type
zStack3x0
Coordinator revision
20210319
Coordinator IEEE Address
0x00124b0024c0ea63
Frontend version
0.6.114

But now the plug is detected as a Lidl plug:

[image]

The only things it reports:

[image]

thve25, 2022-10-12:
Under the latest stable version of zigbee2mqtt,
the device is detected as tuya again

[image]

Koenkk, 2022-10-12:
If no energy value is exposed then this won't work.

> I have a "_TZ3000_r6buo8ba" since several days that does not report energy use.

I can't think of any change that could have impacted this. Has anything been changed in your setup?

thve25, 2022-10-12:
@Koenkk 
What I meant was that this is a new device for me.
It never worked.

nathan-gs, 2022-10-16:
I got it working by copying most from the Tuya file instead of the Lidl items, see:
https://github.com/nathan-gs/nix-conf/blob/main/smarthome/lidl-HG08673-FR-converter.js

thve25, 2022-10-17:
> I got it working by copying most from the Tuya file instead of the Lidl items, see: https://github.com/nathan-gs/nix-conf/blob/main/smarthome/lidl-HG08673-FR-converter.js

I tried this converter, but the measurement values didn't update.
The problem was that the option to poll the device was not present.

I added that option:
```

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
  metering_skip_duplicate: {
      ...fz.metering,
      convert: (model, msg, publish, options, meta) => {
          if (utils.hasAlreadyProcessedMessage(msg)) return;
          return fz.metering.convert(model, msg, publish, options, meta);
      },
  },
  electrical_measurement_skip_duplicate: {
      ...fz.electrical_measurement,
      convert: (model, msg, publish, options, meta) => {
          if (utils.hasAlreadyProcessedMessage(msg)) return;
          return fz.electrical_measurement.convert(model, msg, publish, options, meta);
      },
  },
};

const definition = {
  fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_r6buo8ba'}],
  model: 'HG08673-FR',
  vendor: 'Lidl',
  description: 'Silvercrest smart plug FR with power monitoring',
  ota: ota.zigbeeOTA,
  fromZigbee: [fz.on_off, fzLocal.electrical_measurement_skip_duplicate, fzLocal.metering_skip_duplicate, fz.ignore_basic_report,
    fz.tuya_switch_power_outage_memory, fz.ts011f_plug_indicator_mode, fz.ts011f_plug_child_mode],
  toZigbee: [tz.on_off, tz.tuya_switch_power_outage_memory, tz.ts011f_plug_indicator_mode, tz.ts011f_plug_child_mode],
  configure: async (device, coordinatorEndpoint, logger) => {
    const endpoint = device.getEndpoint(1);
    await endpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
    await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
    await reporting.rmsVoltage(endpoint, {change: 5});
    await reporting.rmsCurrent(endpoint, {change: 50});
    await reporting.activePower(endpoint, {change: 10});
    await reporting.currentSummDelivered(endpoint);
    endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
    endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
    device.save();
  },
  options: [exposes.options.measurement_poll_interval()],
  exposes: [e.switch(), e.power(), e.current(), e.voltage().withAccess(ea.STATE),
    e.energy(), exposes.enum('power_outage_memory', ea.ALL, ['on', 'off', 'restore'])
        .withDescription('Recover state after power outage'),
    exposes.enum('indicator_mode', ea.ALL, ['off', 'off/on', 'on/off', 'on'])
        .withDescription('Plug LED indicator mode'), e.child_lock()],
  onEvent: tuya.onEventMeasurementPoll,
};

module.exports = definition;
```

Now voltage, power and current are updated, but I still get "Null" for energy.

closed 2022-11-25 by github-actions[bot]

pohly, 2023-06-26:
I bought some new Lidl Silvercrest [HG08673](https://www.zigbee2mqtt.io/devices/HG08673.html#lidl-hg08673), Zigbee Model TS011F (same as above).

In my first test, I connected a water boiler and noticed that total "energy" increased over time although the current "power" remained at 0W. That had me a bit worried and brought me here.

But now it's been connected to some other device for a bit longer and now I see "power" 43W, "current" 0.18A, "voltage" 233V, so it seems to be working.

Still a bit odd that it starts reporting total energy increases before it reports current consumption, isn't it?

Anyway, thanks for making this work (at least for me) :sweat_smile: