Koenkk/zigbee2mqtt #14454: [Feature request]: Turn on Lights in a Group individually

tamueller, 2022-10-12:
### Is your feature request related to a problem? Please describe

I have a group of 4 Lights, and want to turn on only 1 Light. This is not possible, as turning on 1 light makes the group state change to on, which then turns on all the lights in the group.

### Describe the solution you'd like

Have a toggle similar to "off_state" that allows to make the group go on only when all lights are on, or when a single light turns on.

Or you could rename "off_state" to "change_state" and have one setting that covers both: Only turn on/off the group when all lights are turned on/off.

### Describe alternatives you've considered

Easy alternativ: Just disable turning on the group when one light is turned on. If I want to turn on all lights in the group, I can just turn on the group, instead of a single light. This would be a breaking change though and might break automations that don't turn in the groups but just one light.

### Additional context

No Additional Context

tamueller, 2022-10-12:
After some debugging on discord (thanks to Rene and Elminster) I have some additional information. Here is a log of what happens when I turn one light on:

```
Received MQTT message on 'zigbee2mqtt/Licht - Flur 4/set' with data '{"state":"ON"}'
2022-10-12 13:57:09 Publishing 'set' 'state' to 'Licht - Flur 4'
Info 2022-10-12 13:57:09MQTT publish: topic 'zigbee2mqtt/Gruppe - Flur Oben', payload '{"brightness":254,"color_mode":"color_temp","color_temp":250,"state":"ON"}'
Info 2022-10-12 13:57:09MQTT publish: topic 'zigbee2mqtt/Licht - Flur 4', payload '{"brightness":254,"color_mode":"color_temp","color_temp":250,"level_config":{"on_level":"previous"},"linkquality":222,"state":"ON"}'
Debug 2022-10-12 13:57:09Received MQTT message on 'zigbee2mqtt/Gruppe - Flur Oben/set' with data '{"state":"ON","transition":1.0,"brightness":254,"color_temp":250}'
Debug 2022-10-12 13:57:09Publishing 'set' 'brightness' to 'Gruppe - Flur Oben'
Debug 2022-10-12 13:57:09Publishing 'set' 'transition' to 'Gruppe - Flur Oben'
Debug 2022-10-12 13:57:09Publishing 'set' 'color_temp' to 'Gruppe - Flur Oben'
Info 2022-10-12 13:57:10MQTT publish: topic 'zigbee2mqtt/Licht - Flur 1', payload '{"brightness":254,"color_mode":"color_temp","color_temp":250,"linkquality":196,"state":"ON","update":{"state":"idle"},"update_available":null}'
Info 2022-10-12 13:57:10MQTT publish: topic 'zigbee2mqtt/Gruppe - Flur Oben', payload '{"brightness":254,"color_mode":"color_temp","color_temp":250,"state":"ON"}'
Info 2022-10-12 13:57:10MQTT publish: topic 'zigbee2mqtt/Licht - Flur 2', payload '{"brightness":254,"color_mode":"color_temp","color_temp":250,"linkquality":76,"state":"ON","update":{"state":null},"update_available":null}'
Info 2022-10-12 13:57:10MQTT publish: topic 'zigbee2mqtt/Licht - Flur 6', payload '{"brightness":254,"color_mode":"color_temp","color_temp":250,"linkquality":163,"state":"ON","update":{"state":null},"update_available":null}'
Info 2022-10-12 13:57:10MQTT publish: topic 'zigbee2mqtt/Licht - Flur 3', payload '{"brightness":254,"color_mode":"color_temp","color_temp":250,"level_config":{"on_level":"previous"},"linkquality":211,"state":"ON"}'
Info 2022-10-12 13:57:10MQTT publish: topic 'zigbee2mqtt/Licht - Flur 4', payload '{"brightness":254,"color_mode":"color_temp","color_temp":250,"level_config":{"on_level":"previous"},"linkquality":222,"state":"ON"}'
Info 2022-10-12 13:57:10MQTT publish: topic 'zigbee2mqtt/Licht - Flur 5', payload '{"brightness":254,"color_mode":"color_temp","color_temp":250,"level_config":{"on_level":"previous"},"linkquality":196,"state":"ON"}'
```

It looks like Z2M publishes an MQTT message that the group turned on, and then receives an MQTT message to turn the group on. I'm not sure if it receives it's "own" message there and reacts on it, or if HA receives the message, and replies with a MQTT-request to turn on the Group again. 

However, I think the logic in Z2M is inconsistent as it's possible to configure the group behaviour for turning off, but not for turning on. So having a "on_state" option that the group is only switched to on when all lights are turned on would solve this issue as well.

m0wlheld, 2022-10-14:
According to [documentation](https://www.zigbee2mqtt.io/guide/configuration/devices-groups.html#groups), you have to set „optimistic“ to false for that group.

tamueller, 2022-10-15:
@m0wlheld 
Sorry, I forgot to mention, that I already tried optimistic: false. The problem then is, that even when I set the group to off, the group and all individual lights still show as on in HomeAssistant, no matter how I set Off_State. This makes it impossible to turn the lights on again with HA :-(
Therefore I couldn't really test it with that setting, but if this already does what I want I should probably close the feature request, and see If I can pinpoint the bug with optimistic: false, or if it's just some kind of config problem in my case.

On the original problem: I solved that, by disabling the group in HA, it seems like the problem really was, that HA received the message that the group turned on, and then sent a message to turn the group on. By disabling the entity, I stopped that, the only downside is, that I can't use the HA "toggle service" anymore for that group, and zigbee doesn't have something like this, so my automations now have to check themself if the light is on/off and switch it off/on respectively, but at least I got it working

closed 2022-10-15 by tamueller