Koenkk/zigbee2mqtt #14455: Philips Hue Buckram spotlight 929003048301 not recognized

thbe51, 2022-10-12:
### Link

https://www.philips-hue.com/en-gb/p/hue-white-ambiance-buckram-quadruple-spotlight/8719514339163#specifications

### Database entry

?

### Comments

This seems to "Generation 2" of Philips 5047430P6. Same hardwavre but unrecognized. Other bulbs perhaps?

danieledwardgeorgehitchcock, 2022-10-12:
I've created a PR - the difference appears to be Bluetooth. The modern model generations also seem to differ between device color. I have kept it generic for now, but the description may change if the model id differs between white and black fixtures

closed 2022-10-12 by Koenkk

thbe51, 2022-10-15:
Hi and thanks for your jobb with this. Unfortunately it did no worked. It turned out to be some errors in philips.js. I've adjusted the file-paths in the header of the file: const exposes = require('/app/node_modules/zigbee-herdsman-converters/lib/exposes'); const fz = {...require('/app/node_modules/zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('/app/node_modules/zigbee-herdsman-converters/lib/legacy').fromZigbee}; const tz = require('/app/node_modules/zigbee-herdsman-converters/converters/toZigbee'); const ota = require('/app/node_modules/zigbee-herdsman-converters/lib/ota/'); const reporting = require('/app/node_modules/zigbee-herdsman-converters/lib/reporting'); const globalStore = require('/app/node_modules/zigbee-herdsman-converters/lib/store'); const utils = require('/app/node_modules/zigbee-herdsman-converters/lib/utils'); const e = exposes.presets; const ea = exposes.access; it then turned out the the lamp still was unsuported! Have added your patch. With this it seem to work. The lamp has three bulbs and as I understand they have different addresses or is there another way to solve this. At the same time I bought a lamp with four of these bulbs. The Type of the bulbs is 9290019533 { zigbeeModel: ['929003048101_01'], model: '9290019533', vendor: 'Philips', description: 'Hue white ambiance Buckram quadruple spotlight with Bluetooth', extend: hueExtend.light_onoff_brightness_colortemp({colorTempRange: [153, 454]}), }, { zigbeeModel: ['929003048101_02'], model: '9290019533', vendor: 'Philips', description: 'Hue white ambiance Buckram quadruple spotlight with Bluetooth', extend: hueExtend.light_onoff_brightness_colortemp({colorTempRange: [153, 454]}), }, { zigbeeModel: ['929003048101_03'], model: '9290019533', vendor: 'Philips', description: 'Hue white ambiance Buckram quadruple spotlight with Bluetooth', extend: hueExtend.light_onoff_brightness_colortemp({colorTempRange: [153, 454]}), }, THOMAS in Stockholm. > > ---------------------------------------- > > Hälsn. > > //Thomas >

danieledwardgeorgehitchcock, 2022-10-15:
Hi,

I am not sure what you are trying to do here.

In order to use the change that was merged, you need to swap to the Dev branch of Z2M following [this guide](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

I am using this branch with many Philips Hue products so, I can confirm that the paths don't need to be modified from being relative to full which is what you appear to be doing above.

I also see that you mention that your fixture has 3 bulbs whereas the link you copied above is a quad unit - the model number also corresponds with a quad unit.

Please try switching to the Dev branch first and see if that resolves your issues

thbe51, 2022-10-15:
Thanks. I was not aware of the dev-brach! I'll check it out as soon my lamps updating has finished which is quite slow. Side note: I bought two lamps, one with three bulbs and one with 4 bulbs. The type is the same of all of them. cheers!! > > ---------------------------------------- > > Hälsn. > > //Thomas >

thbe51, 2022-10-17:
Hi, The dev-version did not work either and I now understand why. I gave you the wrong info. The Philips unit '929003048301' is only iron and some cables. It consist of four G10 bulbs of modell '9290019533' They have four independent Zigbee-addresses of 929003048301_01 to _04. The other Philips unit is '929003048101' with three G10 bulbs, same modell but with addresses of 929003048101_01 to _03. The bulbs can independently be dimmed, turned off/on a.s.o. I came up with this external-converter file: { zigbeeModel: ['929003048101_01'], model: '9290019533', vendor: 'Philips', description: 'Hue white ambiance Buckram quadruple spotlight with Bluetooth', extend: hueExtend.light_onoff_brightness_colortemp({colorTempRange: [153, 454]}), }, { zigbeeModel: ['929003048101_02'], model: '9290019533', vendor: 'Philips', description: 'Hue white ambiance Buckram quadruple spotlight with Bluetooth', extend: hueExtend.light_onoff_brightness_colortemp({colorTempRange: [153, 454]}), }, { zigbeeModel: ['929003048101_03'], model: '9290019533', vendor: 'Philips', description: 'Hue white ambiance Buckram quadruple spotlight with Bluetooth', extend: hueExtend.light_onoff_brightness_colortemp({colorTempRange: [153, 454]}), }, { zigbeeModel: ['929003048301_01'], model: '9290019533', vendor: 'Philips', description: 'Hue white ambiance Buckram quadruple spotlight with Bluetooth', extend: hueExtend.light_onoff_brightness_colortemp({colorTempRange: [153, 454]}), }, { zigbeeModel: ['929003048301_02'], model: '9290019533', vendor: 'Philips', description: 'Hue white ambiance Buckram quadruple spotlight with Bluetooth', extend: hueExtend.light_onoff_brightness_colortemp({colorTempRange: [153, 454]}), }, { zigbeeModel: ['929003048301_03'], model: '9290019533', vendor: 'Philips', description: 'Hue white ambiance Buckram quadruple spotlight with Bluetooth', extend: hueExtend.light_onoff_brightness_colortemp({colorTempRange: [153, 454]}), }, { zigbeeModel: ['929003048301_04'], model: '9290019533', vendor: 'Philips', description: 'Hue white ambiance Buckram quadruple spotlight with Bluetooth', extend: hueExtend.light_onoff_brightness_colortemp({colorTempRange: [153, 454]}), }, Sorry for all the fuzz !! > > ---------------------------------------- > > Hälsn. > > //Thomas >