Koenkk/zigbee2mqtt #14456: Aurora AU-A1ZBPIAB  Plug-in Socket not reporting all values

dizzi77, 2022-10-12:
### What happened?

This Socket is only reporting Switch-State, Power and LQI. 

'Zigbee Model' reports as 'SingleSocket50AU'

### What did you expect to happen?

I expected that Current and Voltage would also be reported - only reports 'NULL'

I had expected 'Zigbee Model' to be - 'SingleSocket51AU' - Blakadder's Repository

### How to reproduce it (minimal and precise)

Inspecting the Herdsman convertor the older AU-A1ZBPIA version references Zigbee Model 'SingleSocket51AU', not SingleSocket50AU as I would have expected. Similarly the newer AU-A1ZBPIAB version references Zigbee Model 'SingleSocket50AU'. 

It looks although these Sockets have their Versions transposed

### Zigbee2MQTT version

1.28.0-1

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle PLUS -P

closed 2022-11-20 by github-actions[bot]

jimmer74, 2023-06-21:
Hi, hate to me-too, but I've got a bunch of these and they all report power usage fine and can be turned on and off just fine, but no current or voltage measurements (frontend just reports "Null" as their values).

[image]

[image]

Is there anything I can try to change/tweak in the settings? Appreciate any help/pointers you can provide!

jimmer74, 2023-06-21:
Just to add, on blakadder: it says it exposes power measurement & on/off, and has the correct image:
 
[image]

However on the zigbee2mqtt info page it says it exposes on/off, power, voltage, current, linkquality, but the image is of the prior model (AU-A1ZBPIA - without the B):

[image]

Have tried transposing the models in ./zigbee-herdsman-converters/devices/aurora_lighting.js, but this effectively does nothing. What's the next thing I can try (or does this model not have voltage/current caps)?