Koenkk/zigbee2mqtt #14462: Lidl Silvercrest 368308_2010 not showing week schedule values in frontend

Element2, 2022-10-13:
### What happened?

When using the Z2M frontend the week schedule values are not showing values on the 'Exposes' tab. The temp slider is showing 0 on all days for all settings, also the hour and minutes are not filled in. If the blue refresh button in front of a slider (for example monday) slider is clicked a red popup message appears with 'No convertor available for 'get' 'monday' [object Object].
The slider and hour and minute settings do work when setting a value, but the values are not kept when refreshing the page.  
The values of the week schedule can be seen in the 'State' tab 

### What did you expect to happen?

I expected to see all the week schedule values of temp, time and minutes for all the settings per day being displayed as set in the schedule on the thermostat.

### How to reproduce it (minimal and precise)

Pair a 368308_2010 Lidl Silvercrest thermostat, go to the frontend and on the 'Devices' tab select the paired thermostat. Then go to the 'Exposes' tab and scroll down to the week schedule. 

### Zigbee2MQTT version

1.28.0-dev commit e3e4a0a4

### Adapter firmware version

6.7.8.0 build 373

### Adapter

Hacked Lidl gateway EZSP v8

nathan-gs, 2022-11-16:
Same problem here. When refreshing it shows: cannot get object Monday.

realglotzi, 2022-11-29:
Same here. 
The problem exists since 1.27.1. Version 1.27.0 works as expected.

closed 2023-01-10 by github-actions[bot]

realglotzi, 2023-01-10:
Still not working with 1.29.1

veeroohre, 2023-04-12:
Any progress?