Koenkk/zigbee2mqtt #14463: zigbee2mqtt unexpected stop

wertherc, 2022-10-13:
### What happened?

The zigbee2mqtt docker stopped unexpectedly

### What did you expect to happen?

Not stopping unexpectedly

### How to reproduce it (minimal and precise)

Not reproduced so far

### Zigbee2MQTT version

1.27.2 commit: 7dc48fb

### Adapter firmware version

0x26720700

### Adapter

deconz (conbee II)

### Debug log

zigbee2mqtt      | Zigbee2MQTT:info  2022-10-12 21:23:48: MQTT publish: topic 'zigbee2mqtt/ingresso_piano1_luce', payload '{"brightness":254,"color_mode":"color_temp","color_temp":250,"color_temp_startup":250,"device":{"applicationVersion":16,"dateCode":"20201111","friendlyName":"ingresso_piano1_luce","hardwareVersion":1,"ieeeAddr":"0x8cf681fffe03f438","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"LED2003G10","networkAddress":38727,"powerSource":"Mains (single phase)","softwareBuildID":"1.0.012","stackVersion":103,"type":"Router","zclVersion":3},"linkquality":255,"power_on_behavior":"previous","state":"OFF","update":{"state":"idle"},"update_available":false}'
zigbee2mqtt      | Zigbee2MQTT:info  2022-10-12 21:23:48: MQTT publish: topic 'zigbee2mqtt/ingresso_piano2_luce', payload '{"brightness":254,"color_mode":"color_temp","color_temp":250,"color_temp_startup":250,"device":{"applicationVersion":16,"dateCode":"20201111","friendlyName":"ingresso_piano2_luce","hardwareVersion":1,"ieeeAddr":"0x2c1165fffedb7fa6","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"LED2003G10","networkAddress":32631,"powerSource":"Mains (single phase)","softwareBuildID":"1.0.012","stackVersion":103,"type":"Router","zclVersion":3},"linkquality":255,"power_on_behavior":"previous","state":"OFF","update":{"state":"idle"},"update_available":false}'
zigbee2mqtt      | Zigbee2MQTT:warn  2022-10-12 21:28:22: Failed to ping 'smart_plug' (attempt 1/2, Read 0x54ef4410004e9a6d/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received))
zigbee2mqtt      | UnhandledPromiseRejection: This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason "APS TIMEOUT".

danieledwardgeorgehitchcock, 2022-10-13:
Could you enable Z2M debug logging and ZHA debug logging and try and recreate the issue? Also, what device model / manufacturer is your device labelled 'smart_plug'?

wertherc, 2022-10-14:
Hello, I changed the logging in configuration.yaml
==============================================
advanced:
  log_level: debug
  timestamp_format: 'YYYY-MM-DD HH:mm:ss'
  log_file: log_%TIMESTAMP%.txt
  log_rotation: true
  log_output:
    - console
    - file
  log_symlink_current: true
=============================================
and added to the docker compose file this:
===============================================
 zigbee2mqtt:
    container_name: zigbee2mqtt
    image: docker.io/koenkk/zigbee2mqtt:1.27.2
    volumes:
      - "/opt/zigbee2mqtt:/app/data"
    devices:
      - "/dev/ttyACM0"
    environment:
      - "TZ=Europe/Rome"
      **- "DEBUG=zigbee-herdsman*"**
    restart: unless-stopped
    ports:
      - "8087:8087"
    networks:
      - home-automation-network
    depends_on:
      - mosquitto
    healthcheck:
      test: ["CMD-SHELL", "wget --no-verbose --tries=1 --spider http://localhost:8087 || exit 1"]
      interval: "60s"
      timeout: "10s"
      retries: 2
      start_period: "60s"
=============================================================================

wertherc, 2022-10-14:
Hello, this is the device:
https://www.zigbee2mqtt.io/devices/SP-EUC01.html#xiaomi-sp-euc01

I have added this part to the configuration.yml to enable debug logging:
`advanced:`
`log_level: debug`
`timestamp_format: 'YYYY-MM-DD HH:mm:ss'`
`log_file: log_%TIMESTAMP%.txt`
`log_rotation: true`
`log_output:`
`    - console`
`    - file`
`log_symlink_current: true`

Then I added the DEBUG variable in the docker-compose file:
`zigbee2mqtt:`
`    container_name: zigbee2mqtt`
`    image: docker.io/koenkk/zigbee2mqtt:1.27.2`
`    volumes:`
`      - "/opt/zigbee2mqtt:/app/data"`
`    devices:`
`      - "/dev/ttyACM0"`
`    environment:`
`      - "TZ=Europe/Rome"`
`      **- "DEBUG=zigbee-herdsman*"**`
`    restart: unless-stopped`
`    ports:`
`      - "8087:8087"`
`    networks:`
`      - home-automation-network`
`    depends_on:`
`      - mosquitto`
`    healthcheck:`
`      test: ["CMD-SHELL", "wget --no-verbose --tries=1 --spider http://localhost:8087 || exit 1"]`
`      interval: "60s"`
`      timeout: "10s"`
`      retries: 2`
`      start_period: "60s"`
Then if I understand correctly when and if there will be the crash, I will have to launch this command and send you the log.txt file. 
`docker logs zigbee2mqtt  > log.txt 2>&1`

Just to confirm that everything is correct. Is it? 
Thank you.

danieledwardgeorgehitchcock, 2022-10-14:
I'll look into this, see if I can find anything

closed 2022-11-21 by github-actions[bot]