Koenkk/zigbee2mqtt #14470: Error setting interface mode for Develco Energy Meter

msp1974, 2022-10-13:
### What happened?

I had previously set the interface mode on one of my Develco (Frient) Energy meters to gas.  This was using a previous version of zigbee2mqtt.

After upgrading to the current HA Addon version 1.28.0-1 I now get an error trying to change it back to Electricity.
No changes to the device firmware have been made between working and not working. Only upgrade of zigbee2mqtt HA addon.

This error occurs using either the zigbee2mqtt web interface or the HA entities.  Extract from the web interface debug log below:

### What did you expect to happen?

That the interface mode changed to the selected option

### How to reproduce it (minimal and precise)

Try and change the interface mode using the web interface on a Develco Energy Meter

### Zigbee2MQTT version

1.28.0

### Adapter firmware version

20210708

### Adapter

Sonoff ZBDongle-P

### Debug log

```
Debug 2022-10-13 16:08:00Received MQTT message on 'zigbee2mqtt/0x0015bc001b021df5/set' with data '{"interface_mode":"electricity"}'
Debug 2022-10-13 16:08:00Publishing 'set' 'interface_mode' to '0x0015bc001b021df5'
Info 2022-10-13 16:08:00MQTT publish: topic 'zigbee2mqtt/0x0015bc001b021df5', payload '{"battery_low":false,"check_meter":false,"current_summation":0,"energy":0,"interface_mode":"gas","last_seen":"2022-10-13T16:08:00+01:00","linkquality":123,"power":0,"pulse_configuration":100}'
Error 2022-10-13 16:08:00Publish 'set' 'interface_mode' to '0x0015bc001b021df5' failed: 'Error: Write 0x0015bc001b021df5/2 seMetering({"develcoInterfaceMode":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'INVALID_DATA_TYPE')'
Debug 2022-10-13 16:08:00Error: Write 0x0015bc001b021df5/2 seMetering({"develcoInterfaceMode":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'INVALID_DATA_TYPE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28) at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:397:22) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/devices/develco.js:209:17) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:246:36)
Info 2022-10-13 16:08:00MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'interface_mode' to '0x0015bc001b021df5' failed: 'Error: Write 0x0015bc001b021df5/2 seMetering({\"develcoInterfaceMode\":0}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Status 'INVALID_DATA_TYPE')'","meta":{"friendly_name":"0x0015bc001b021df5"},"type":"zigbee_publish_error"}'
```

msp1974, 2022-10-17:
Have provided a PR in the herdsman-converters repo to resolve this,

GSzabados, 2022-10-18:
Two question, how are you guys using it with your gas meter? If you use it with a reed relay or IR sensor could you please provide some details how have you wired it?

@msp1974, don't you want to fix the `current_summation` as well? It should be fixed at the herdsman converter as the value should be 48bit unsigned. And that should be used for the device.

https://github.com/Koenkk/zigbee-herdsman-converters/issues/4295

Koenkk, 2022-10-19:
Assuming this can be closed as https://github.com/Koenkk/zigbee-herdsman-converters/pull/4793 has been merged.

closed 2022-10-19 by Koenkk