Koenkk/zigbee2mqtt #14471: ROBB ROB_200-017-0 Zigbee smart plug not showing energy values

Tankdoz, 2022-10-13:
### What happened?

All other values are reported correctly (on/off, voltage, current, power) but the energy value is ‘nill’

### What did you expect to happen?

I expected a numerical value

### How to reproduce it (minimal and precise)

(Force) removing and reinstalling the device gives the same issue

### Zigbee2MQTT version

 1.23.0 commit: 6ea61ba

### Adapter firmware version

0x26660700

### Adapter

ConBee2/RaspBee2

Koenkk, 2022-10-18:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['ROB_200-017-0', 'HK-PLUG-A'],
    model: 'ROB_200-017-0',
    vendor: 'ROBB',
    description: 'Zigbee smart plug',
    fromZigbee: [fz.electrical_measurement, fz.on_off, fz.ignore_genLevelCtrl_report, fz.metering],
    toZigbee: [tz.on_off],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
        await reporting.onOff(endpoint);
        await reporting.readEletricalMeasurementMultiplierDivisors(endpoint);
        await reporting.rmsVoltage(endpoint);
        await reporting.rmsCurrent(endpoint);
        await reporting.activePower(endpoint);
        await reporting.currentSummDelivered(endpoint);
    },
    exposes: [e.power(), e.current(), e.voltage().withAccess(ea.STATE), e.switch(), e.energy()],
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

nielsvanbaal, 2022-10-18:
I tried, but I get " Integration error: external_converters - Integration 'external_converters' not found. "

Koenkk, 2022-10-18:
You should add it to the z2m configuration.yaml not the Home assistant one.

nielsvanbaal, 2022-10-18:
Yes, a mate of mine suggested that too. Sometimes I feel such a n00b still. I really appreciate your help!

So it is added now everything boots, but still none kWh unfortunately.

dromer1967, 2022-10-19:
The above ext_converter.js did not solve the problem.

However, after adding

`await reporting.readMeteringMultiplierDivisor(endpoint);`

to it, it started to work :) I looked at the js for ROB_200-017-1 and checked the difference :)

nielsvanbaal, 2022-10-19:
Nice! This works like a charm now. Thnx!

Koenkk, 2022-10-19:
@dromer1967 great work, integrated the changes!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Looney1976, 2024-01-07:
I know this is an old issue, I'm using the 017 and a Robb zigbee build-in switch and both are supported by Z2M but both do not show temperature or power. Do I need to perform some action in Z2M to get this to work?