Koenkk/zigbee2mqtt #14477: [New device support]: Fantem ZB006-X not supported, add new device manufacturer _TZ3210_ebbfkvoy

bladehstream, 2022-10-14:
### Link

https://www.zigbee2mqtt.io/devices/ZB006-X.html

### Database entry

{"id":13,"type":"Router","ieeeAddr":"0x14b457fffe0e4971","nwkAddr":13241,"manufId":4098,"manufName":"_TZ3210_ebbfkvoy","powerSource":"Mains (single phase)","modelId":"TS110F","epList":[1,2,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,4,5,6,8,61184,64770],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS110F","manufacturerName":"_TZ3210_ebbfkvoy","stackVersion":0,"dateCode":"","zclVersion":3}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":0}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":1,"inClusterList":[],"outClusterList":[6,8],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":1,"inClusterList":[],"outClusterList":[6,8],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":160,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1665724276434,"defaultSendRequestWhen":"immediate"}

### Comments

Attempted to pair this device as apparently it is supported, paired fine but failed interview as the manufacturer was not in the fingerprint?

info  2022-10-14 14:59:19: Device '0x14b457fffe0e4971' joined
info  2022-10-14 14:59:19: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x14b457fffe0e4971","ieee_address":"0x14b457fffe0e4971"},"type":"device_joined"}'
info  2022-10-14 14:59:19: MQTT publish: topic 'zigbee2mqtt/0x14b457fffe0e4971/availability', payload '{"state":"online"}'
info  2022-10-14 14:59:19: Starting interview of '0x14b457fffe0e4971'
info  2022-10-14 14:59:19: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x14b457fffe0e4971","ieee_address":"0x14b457fffe0e4971","status":"started"},"type":"device_interview"}'
info  2022-10-14 14:59:19: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x14b457fffe0e4971","ieee_address":"0x14b457fffe0e4971"},"type":"device_announce"}'
info  2022-10-14 14:59:20: Successfully interviewed '0x14b457fffe0e4971', device has successfully been paired
warn  2022-10-14 14:59:20: Device '0x14b457fffe0e4971' with Zigbee model 'TS110F' and manufacturer name '_TZ3210_ebbfkvoy' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
info  2022-10-14 14:59:20: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x14b457fffe0e4971","ieee_address":"0x14b457fffe0e4971","status":"successful","supported":false},"type":"device_interview"}'
warn  2022-10-14 14:59:24: Received message from unsupported device with Zigbee model 'TS110F' and manufacturer name '_TZ3210_ebbfkvoy'

danieledwardgeorgehitchcock, 2022-10-14:
Hi,

Could you post a link to where you purchased the device from?

Just need to check the existing model with the new one you reported to see if I can just 'tag on' the new identifier..

bladehstream, 2022-10-14:
Sure thing @danieledwardgeorgehitchcock ,that was my hope:
https://aliexpress.com/item/1005001907709905.html

danieledwardgeorgehitchcock, 2022-10-14:
Ok, it looks like the same module - I will create a PR with some tweaks

closed 2022-10-18 by Koenkk