Koenkk/zigbee2mqtt #14480: Lidl 14147306L recognized as TuYa TS0502A

akula3, 2022-10-14:
### What happened?

I successfully paired Lidl 14147**3**06L ceiling light with Sonoff Zigbee 3.0 CC2652P Dongle Plus, the device is visible and working.
Strange is that it is recognized as a Tuya device TS0502A.
In the supported devices list I found only Lidl 14147**2**06L.
So, I can't say, if my device behaves as expected or not and if it is normally, that Lidl 14147306L will be recognized as TuYa TS0502A.
Corresponding data base entry is:
{"id":10,"type":"Router","ieeeAddr":"0x588e81fffedbbe84","nwkAddr":38380,"manufId":4098,"manufName":"_TZ3000_5fkufhn1","powerSource":"Mains (single phase)","modelId":"TS0502A","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0502A","manufacturerName":"_TZ3000_5fkufhn1","powerSource":1,"zclVersion":3,"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":""}},"lightingColorCtrl":{"attributes":{}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1665750954874,"defaultSendRequestWhen":"immediate"}

### What did you expect to happen?

Lidl 14147306L should be recognized as Lidl 14147306L

### How to reproduce it (minimal and precise)

add Lidl 14147306L, see wrong data in the device overview

### Zigbee2MQTT version

1.28.0-1

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0 CC2652P Dongle Plus

danieledwardgeorgehitchcock, 2022-10-14:
Hi,

This is correct behaviour - Lidl whitelabels a lot of Heiman devices which in turn use the TuYa platform.

When devices are created, a whitelabel reference to the Lidl device is created for the purposes of documentation and showing support however, Z2M just treats the Lidl product the same as its originating manufacturer

closed 2022-10-18 by Koenkk

akula3, 2022-10-18:
very well, thanks a lot!

AndreKR, 2024-02-29:
> for Lidl devices we make a difference and we detect them as Lidl instead (since the brand is very famous).

In that case you might be interested that the 14147406L I just got is recognized as a TS0502A.

If there's anything you can do about the transition time setting not working, that would be great, too. It has a forced transition time of about 1 second and I hate it so much. If I tap "on" I want light now, not a second later.

[image]

[image]

```
{"id":98,"type":"Router","ieeeAddr":"0x842e14fffe2223e8","nwkAddr":5962,"manufId":4098,"manufName":"_TZ3000_9evm3otq","powerSource":"Mains (single phase)","modelId":"TS0502A","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0502A","manufacturerName":"_TZ3000_9evm3otq","stackVersion":0,"dateCode":""}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{"onLevelSupported":false}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1811148075},"lastSeen":1709170542329, "defaultSendRequestWhen":"immediate"}
```