Koenkk/zigbee2mqtt #14481: Zlinky unable to read "tomorrow color" or "couleur du lendemain" (in french)

BBESINET, 2022-10-14:
### What happened?

I bought a Lixee Zlinky_TIC paired with Zigbee2MQTT (1.28.0-1).

All the relevant informations are OK except one:

Tomorow color (I have a TEMPO contract) is not changing and always reporting "----"

### What did you expect to happen?

I expected to have either a string changing from nothing to Blue , White or Red as explained in documentation or either a numerical value changing from 0 to 3 (original info coded on 2 bits).

### Zigbee2MQTT version

1.28.0

### Adapter firmware version

20210708

### Adapter

SonOff Zigbee 3.0 USB Dongle plus

BBESINET, 2022-10-14:
Sorry this is my first time here... I filled all part but wanted on top to say:
Hello !

and Thank you for your help.

BBESINET, 2022-10-14:
PS: I am using Zigbee2MQTT into Home Assistant as a module. 
No fancy stuff... just Home assistant OS on a R.PI 3 and the Sonoff dongle to start domotic.

Trying to create scenario for my Tempo Elec contract brought me into this issue.

Regards.
BBESINET

BBESINET, 2022-11-03:
Any news on this topic ? 

Most probably there is an issue in the transformation of the information sent by Zlinky into a string. This information is normaly sent by the linky device between 20:00 and 6:00 allowing to determine the next day electricity cost.

It is very useful for automations to anticipate a high cost and thus anticipate some high consumers before 6:00 and the cost changes.

Information send by linky is based on 2 bits (thus coded between 0, 1, 2 or 3) and should be transformed by zlinky into "nothing", "bleu", "blanc" or "rouge" (blue, white, red). Other integration (ZHA for exemple) are able to recover this information, thus the issue is concerning Zigbee2mqtt integration of the Zlinky.

We must not be numerous using this TEMPO contract, but if this could be fixed in another Z2M soft, that would be great !

Best regards to all.
BBESINET

BBESINET, 2022-11-23:
Dear all.

Some update:
Despite updates of both Zlinky to the latest firmware and Zigbee2MQTT to version 1.28.2-1. The problem is still there.

Regards.

BBESINET

vk496, 2022-11-24:
Hello!

You Linky is in Historique or Standard? Have you tried to manually setting that configuration in your zlinky (and restarting)?

BBESINET, 2022-11-24:
My Zlinky is in historique mode.

In parallel I raised the issue there : [https://github.com/fairecasoimeme/Zlinky_TIC/issues/139](url)

And the "issue" might be on the source side.
I checked on my linky, and while I had in my internal house led device (a kind of CPL) the tomorow color display, directly on the linky, the tomorow color was indicated ---- and thus coherent with Zlinky output.

The behaviour might be that this indication is only used when the tomorow color is diferent from the "today" color...

To be confirmed when we will have color change this winter...

vk496, 2022-11-24:
But your Linky is in Historique too?

An empty string could indicate some issue from Z2M implementation, but those ---- seems to be valid output from Linky.

Btw, there is a technical pdf of enedis which describe EXACTLY the outputs of Linky. I suggest you to check it. Pheraps there is some other attribute which give you the warning/notification you expect and is not working correctly...

BBESINET, 2022-11-24:
Yes. 

Everything is coherent except one thing about this "tomorow color" which is very specific to my kind of contract (Tempo).

What is misleading is that every day around 20h00 I have a small box in the house (working on a kind of CPL technology) that gives me the tomorow color thanks to 3 LED. 
This box is always working thus I was believing that every night this info was available on TIC.

Yesterday evening I checked my LINKY and this box and had the following: 
- the box had led ON indicating current cost "HP Blue" and tomorow color "Blue"
- the LINKY displayed current cost as "HP Blue" but tomorow color ("DEMAIN" key) as "----"
- Zlinky was fully coherent with LINKY

Thus there is a possibility that this information is only updated when there is a color change... 

I will investigate it next time there is a color change (probable in the comming weeks due to winter approaching...)

I will keep you posted and change the status of the issue if it is finally working...

BBE

vk496, 2022-11-24:
Pheraps is a momentaneous change? Did you checked the last 48h history (I use Home Assistant for that for example)

BBESINET, 2022-11-24:
Yes I checked history and journal through Home assistant...

rdorys, 2022-11-24:
Same here, the tomorrow color never changes on the evening.
[image]

rdorys, 2022-11-28:
Hi, update on my side,

The tomorrow color is now 'BLAN'.

[image]

Maybe it changes only when tomorrow has a different color than today.

We need more days to understand the behaviour.

BBESINET, 2022-11-28:
Confirmed here too.

No issue from Zigbee2mqtt nor Zlinky.

Enedis is only sending this information when the color of tomorow is different from today...

I close the issue. 
Thanks for the support

BBESINET, 2022-11-28:
Thanks to all for the support

closed 2022-11-28 by BBESINET

BBESINET, 2022-12-02:
Following several white days and having tomorrow a blue one, I correct a little bit the statement above for people having the same question in the future.

Enedis is apparently sending the information only if tomorrox color is BLAN or ROUG. When tomorrow color is BLEU there is no feedback, and Zlinky send "----". (validated for "white" days, but still to be confirm next "red" day...)

Apparent behaviour is the following :
Tomorrow color blue   => value is "----" between 22h and 6h
Tomorrow color white => value is "BLAN" between 22h and 6h
Tomorrow color red    => value is "ROUG" between 22h and 6h
value always "----" between 6h and 22h

BBESINET, 2022-12-05:
I do not use this value (I am in single phase). And those values are not reported since I set following Zlinky settings (specific) in Z2M:
- linky_mode to historic
- energy_phase to single_phase
- production to false
- tarif to "Historique - BBR"
- KWh precision to 3

The values reported by my Zlinky are:
- tomorow color
- active tier register delivered
- ADPS
- the 6 indexes (BBRHCJB to BBRHPJR)
- HHPHC (unused but reported)
- IINST
- IMAX
- ISOUSC
- PAPP
- Current Tarif
- Serial number

All my indexes are progressing (at least for blue and white... still no red days for the time being):
BBRHCJB, BBRHPJB, BBRHCJW and BBRHPJW.

BBRHCJR and BBRHPJR not yet observed alive, but coherent with the value displayed on Linky...

BBESINET, 2022-12-09:
Following first "red" day. I can confirm the hypothesis above for the behaviour of both Linky and Zlinky:

Tomorrow color blue => value is "----" between 22h and 6h (and not "BLEU" as I was expecting and why I created this issue!)
Tomorrow color white => value is "BLAN" between 22h and 6h
Tomorrow color red => value is "ROUG" between 22h and 6h
value always "----" between 6h and 22h when tomorrow color is unknown on linky.