Koenkk/zigbee2mqtt #14489: SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P LQI low

maframan, 2022-10-15:
### What happened?

I've problem with zigbee signal. The sonoff gateway is plugged on USB 3.0 NUC port and the first end device is 70 cm away and the LQI is equal to 134. I've tried
- with usb 3.0 extension cable for gateway without success
- with 8dbi antenna on gateway without success

Now the gateway is plugged directly to NUC on USB 3.0 front side.

How to troubleshoot my problem? what could my problem be? Before the NUC I had the raspberry 4 model b in the same position without any problem.  The problem is the usb 3.0 port?

### What did you expect to happen?

LQI too high

### How to reproduce it (minimal and precise)

N/A

### Zigbee2MQTT version

1.28.0-1

### Adapter firmware version

CC1352P2_CC2652P_launchpad_coordinator_20220219.zip

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Debug log

N/A

JacekKac, 2022-10-15:
don't know what is "NUC" but all my LQI issues went away after switching all coordinator/router devices to sonoff "E" dongle. dramatic improvement of range and stability.
Befor that i have tried all "hints' like usb extension, different channels, aways from wifi router etc. There was no improvement at all.

mickadam29, 2022-10-22:
Switch to ZBDongle-E. This model is experimental but you can use it. And flash you PLUS with router firmware. If needed.
My feedback

For higher LQI reasons, I installed a ZBDongle-E to replace a ZBdongle-Plus.
I know it's experimental.
I didn't flash it. I kept the original SONOFF firmware.
I managed to include most of my ZIGBEE equipment, with some difficulties for some.
Those who could not be interviewed are SONOFF equipment:
SNZB-01
SNZB-02
SNZB-03
SNZB-04
Here is a solution while waiting for ZBDongle-E support
I flashed my ZBDongle-Plus in router mode with the firmware provided on Koenkk's git.
Version: CC1352P2_CC2652P_launchpad_router_20220125
When including equipment, I do so near the ZBDongle-Plus. The interview is going well and the equipment is correctly added.
Then I disconnect the ZBDongle-Plus dongle, in order to force the equipment to change router.
The 4 types of equipment are perfectly included, functional. And only the ZBDongle-E remains on my network

closed 2022-11-29 by github-actions[bot]