Koenkk/zigbee2mqtt #14495: Heiman  HS3CG Combustible Gas Sensor wrong converter

mikevlz, 2022-10-15:
### What happened?

Currently(v1.28) this gas sensor always reports no gas leak. 
`    {
        zigbeeModel: ['GASSensor-N', 'GASSensor-N-3.0'],
        model: 'HS3CG',
        vendor: 'HEIMAN',
        description: 'Combustible gas sensor',
        fromZigbee: [fz.ias_gas_alarm_1],
        toZigbee: [],
        exposes: [e.gas(), e.battery_low(), e.tamper()],
    },
`
Here fromZigbee should be changed to fz.ias_gas_alarm_2 because device reports `type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":32}' from endpoint 1 with groupID 0` when no gas leak/not testing, and `type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":34}' from endpoint 1 with groupID 0` when gas leak detected/test button pressed.

### What did you expect to happen?

Correct reports "gas:true" when leak detected/test button pressed

### How to reproduce it (minimal and precise)

Press test button on sensor. Mqtt status will report "gas:false".

### Zigbee2MQTT version

1.28.0

### Adapter firmware version

20201010

### Adapter

cc2538

### Debug log

debug 2022-10-15 15:10:21Received Zigbee message from '0x00158d0004fcf02c', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":32}' from endpoint 1 with groupID 0
info 2022-10-15 15:10:21MQTT publish: topic 'zigbee2mqtt/0x00158d0004fcf02c', payload '{"battery_low":false,"gas":false,"last_seen":"2022-10-15T12:10:21.026Z","linkquality":131,"tamper":false}'
debug 2022-10-15 15:10:52Received Zigbee message from '0x00158d0004fcf02c', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":34}' from endpoint 1 with groupID 0
info 2022-10-15 15:10:52MQTT publish: topic 'zigbee2mqtt/0x00158d0004fcf02c', payload '{"battery_low":false,"gas":false,"last_seen":"2022-10-15T12:10:52.626Z","linkquality":139,"tamper":false}'

mikevlz, 2022-10-15:
PR: https://github.com/Koenkk/zigbee-herdsman-converters/pull/4787

Koenkk, 2022-10-18:
Fixed by https://github.com/Koenkk/zigbee-herdsman-converters/pull/4787

closed 2022-10-18 by Koenkk

Mad-Andy, 2025-07-21:
Hello, I've requested the support for this device years ago, and everything worked like a charm (or so I thought), but I've noticed just today (v. 2.5.1) performing a test on my HS3CG gas sensor, that my gas sensor always now reports no gas leak.

Maybe something has changed with firmware updates and mine has a older version, but I cannot find any advice on how to update it.

Do you think is possible to add a check on firmware version so that you can expose fz.ias_gas_alarm_2 for newer or fz.ias_gas_alarm_1 for older ones?

here are my logs (removing and re-pairing)

`info 2025-07-20 22:00:26z2m: Removing device 'Gas Cucina' (block: false, force: true)
info 2025-07-20 22:00:26z2m:mqtt: MQTT publish: topic 'homeassistant/binary_sensor/0x00158d0001deec28/gas/config', payload ''
info 2025-07-20 22:00:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Gas Cucina', payload ''
info 2025-07-20 22:00:26z2m: Successfully removed device 'Gas Cucina' (block: false, force: true)
info 2025-07-20 22:00:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/remove', payload '{"data":{"block":false,"force":true,"id":"Gas Cucina"},"status":"ok","transaction":"npk2r-1"}'
info 2025-07-20 22:00:26z2m:mqtt: MQTT publish: topic 'homeassistant/binary_sensor/0x00158d0001deec28/battery_low/config', payload ''
info 2025-07-20 22:00:26z2m:mqtt: MQTT publish: topic 'homeassistant/binary_sensor/0x00158d0001deec28/tamper/config', payload ''
info 2025-07-20 22:00:26z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x00158d0001deec28/linkquality/config', payload ''
info 2025-07-20 22:00:44z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/health', payload '{"response_time":1753041644109,"os":{"load_average":[0.17,0.12,0.1],"memory_used_mb":1283.68,"memory_percent":66.1197},"process":{"uptime_sec":87610,"memory_used_mb":106.63,"memory_percent":2.8144},"mqtt":{"connected":true,"queued":0,"published":986,"received":65},"devices":{"0x00158d0007d39e2c":{"messages":7,"messages_per_sec":0.0001,"leave_count":0,"network_address_changes":0},"0x00158d000288ad56":{"messages":29,"messages_per_sec":0.0003,"leave_count":0,"network_address_changes":0},"0x00158d00034983a4":{"messages":152,"messages_per_sec":0.0018,"leave_count":0,"network_address_changes":0},"0x00158d0007e56e5f":{"messages":102,"messages_per_sec":0.0013,"leave_count":0,"network_address_changes":0}}}'
info 2025-07-20 22:01:41z2m: Zigbee: allowing new devices to join.
info 2025-07-20 22:01:41z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254},"status":"ok","transaction":"npk2r-2"}'
info 2025-07-20 22:02:20zh:controller: Interview for '0x00158d0001deec28' started
info 2025-07-20 22:02:20z2m: Device '0x00158d0001deec28' joined
info 2025-07-20 22:02:20z2m: Starting interview of '0x00158d0001deec28'
info 2025-07-20 22:02:20z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d0001deec28","ieee_address":"0x00158d0001deec28"},"type":"device_joined"}'
info 2025-07-20 22:02:20z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d0001deec28","ieee_address":"0x00158d0001deec28","status":"started"},"type":"device_interview"}'
info 2025-07-20 22:02:20z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d0001deec28","ieee_address":"0x00158d0001deec28"},"type":"device_announce"}'
info 2025-07-20 22:02:22zh:controller: Succesfully interviewed '0x00158d0001deec28'
info 2025-07-20 22:02:22z2m: Successfully interviewed '0x00158d0001deec28', device has successfully been paired
info 2025-07-20 22:02:22z2m: Device '0x00158d0001deec28' is supported, identified as: Heiman Combustible gas sensor (HS3CG)
info 2025-07-20 22:02:22z2m:mqtt: MQTT publish: topic 'homeassistant/binary_sensor/0x00158d0001deec28/gas/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":**{"hw_version":16,**"identifiers":["zigbee2mqtt_0x00158d0001deec28"],"manufacturer":"Heiman","model":"Combustible gas sensor","model_id":"HS3CG","name":"0x00158d0001deec28","via_device":"zigbee2mqtt_bridge_0x00124b0018e1b298"},"device_class":"gas","object_id":"0x00158d0001deec28_gas","origin":{"name":"Zigbee2MQTT","sw":"2.5.1","url":"https://www.zigbee2mqtt.io"},"payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0x00158d0001deec28","unique_id":"0x00158d0001deec28_gas_zigbee2mqtt","value_template":"{{ value_json.gas }}"}'
info 2025-07-20 22:02:22z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Combustible gas sensor","exposes":[{"access":1,"description":"Indicates whether the device detected gas","label":"Gas","name":"gas","property":"gas","type":"binary","value_off":false,"value_on":true},{"access":1,"category":"diagnostic","description":"Indicates if the battery of this device is almost empty","label":"Battery low","name":"battery_low","property":"battery_low","type":"binary","value_off":false,"value_on":true},{"access":1,"description":"Indicates whether the device is tampered","label":"Tamper","name":"tamper","property":"tamper","type":"binary","value_off":false,"value_on":true},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"HS3CG","options [... cut ...] sages_per_sec":0.0001,"leave_count":0,"network_address_changes":0},"0x00158d000288ad56":{"messages":29,"messages_per_sec":0.0003,"leave_count":0,"network_address_changes":0},"0x00158d00034983a4":{"messages":152,"messages_per_sec":0.0018,"leave_count":0,"network_address_changes":0},"0x00158d0007e56e5f":{"messages":102,"messages_per_sec":0.0012,"leave_count":0,"network_address_changes":0},"0x00158d0001deec28":{"messages":12,"messages_per_sec":0.0238,"leave_count":0,"network_address_changes":0}}}'`