Koenkk/zigbee2mqtt #14497: [Feature request]: compatibility with matter usb stick

SeveranExp, 2022-10-15:
### Is your feature request related to a problem? Please describe

Not a problem.

### Describe the solution you'd like

Home assistant gang have just come out with the following dongle:
https://www.home-assistant.io/skyconnect/

It’s both a zigbee and matter dongle, with the thread compatibility coming later.
I don’t know whether it makes sense to bundle the matter devices compatibility on this project, or, to have something else entirely different. But since the usb dongle has compatibility with z2m already, maybe there is an intelligent way to make this happen, without too much work?

### Describe alternatives you've considered

Matter and thread are so new that there’s barely any alternatives around. Im an openHAB user, so whatever I get, will have to be compatible with homeassistant/openHAB/zigbee2mqtt since that’s the basis of my setup.

### Additional context

Just looking to bring the attention up for this topic. I’m not expecting an instant solution :)

jerrm, 2022-10-15:
It is not a "Matter dongle."  Matter and Thread are not the same thing but are often conflated. 

Initially it likely works out of the box with z2m as zigbee only.  It uses the same SI Labs chip/ezsp protocol already supported by z2m (albeit currently considered  experimental).

If it will work when/if dual protocol is enabled is unknown to me, but expect it will be possible.   Assuming the ezsp protocol stays the same, z2m could connect to the zigbee daemon running on the server via a socket, vs connecting serially to the daemon running on the stick.

But... we will have to see the final implementation.

SeveranExp, 2022-10-15:
> If it will work when/if dual protocol is enabled is unknown to me, but expect it will be possible. Assuming the ezsp protocol stays the same, z2m could connect to the zigbee daemon running on the [...]

Yep, I fully understand that. That is why I said that I’m not particularly looking for an “instant solution” - more like “raising awareness”. It will certainly take some time and experimentation, but as long as this request is here, hopefully someone will eventually pick it up :)

SeveranExp, 2022-11-15:
Just to add (before the bot closes this), matter v1.0 is out and some devices have begun to appear online. Not sure when it would be good to start to take a look at this but things seem to be looking real nowadays.

Hedda, 2022-11-18:
FYI, you guys can follow kirovilya's development and help out as a tester in the dedicated zigbee-herdsman issue about EZSP here:

https://github.com/Koenkk/zigbee-herdsman/issues/319

I just posted this question there which is directly related as it to support a dependency needed in zigbee-herdsman to support Silicon Labs based adapters with RPC (Radio Co-Processor) firmware instead of NCP (Network Co-Processor) firmware:

> kirovilya wondering if you have looked or are planing to look at the option to include Zigbeed (Silicon Labs Zigbee Daemon) for use for ezsp adapter in zigbee-herdsman to allow use of RPC firmware as alternative to NCP firmware? See feature request here:
> 
> [Koenkk/zigbee2mqtt#14497](https://github.com/Koenkk/zigbee2mqtt/issues/14497)
> 
> The real-world use-case is that some adapters will sooner or later probably start to ship with RPC (Radio Co-Processor) firmware instead of NCP (Network Co-Processor ) firmware by default (especially now that Matter 1.0 for Thread has been released) to enable concurrent protocols for running Zigbee and OpenThread at the same time on a single adapter by offloading parts of the protocol stacks to software running on the host.
> 
> https://openthread.io/platforms/co-processor
> 
> At least I think that the soon-to-be-released Home Assistant SkyConnect USB Stick (based on Silicon Labs EFR32MG21) will sooner or later ship with RPC firmware by default even if they have recently said that it will initially ship with NCP firmware:
> 
> https://www.home-assistant.io/skyconnect/
> 
> FYI, "Silicon Labs Zigbee/OpenThread Multiprotocol Add-on" will soon ship with Home Assistant to enable use of Silabs radios with RCP Multi-PAN firmware installed as those needs Silicon Labs Zigbee Daemon (Zigbeed), and I believe that they might have chosen to connect or map Zigbeed via a TCP socket port (for communicating with EZSP) instead of serial port(?).
> 
> https://github.com/home-assistant/addons/tree/master/silabs-multiprotocol
> 
> Zigbeed runs the EmberZNet Zigbee networking stack on the host instead of on the MCU when using RPC firmware, see related discussions:
> 
> [zigpy/zigpy#894](https://github.com/zigpy/zigpy/discussions/894)
> 
> [grobasoz/zigbee-firmware#27](https://github.com/grobasoz/zigbee-firmware/issues/27)
> 
> https://www.silabs.com/documents/public/application-notes/an1333-concurrent-protocols-with-802-15-4-rcp.pdf
> 
> https://www.silabs.com/documents/public/user-guides/ug103-03-fundamentals-design-choices.pdf
> 
> As I understand Zigbeed can be built from GSDK sources (Silicon Labs Zigbee Daemon)?
> 
> https://github.com/SiliconLabs/gecko_sdk
> 
> PS: zigpy bellows library does already have support for zigbeed and thus Silabs RPC firmware https://github.com/zigpy/bellows

Hedda, 2022-11-18:
> I don’t know whether it makes sense to bundle the matter devices compatibility on this project

FYI, there is a separate discussion with the suggestion to add Matter/CHIP support in the Z2M project or as a fork here -> https://github.com/Koenkk/zigbee2mqtt/discussions/7443

Koenkk, 2022-11-20:
> Adding just "Matter/CHIP bridge support" would still only support Zigbee devices however it would allow Zigbee2MQTT to act as a Matter bridge (as an alternative to MQTT)

This is what I'm planning to support (via https://github.com/mfucci/node-matter)

Hedda, 2022-12-12:
> kirovilya wondering if you have looked or are planing to look at the option to include Zigbeed (Silicon Labs Zigbee Daemon) for use for ezsp adapter in zigbee-herdsman to allow use of RPC firmware as alternative to NCP firmware? 
> 
> ...
> 
> The real-world use-case is that some adapters will sooner or later probably start to ship with RPC (Radio Co-Processor) firmware instead of NCP (Network Co-Processor ) firmware by default (especially now that Matter 1.0 for Thread has been released) to enable concurrent protocols for running Zigbee and OpenThread at the same time on a single adapter by offloading parts of the protocol stacks to software running on the host.
> 
> https://openthread.io/platforms/co-processor
> 
> At least I think that the soon-to-be-released Home Assistant SkyConnect USB Stick (based on Silicon Labs EFR32MG21) will sooner or later ship with RPC firmware by default even if they have recently said that it will initially ship with NCP firmware

FYI, it looks like Nabu Casa has officially released (experimental) RPC Multi-PAN firmware to allow multiprotocol support using their EmberZNet/EZSP based "Home Assistant SkyConnect" USB adapter as well as for the embedded radio module inside their "Home Assistant Yellow" appliance (both based on Silicon Labs EFR32MG21 SoC). If a user flash this RPC Multi-PAN firmware today then they can at this not use it with Zigbee2MQTT as zigbee-herdsman do not yet include support for Zigbeed (Silicon Labs Zigbee Daemon) to be able to use for ezsp adapters wih EmberZNet RPC firmware (instead of EmberZNet NCP firmware).

https://github.com/NabuCasa/silabs-firmware -> https://github.com/NabuCasa/silabs-firmware/tree/main/RCPMultiPAN

The reason for this is that Home Assistant 2022.12 has now been released with initial support for Matter including Matter over Thread with "Home Assistant SkyConnect" and "Home Assistant Yellow", so RPC Multi-PAN firmware for multiprotocol support is flashed if the user chooses to enable multiprotocol support in the "Home Assistant SkyConnect" and "Home Assistant Yellow" add-ons for Home Assistant, which is listed as an experimental feature. And at this point, moving back to an EmberZNet Zigbee NCP firmware requires manually reflashing the adapter.

https://www.crowdsupply.com/nabu-casa/home-assistant-yellow/updates/more-yellows-shipping-and-thread-plus-matter-updates

https://www.home-assistant.io/blog/2022/12/07/release-202212/

https://community.home-assistant.io/t/home-assistant-skyconnect-usb-stick-announced-will-be-compatible-with-both-zigbee-and-thread-including-matter-chip-over-thread/433594/

[image]

closed 2023-01-20 by github-actions[bot]

spartandrew18, 2023-01-24:
BUMP. The multipan firmware bug has now been fixed. I was wondering is there anyway to add the zigbee configuration to zigbee2mqtt? 

When I try to add the information below I get an error from the logs.

port: socket://core-silabs-multiprotocol:9999
adapter: ezsp

[17:13:38] INFO: Preparing to start...
[17:13:39] INFO: Socat not enabled
[17:13:41] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:info  2023-01-24 17:13:48: Logging to console and directory: '/config/zigbee2mqtt/log/2023-01-24.17-13-48' filename: log.txt
Zigbee2MQTT:info  2023-01-24 17:13:48: Starting Zigbee2MQTT version 1.29.2-dev (commit #7639ce4)
Zigbee2MQTT:info  2023-01-24 17:13:48: Starting zigbee-herdsman (0.14.88)
Zigbee2MQTT:error 2023-01-24 17:13:48: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-01-24 17:13:48: Failed to start zigbee
Zigbee2MQTT:error 2023-01-24 17:13:48: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-01-24 17:13:48: Exiting...
Zigbee2MQTT:error 2023-01-24 17:13:48: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open socket://core-silabs-multiprotocol:9999'
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:91:28)
    at SerialPort._error (/app/node_modules/@serialport/stream/dist/index.js:76:22)
    at /app/node_modules/@serialport/stream/dist/index.js:112:18

kirovilya, 2023-01-25:
@spartandrew18 your port name must started with 'tcp://", not "socket://"

samuelthng, 2023-02-15:
@spartandrew18 were you able to get it working?
I'm facing an issue where the port closes instead.

```
Zigbee2MQTT:info  2023-02-15 13:33:52: Starting Zigbee2MQTT version 1.30.1 (commit #unknown)
Zigbee2MQTT:info  2023-02-15 13:33:52: Starting zigbee-herdsman (0.14.89)
2023-02-15T05:33:52.307Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[20]},"serialPort":{"path":"tcp://core-silabs-multiprotocol:9999","adapter":"ezsp"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2023-02-15T05:33:52.310Z zigbee-herdsman:adapter:ezsp:uart Opening TCP socket with core-silabs-multiprotocol:9999
2023-02-15T05:33:52.324Z zigbee-herdsman:adapter:ezsp:uart Socket connected
2023-02-15T05:33:52.325Z zigbee-herdsman:adapter:ezsp:uart Socket ready
2023-02-15T05:33:52.325Z zigbee-herdsman:adapter:ezsp:uart Uart reseting
2023-02-15T05:33:52.325Z zigbee-herdsman:adapter:ezsp:uart --> Write reset
2023-02-15T05:33:52.327Z zigbee-herdsman:adapter:ezsp:uart --> [1ac038bc7e]
2023-02-15T05:33:52.341Z zigbee-herdsman:adapter:ezsp:uart <-- []
2023-02-15T05:33:52.342Z zigbee-herdsman:adapter:ezsp:uart <-- [c1020b0a527e]
2023-02-15T05:33:52.343Z zigbee-herdsman:adapter:ezsp:uart <-- RSTACK: c1020b0a527e
2023-02-15T05:33:52.344Z zigbee-herdsman:adapter:ezsp:uart RSTACK Version: 2 Reason: RESET_SOFTWARE frame: c1020b0a527e
2023-02-15T05:33:52.345Z zigbee-herdsman:adapter:ezsp:ezsp ==> version: {"desiredProtocolVersion":4}
2023-02-15T05:33:52.347Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"version","_id_":0,"_isRequest_":true,"desiredProtocolVersion":4}
2023-02-15T05:33:52.348Z zigbee-herdsman:adapter:ezsp:uart --> DATA (0,0,0): 00000004
2023-02-15T05:33:52.350Z zigbee-herdsman:adapter:ezsp:uart --> [004221a850ed2c7e]
2023-02-15T05:33:52.350Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (1)
2023-02-15T05:33:54.354Z zigbee-herdsman:adapter:ezsp:uart --> Error: Error: {"sequence":1} after 2000ms
2023-02-15T05:33:54.354Z zigbee-herdsman:adapter:ezsp:uart -!- break waiting (1)
2023-02-15T05:33:54.354Z zigbee-herdsman:adapter:ezsp:uart Can't send DATA frame (0,0,0): 00000004
2023-02-15T05:33:54.354Z zigbee-herdsman:adapter:ezsp:uart ->> DATA (0,0,1): 00000004
2023-02-15T05:33:54.355Z zigbee-herdsman:adapter:ezsp:uart --> [084221a850ef017e]
2023-02-15T05:33:54.355Z zigbee-herdsman:adapter:ezsp:uart -?- rewaiting (1)
2023-02-15T05:33:56.356Z zigbee-herdsman:adapter:ezsp:uart --> Error: Error: {"sequence":1} after 2000ms
2023-02-15T05:33:56.356Z zigbee-herdsman:adapter:ezsp:uart -!- break rewaiting (1)
2023-02-15T05:33:56.357Z zigbee-herdsman:adapter:ezsp:uart Can't resend DATA frame (0,0,1): 00000004
2023-02-15T05:33:56.357Z zigbee-herdsman:adapter:ezsp:driv Reset connection. Try 0
2023-02-15T05:33:56.357Z zigbee-herdsman:adapter:ezsp:driv Stop driver
2023-02-15T05:33:56.357Z zigbee-herdsman:adapter:ezsp:ezsp Stop ezsp
Zigbee2MQTT:error 2023-02-15 13:33:56: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-02-15 13:33:56: Failed to start zigbee
Zigbee2MQTT:error 2023-02-15 13:33:56: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-02-15 13:33:56: Exiting...
Zigbee2MQTT:error 2023-02-15 13:33:56: Error: Failure send version:{"type":"Buffer","data":[0,0,0,4]}
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:529:23
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
2023-02-15T05:33:56.475Z zigbee-herdsman:adapter:ezsp:uart Port closed
2023-02-15T05:33:56.476Z zigbee-herdsman:adapter:ezsp:ezsp Close ezsp
2023-02-15T05:33:56.477Z zigbee-herdsman:adapter:ezsp:driv Close driver
```

merlinpimpim, 2023-08-06:
Hello,
Any news on this topic since February?

Hedda, 2023-08-07:
> Any news on this topic since February?

In what regards? As you can read above it was not clear what the original poster was asking about. For example; If the question is about the Matter standard then instead see diacussion in  https://github.com/Koenkk/zigbee2mqtt/discussions/7443

merlinpimpim, 2023-08-07:
Thank you.

Yes, the point is to know how to make the SkyConnect stick work in multi-protocol RCP Multi-PAN mode (Zigbee/Thread, with ad hoc firmware) with Z2M in Home Assistant.